iNavFlight/inav #6338: NEO-M9N doesn't work with 2.6.0

slackista, 2020-11-23:
#### Current Behavior
Matek M9N-5883 doesn't work with 2.6.0 RC3

#### Steps to Reproduce
1. Soldier GPS to UART4.
2. Configure ports tab. Only check GPS in UART4. Save and reboot.
3. Enable "GPS for Navigation" and UBLOX7 protocol in Configuration tab. Save and reboot.

#### Expected behavior
GPS have fix(green led blinks) but no data in GPS tab and GPS has red label on top.

#### Additional context
I have connected that GPS unit with usb-ttl to u-center and works as expected.

inav-configurator 2.6-RC1
Matek405CTR
INAV/MATEKF405 2.6.0 Nov 20 2020 / 23:34:41 (27266e34)

digitalentity, 2020-11-23:
This is strange as M9N was tested on MATEKF405SE target was was working fine. Can you post a diff of your config and photo of how it's connected?

digitalentity, 2020-11-23:
Matek M9N-5883 connected to PIXRACER and working fine:

[image]

DzikuVx, 2020-11-23:
I have M9N connected to KakuteF7 and it's also working fine with 2.6

slackista, 2020-11-23:
```
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/MATEKF405 2.6.0 Nov 20 2020 / 23:34:41 (27266e34)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

# servo mix

# servo

# safehome

# logic

# gvar

# feature
feature GPS

# beeper

# map

# serial
serial 0 0 115200 115200 0 115200
serial 1 0 115200 115200 0 115200
serial 3 2 115200 115200 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 110
set gyro_lpf_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 120
set acc_hardware = MPU6000
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = NONE
set blackbox_rate_denom = 2
set applied_defaults = 2
set gps_provider = UBLOX7
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set mc_airmode_type = THROTTLE_THRESHOLD

# profile
profile 1

set mc_p_pitch = 44
set mc_i_pitch = 75
set mc_d_pitch = 25
set mc_i_roll = 60
set mc_p_yaw = 35
set mc_i_yaw = 80
set dterm_lpf_hz = 110
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 170
set dterm_lpf2_type = PT1
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set setpoint_kalman_enabled = ON
set setpoint_kalman_q = 200
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# battery_profile
battery_profile 1

# end the command batch
batch end
```

After command `gpspasstrough` I can see bytes in console. After closing configurator I open u-center and it works but poorly, I can't change baudrate and interface shows tearing. After minute several satiates in the u-center.

[image]
PS: I've checked connection wires with multimeter before power on.

stronnag, 2020-11-23:
And with `gps_ublox_use_galileo = OFF` ?

slackista, 2020-11-23:
* `gps_ublox_use_galileo = OFF` does nothing.
* `gps_auto_config = OFF`:
[image]

MATEKSYS, 2020-11-24:
Just tested. M9N works fine with F405-CTR & 2.6_RC3

@slackista Type CLI "defaults", reconfigure it

slackista, 2020-11-24:
Did it several times. Same behavior. I also have SAM-M8N and works without issues with the same UART on this FC.

slackista, 2020-11-24:
I think my M9N unit have defect. In statistic `TimeStartup` variable doesn't increment like it should.
### NEO-M9N:
[image]
### SAM-M8N:
[image]
[image]
### And iNav with M8N:
[image]

slackista, 2020-11-24:
It turns out I had faulty cheap usb-ttl and unable to properly reset cfg to default values of M9N-5883. After connecting to better adapter and reset to defaults it works. Thanks guys for support!

closed 2020-11-24 by slackista