iNavFlight/inav #6346: Unable to select cruise mode / alt hold or similar, with rotated FC

uk-Pedro, 2020-11-26:
There seems to be a bug whereby iNav will not go into a few modes if the board is rotated.  I can't select cruise mode or alt hold but RTH works fine.  FC is Matek 722 wing.  Below is my "diff"

# diff

# version
# INAV/MATEKF722SE 2.5.1 Jun 16 2020 / 12:47:07 (098a574b3)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 66 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -66 0 -1
smix 3 4 1 50 0 -1

# servo
servo 4 1000 2000 1500 -100

# logic

# gvar

# gf

# feature
feature -BLACKBOX
feature -AIRMODE
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 2 2 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 2100
aux 1 1 0 900 1150
aux 2 2 0 1200 1475
aux 3 3 2 1300 2100
aux 4 11 2 1725 2100
aux 5 10 0 1725 2100
aux 6 12 0 1500 1800
aux 7 37 3 900 1075
aux 8 21 3 1850 2100
aux 9 13 5 1600 2100
aux 10 42 4 1525 1700
aux 11 43 4 1450 1550
aux 12 44 4 1400 1475

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 1 13 V
osd_layout 0 1 1 15 H
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 5 23 15 V
osd_layout 0 6 23 14 V
osd_layout 0 7 1 1 V
osd_layout 0 9 1 12 V
osd_layout 0 10 1 4 V
osd_layout 0 11 7 15 V
osd_layout 0 12 13 15 V
osd_layout 0 13 25 0 V
osd_layout 0 14 9 1 V
osd_layout 0 15 20 0 V
osd_layout 0 20 19 2 V
osd_layout 0 21 19 1 V
osd_layout 0 22 14 1 V
osd_layout 0 23 14 0 V
osd_layout 0 28 23 14 H
osd_layout 0 30 2 9 V
osd_layout 0 32 1 14 V
osd_layout 1 32 1 14 V
osd_layout 2 22 14 1 V
osd_layout 2 32 1 14 V
osd_layout 2 34 9 15 H
osd_layout 3 13 25 0 V
osd_layout 3 32 1 14 V

# master
set acc_hardware = MPU6000
set acczero_x = 95
set acczero_y = -14
set acczero_z = -28
set accgain_x = 4111
set accgain_y = 4053
set accgain_z = 4016
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = ADC
set rssi_channel = 15
set serialrx_provider = CRSF
set serialrx_halfduplex = ON
set failsafe_procedure = RTH
set align_board_roll = 50
set align_board_yaw = -900
set current_meter_scale = 250
set platform_type = AIRPLANE
set model_preview_type = 8
set 3d_deadband_low = 1425
set 3d_neutral = 1475
set small_angle = 180
set disarm_kill_switch = OFF
set applied_defaults = 3
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 7500

# profile
profile 1

set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set rc_yaw_expo = 70
set roll_rate = 30
set yaw_rate = 10

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 330

# end the command batch
batch end

digitalentity, 2020-11-26:
If you are trying on the ground - this is expected. You need to be flying.

uk-Pedro, 2020-11-26:
> If you are trying on the ground - this is expected. You need to be flying.

Nope - flying.

digitalentity, 2020-11-26:
Can you record a blackbox log of the flight where, for example, althold did not engage?

uk-Pedro, 2020-11-26:
> Can you record a blackbox log of the flight where, for example, althold did not engage?

Great idea.  I'll do that next time I'm out.  Thanks for the suggestion.

uk-Pedro, 2020-12-05:
I went out today to log a flight with blackbox.  Tried heading hold and alt hold in manual mode, no good, didn't work.  Tried it in angle mode, perfect, worked like a charm.  I must have missed that in the docs - my bad.

Thanks for your suggestion digitalentity.  I've not uploaded as the issue was pilot error or documentation error.

closed 2020-12-05 by uk-Pedro