iNavFlight/inav #6348: No DJI osd using FLYWOOF411 on Flywoo GOKU GN413s AIO

GavFPV, 2020-11-28:
## Current Behavior

No Voltage or DJI osd output from this target on this board.
I have tried every combination known to man, then tested in betaflight and it works as expected, i believe it might be target issue? 
Happy to test if someone can have a look 

## Steps to Reproduce

1. Flash, and it doesn't work, works fine on betaflight
2.
3.
4.

## Expected behavior

OSD to work 

## Suggested solution(s)

No idea, but happy to test if someone can have a look at this target 

## Additional context

Also tested the new 2.6RC, rolled back inav configurator, every combination of msp or DJI VTX combo in the ports tab, just in case, rebuilt the quad twice, until I tested in BF and it worked.
really want to use Inav for alt hold on this project 

---

- FC Board name and vendor: Flywoo GOKU GN413s AIO
- INAV version string:  INAV/FLYWOOF411 2.5.2 Aug  4 2020 / 10:52:51 (faaedc7cb)

digitalentity, 2020-11-29:
Please share your config `diff` or a screenshot of your Ports tab in the Configurator.

GavFPV, 2020-11-29:
I'm at work today, but can get one when home, I tried every combination of uarts, dji vtx on and off, msp on and off
When I tested in betaflight, it worked as expected, flash back to inav and on osd no matter what settings I tried

digitalentity, 2020-11-30:
Please note,  that Betaflight and INAV are different in many settings. You issue is likely a settings problem. Please share your `diff`

GavFPV, 2020-11-30:
Hi,

Below is the diff all

Diff all

# version
# INAV/FLYWOOF411 2.5.2 Aug  4 2020 / 10:52:51 (faaedc7cb)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# logic

# gvar

# gf

# feature
feature -SOFTSERIAL
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 64 115200 115200 0 115200
serial 1 2097152 115200 115200 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 110
set gyro_lpf_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 120
set acc_hardware = MPU6000
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set motor_pwm_rate = 16000
set motor_pwm_protocol = DSHOT600
set model_preview_type = 3
set applied_defaults = 2
set mc_airmode_type = THROTTLE_THRESHOLD

# profile
profile 1

set mc_p_pitch = 44
set mc_i_pitch = 75
set mc_d_pitch = 25
set mc_i_roll = 60
set mc_p_yaw = 35
set mc_i_yaw = 80
set dterm_lpf_hz = 110
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 170
set dterm_lpf2_type = PT1
set use_dterm_fir_filter = OFF
set dterm_setpoint_weight =  0.750
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

#

GavFPV, 2020-11-30:
[Image]

Also here is a pic of the ports tab

ShikOfTheRa, 2020-12-02:
Just as an fyi, I was having problems with betaflight and I found there are different versions of the Goku 411 FC using different resource mappings.
Haven't seen it documented anywhere but figured it out by looking at some posted CLI dumps from betaflight before it was confirmed by Flywoo.

Suggest contacting Flywoo just in case they did the same with the 413

KnzHz, 2020-12-02:
I seem to have a version2, where soft serial is motor5 and 6, it seems older board had it mapped some random places ;) 

I'm new to iNav(as in just flashed it) but can understand that resource remapping is not something iNav does, you compile a new target with the resources where you want them correct? 

I have the same problem, but with my GPS as I switched OSD and GPS(as OSD requires way more CPU than GPS, at least on BF). 

Are the resource points the same between BF and iNav? What I mean is "A10" the same "A10" in both firmwares?

GavFPV, 2020-12-03:
> I seem to have a version2, where soft serial is motor5 and 6, it seems older board had it mapped some random places ;)
>
> I'm new to iNav(as in just flashed it) but can understand that resource [...]

I think your onto something, below is the betaflight resource and the inav

Betaflight

resource BEEPER 1 C14
resource MOTOR 1 A08
resource MOTOR 2 A09
resource MOTOR 3 A10
resource MOTOR 4 B00
resource MOTOR 5 B04
resource PPM 1 A02
resource PWM 1 B01
resource PWM 2 A03
resource LED_STRIP 1 A15
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C13
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource CAMERA_CONTROL 1 B10
resource ADC_BATT 1 A00
resource ADC_RSSI 1 B01
resource ADC_CURR 1 A01
resource PINIO 1 B05
resource FLASH_CS 1 B02
resource OSD_CS 1 B12
resource GYRO_EXTI 1 B03
resource GYRO_CS 1 A04
resource USB_DETECT 1 C15

INAV

A00: ADC CH2
A01: FREE 
A02: SOFTSERIAL1 UART TX
A03: SERIAL2 UART RX
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MOTOR1 OUT
A09: MOTOR2 OUT
A10: MOTOR3 OUT
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: FREE 
B00: MOTOR4 OUT
B01: FREE 
B02: FLASH CS
B03: MPU EXTI
B04: FREE 
B05: FREE 
B06: FREE 
B07: SERIAL1 UART RX
B08: I2C1 SCL
B09: I2C1 SDA
B10: FREE 
B11: FREE 
B12: OSD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: FREE 
C03: FREE 
C04: FREE 
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: FREE 
C11: FREE 
C12: FREE 
C13: LED1 OUT
C14: BEEPER OUT
C15: FREE 
D02: FREE

GavFPV, 2020-12-03:
uarts aren't set up as full ones in INAV compared to Betaflight 

Anyone know how to do a custom build for INAV?

digitalentity, 2020-12-03:
It seems we need a new target for FLYWOOF411 V2

KnzHz, 2020-12-03:
After playing with it, I got GPS to work, apparently UART1 is the resources that are soft serial in BF(which is like, whaaaat?). My guess is that this target was made with v2.1(which is what I think mine is based on https://flywoo.net/pages/q-a-about-explorer-lr) , so people with the OG board needs another target. In this file: https://cdn.shopify.com/s/files/1/0010/7410/2324/files/FLYWOO_EXPOLR_LR_4INCH-VISTA_HD_BF4.11_CLI_201029.rar?v=1603913157 there's dumps to the different versions, I assume one can map the resources from that?

ShikOfTheRa, 2020-12-03:
If not in a hurry I can have a look at identifying / creating ones for Flywoo goku v1 and v2 goku f411 this next week. Unfortunately I'm tied up with work at moment.

GavFPV, 2020-12-09:
any luck?

ShikOfTheRa, 2020-12-09:
@GavFPV Really the manufacturer should be doing this. its a time consuming task trying to cross check and migrate.  
You can give this a try..

I have tested basic functionality only on a goku F411 V2. (motors operate correctly / GSP only. ) I do not know if it is the same, but there is more working than before!
So make sure you have everything backed up and test carefully...! 

[inav_2.6.0_FLYWOOF411GV2.zip](https://github.com/iNavFlight/inav/files/5667705/inav_2.6.0_FLYWOOF411GV2.zip)

GavFPV, 2020-12-09:
Thanks for your help, I have contacted flywoo, not getting alot of traction with them, their resource list was different from mine so am chasing a target from them too

KnzHz, 2020-12-11:
@ShikOfTheRa I tried your build but there's still resources that are set wrong, motors work now though. Is there a wiki page or resource where I can read how I do my own builds? So I don't have to bother you each of my tests. I have version 2.1.

I assume resource mapping is the same as Betaflight, although it's done on build, By that a mean, the pins on the chip should still point to the same resources in both firmware types, yes?

ShikOfTheRa, 2020-12-11:
Damn - I was just going to put in a Pull request.
What doesn't work? I tried on mine and verified:
Motors 1-4 correct operation
Serial1 - CRSF
Serial2 - GPS
SoftSerial - VTX control
Ledstrip
Buzzer
Voltage

Untested:
I2C - but that didn't change from original F411 pins so should be OK.
Current sensor

My branch is here:
https://github.com/ShikOfTheRa/inav/tree/FlywooF411

Build guide:
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20with%20Linux%20Subsystem.md

KnzHz, 2020-12-11:
Nice, I've reversed DJI and GPS so that GPS is on a Softserial and DJI is on a real UART as that at least is a mere supported way in BF. I can get the GPS to work, but not the DJI, maybe I'm just doing it wrong ;)

Do you have the v2.1?

I'll test some more to see if it's me that's just an idiot ;) I've never used iNav but I'm an old Cleanflight and current Betaflight user.

ShikOfTheRa, 2020-12-11:
I kept it as close to original as possible, but maybe try this - with ppm support disabled as it uses shared pin with serial 2. I assumed they can co-exist if PPM not used but maybe this is not possible. 
[inav_2.6.0_FlywooGokuF411.zip](https://github.com/iNavFlight/inav/files/5679043/inav_2.6.0_FlywooGokuF411.zip)

KnzHz, 2020-12-11:
Okay, so I might be a complete tool but I can't get GPS to work on the softserial port, I don't even see softserial mapped in resource. I've set Softserial1 to GPS, when I try to enable it in configuration, it is disabled on reboot. I was pretty sure I had GPS working before I I fiddled with the DJI setup, I have no idea what I've differently. This is with your initial v2. I got motors and OSD working. AFAIK from the BF dump SoftSerial should be B06 and b07, but they are "FREE", though that might some configuration thing as it seems resources are only displayed once in use in iNav. I'm almost positive that I'm the error and not your build :P

ShikOfTheRa, 2020-12-11:
Is your board definitely a v2.1 - it should have 4 pads like in picture:
https://cdn.shopifycdn.net/s/files/1/0010/7410/2324/products/20201126012224_2048x2048.png?v=1606326840

This is V2 from last version posted with CRSF on Ser 1, GPS on Ser 2 and VTX on soft ser 1 (but single wire TX only... )

[image]

KnzHz, 2020-12-11:
This is what I don't have when settings Softseriel to GPS:

B06: SOFTSERIAL1 UART TX
B07: SOFTSERIAL1 UART RX

[image]

This is what I can't enable:

[image]

On the v1(or official) build, the softserial pins m5 and m6 on the v1 or mapped as uart1, and with that I can get GPS to work. 

This is my dump from resource:

# resource
IO:
----------------------
A00: FREE 
A01: ADC CH1
A02: SERIAL2 UART TX
A03: SERIAL2 UART RX
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MOTOR1 OUT
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: MOTOR4 OUT
B00: FREE 
B01: ADC CH2
B02: FLASH CS
B03: MOTOR2 OUT
B04: FREE 
B05: MPU EXTI
B06: FREE 
B07: FREE 
B08: I2C1 SCL
B09: I2C1 SDA
B10: MOTOR3 OUT
B11: FREE 
B12: OSD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: FREE 
C03: FREE 
C04: FREE 
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: FREE 
C11: FREE 
C12: FREE 
C13: LED1 OUT
C14: BEEPER OUT
C15: FREE 
D02: FREE 

and version:

# version
# INAV/FLYWOOF411GV2 2.6.0 Dec  9 2020 / 17:26:09 (ac9d91bd)
# GCC-9.3.1 20200408 (release)

ShikOfTheRa, 2020-12-11:
That version one looks like first cut from a couple of days ago...
I have just re-uploaded with different name in case github is caching or something.

I have just flashed this version quick test on serial port to be sure and you should get the following.
INAV/FLYWOOF411GV2 2.6.0 Dec 11 2020 / 13:44:03 (8d1b3f15)

[inav_2.6.0_FLYWOOF411GV2_latest.zip](https://github.com/iNavFlight/inav/files/5679932/inav_2.6.0_FLYWOOF411GV2_latest.zip)

KnzHz, 2020-12-11:
Holy smokes batman!

It worked with the one you just uploaded!

Thanks!

ShikOfTheRa, 2020-12-11:
Excellent! 

I uploaded with same name - I wonder in github did not like that. I have removed previous links to avoid anyone misusing.
Thanks for confirming with testing - it helped confirm serial operation and we now have bidirectional soft serial which I did not add before.

There are a lot of these out there in the FlyWoo Micro Long Range and I suspect a fair few guys will want to use iNav on it.

Test carefully and let us know how it goes. and I'll submit a PR

I see there is a spare pin for CC on the diag I guess this is camera control but it is not used in betaflight dump. Potentially that could be used for something, but no idea what it connects to so it will have to remain until we find details.

KnzHz, 2020-12-11:
Perfect ;)

I will test this when it stops raining all the time here in Denmark ;)

My next plan is to switch the GPS with one with compass and solder SCL and SDA, the resources are mapped to the same pin so I assume(tm) it'll work fine. I don't have a GPS with compass on hand so that'll have to wait until my children has raped my wallet for Christmas ;)

Yeah, I think Flywoo did bear minimum to get it to work. I bought this so late that I didn't see the original BF4.2 dump, only the "OSD fix" 4.1.1 - maybe it CC mapped. Anyway, for digital I don't need it but yeah, CC should have a timer so it might be useful if you needed another accessory that could use 1pin uart. 

A quick hover test with GPS arming disabled shows that it at least flies now, woohoo!

ShikOfTheRa, 2020-12-11:
@GavFPV could you post a resource map from a know good working config of the GN413 AIO ? See if we can get that one running too

GavFPV, 2020-12-12:
Hi,

Here is the resource dump from my board, this is the one where the DJI osd will not work.
I have since changed out the fc, but i make something up to test 

I haven't had a chance to test the target posted above 

# resource
IO:
----------------------
A00: ADC CH2
A01: FREE 
A02: FREE 
A03: SERIAL2 UART RX
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MOTOR1 OUT
A09: MOTOR2 OUT
A10: MOTOR3 OUT
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: FREE 
B00: MOTOR4 OUT
B01: FREE 
B02: FLASH CS
B03: MPU EXTI
B04: FREE 
B05: FREE 
B06: SERIAL1 UART TX
B07: SERIAL1 UART RX
B08: I2C1 SCL
B09: I2C1 SDA
B10: FREE 
B11: FREE 
B12: OSD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: FREE 
C03: FREE 
C04: FREE 
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: FREE 
C11: FREE 
C12: FREE 
C13: LED1 OUT
C14: BEEPER OUT
C15: FREE 
D02: FREE

GavFPV, 2020-12-12:
below is a resource dump from betaflight where the osd works fine on uart 2

Flywoo411 is the target 

# resource
resource BEEPER 1 C14
resource MOTOR 1 A08
resource MOTOR 2 A09
resource MOTOR 3 A10
resource MOTOR 4 B00
resource MOTOR 5 B04
resource PPM 1 A02
resource PWM 1 B01
resource PWM 2 A03
resource LED_STRIP 1 A15
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C13
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource CAMERA_CONTROL 1 B10
resource ADC_BATT 1 A00
resource ADC_RSSI 1 B01
resource ADC_CURR 1 A01
resource PINIO 1 B05
resource FLASH_CS 1 B02
resource OSD_CS 1 B12
resource GYRO_EXTI 1 B03
resource GYRO_CS 1 A04
resource USB_DETECT 1 C15

GavFPV, 2020-12-12:
@ShikOfTheRa 
I have tested target inav_2.6.0_FLYWOOF411GV2.zip as above and the osd works on uart2.

 Nice work!

There is an issue with the board detecting battery voltage, it reads 9v when plugged into usb and also a 4s lipo, when i turn on current sensor this then reads 3.8v and doesn't change, might be a setting I'm missing somewhere

ShikOfTheRa, 2020-12-12:
@GavFPV – at a quick look that resource list from BF looks pretty much like the stock Flywoo f411,
But 
Flywoo F411 has:
UART1 TX and RX (2 pins)
UART2 RX single pin
Softerial single wire TX/RX on the UART2 TX
So should be connected accordingly

In your working BF resources you have the following
UART1 TX and RX (2 pins)
UART2 TX and RX (2 pins)
No Soft serial

So probably generating an alternative F411 target would be appropriate if cannot connect up to standard 411. The bat/curr will start working then too!

What exactly do you have connected to al the serial pins?

GavFPV, 2020-12-12:
I have crossfire on uart 1 both Rx and tx
And dji vtx on uart 2 both Rx and tx

KnzHz, 2020-12-12:
> So probably generating an alternative F411 target would be appropriate if cannot connect up to standard 411. The bat/curr will start working then too!

There's also an aio board that uses the same target, I have one I haven't used for anything yet but I'll hook it up to my computer later at look at the resource mapping in BF and see if it's the same as version2. Flywoo has some serious documentation issues. The website says that the v2 doesn't have barometer but both iNav and BF has it as working? 

I took mine out for one battery between rainshowers today and it flew perfectly, I can't believe flywoo didn't invest 5 more seconds into the explorer LR and put iNav and a compass on it! 

So from my pov : Target approved ;-) I flew it on 4s and at least the osd showed correct voltage for me ;-)

ShikOfTheRa, 2020-12-12:
@KnzHz  – thanks for feedback and testing

@GavFPV 

So have had a think and maybe this is best way forward to maximise compatibility for others:
Serial1 and 2 are full hardware ports so can be used for vista etc.
Soft Serial is single wire only and available on M5 if anyone needs it
Still retain RSSI, Current, voltage and ledstrip. Hopefully :)

Could do more if knew what pins mapped to what pads on these boards.  KnzHz is absolutely correct – Flywoo documentation absolutely sucks. 

Anyway, try this one for your 413 board
[inav_2.6.0_FLYWOOF411_HWSRL.zip](https://github.com/iNavFlight/inav/files/5683721/inav_2.6.0_FLYWOOF411_HWSRL.zip)

KnzHz, 2020-12-12:
Okay, I found the print from my flywoo box. Apparently version 2.0 has barometer but v2.1 doesn't. I just got a second unit yesterday that I hadn't unpacked, it has v2.1 and the print says there's no barometer and Betaflight won't detect one either. They're not making it easy. I have no idea where I put the AIO right now, so I'll have to look for it tomorrow. 

@ShikOfTheRa You have the 2.1 right? Does it have Baro in iNav?

If not, then there's apparently 4 versions, as there's 1.0, 1.2 and 2.0 in the file I linked earlier from their website. Weird.

ShikOfTheRa, 2020-12-13:
@KnzHz 
I guess mine is 2.0  as it has the Baro - baro show in iNav. I'll look next time I take frame apart.
The target is still the same as 2.1 . iNav will autodetect if the Baro is there or not.

Its a great little 16mm FC if they leave it with Baro on. They absolutely should leave it on when selling it with Micro Long Range - per4fect combo. 

Anyway, lets hope we have sorted @GavFPV  board too. I am quietly confident :)

KnzHz, 2020-12-14:
ON TOPIC: Put a few batteries through it yesterday. Works perfectly.

GENERAL F411 from flywoo stuff:
Maybe they just disabled BARO in firmware, I'll try to flash your iNav build to my new one and see if Baro's enabled. The support website says to disable Baro as it uses to much cpu(as the default is MSP on SoftSerial) so maybe it's just an effort from their side to not crash the cpu. I'll test. Yeah, I just testet: No Baro. Weird company.

Anyway, I found my AIO board that uses the same target: 

https://flywoo.net/products/flywoo-goku-gn413s-f411-f4-flight-controller-aio-osd-bec-13a-bl_s-2-4s-4in1-esc-for-rc-drone-fpv-racing?_pos=1&_sid=665356fc8&_ss=r

It doesn't make life easier. The motor mappings are as v1 but there's no softserial set by default in the BF conf from Flywoo. I assume it would be prudent to use m5 and led. 

To make matters worse, mine has to pads that are not on the pictures on the website: 
[image]

I can't tell if it says "p85" and "p810" or "PB5" and "PB10" If it refers to "Pin B5 and Pin B10" then it's resource mapped in BF as PINIO and Camera Control respectively. No SCA and SDL pads though, so the AIO can't be used with Compass, sucks.  

I'm trying to get some more info from Flywoo!

ShikOfTheRa, 2020-12-14:
So This is what I think. Just the two targets for now
1 FLYWOOF411 (the original)
2 FLYWOOF411GV2 (Flywoo Goku V2 with 2 wire soft serial on M5M6) for those who have a Baro on board

V2 is probably a good base for future targets if Flywoo.
I'll update the flywoo md docs if targets reach master.

After a lot of wasted time, research and reading I am not sure it is worth doing any more targets as the boards either do not have baro or I2c... Guess they were never intended to use iNav. I do like the Goku 411 v2 with baro - nice little board.

I have also had  no more response from Flywoo. 

@KnzHz "Weird company." - oh yes!!

KnzHz, 2020-12-14:
I tend to agree, I will consider myself lucky to have gotten the exact right version that is the most iNav friendly ;)

I guess I'll buy a new stack for the other one or use it for something else. 

Thank you for all your help!

GavFPV, 2020-12-14:
Thanks all for your input, but I moved to a different board for my current project, hope flywoo figures this all out

ShikOfTheRa, 2020-12-14:
Thanks for testing 

PR raised for V2 target
https://github.com/iNavFlight/inav/pull/6400

Think this can be closed.

GQ4Life, 2021-03-10:
> @KnzHz
> I guess mine is 2.0 as it has the Baro - baro show in iNav. I'll look next time I take frame apart.
> The target is still the same as 2.1 . iNav will autodetect if the Baro is there or not. [...]

Yeah pretty sure my flywoo f411 explorer has baro. Doesn't work on the inav 2.5.2 or 2.6.1 firmware. None of the resources match.  But everything works on betaflight.  
Betaflight some of the resources
resource BEEPER 1 C14
resource MOTOR 1 A08
resource MOTOR 2 B03
resource MOTOR 3 B10
resource MOTOR 4 A15
resource PPM 1 A02
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 11 B06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 11 B07
resource ADC_BATT 1 B01
resource ADC_CURR 1 A01

iNAV firmware I had that didn't work resources
A00: ADC CH2
A01: FREE 
A02: FREE 
A03: SERIAL2 UART RX
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MOTOR1 OUT
A09: MOTOR2 OUT
A10: MOTOR3 OUT
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: FREE 
B00: MOTOR4 OUT
B01: FREE 
B02: FLASH CS
B03: MPU EXTI
B04: FREE 
B05: FREE 
B06: SERIAL1 UART TX
B07: SERIAL1 UART RX

How can I see resources on firmware before trying?

DzikuVx, 2021-11-06:
This was fixed in 3.0.1

closed 2021-11-06 by DzikuVx