iNavFlight/inav #6349: Right servo moving when throttling

uldariel, 2020-11-29:
## Current Behavior
The right servo is moving when I push the throttle. It seems like the motor output is also being sent to the right servo (output 2 of the FC). The servos are working ok on roll and pitch, I have checked I have no mixes on my transmitter, or in inav. The bar showing the servo movement stays on 1500 while reving on inav. I am using an omnibus clone from aliexpress on a fixed wing with a single motor

## Steps to Reproduce
1. Flash Omnibus F4 V3 target
2. Set the mixes
3. Arm the controller and push the throttle

## Expected behavior
I expected the servos to be isolated from the motor but one of them is moving along with the thorttle

## Suggested solution(s)
I don´t know, may it be a bug?

## Additional context
https://pastebin.pl/view/9bc67cff

---

# version
# INAV/OMNIBUSF4V3 2.5.2 Aug  4 2020 / 10:05:34 (faaedc7cb)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

- FC Board name and vendor: Omnibus F4 clone from aliexpress (I don´t know if I can paste the link)
- INAV version string: 2.5.2

stronnag, 2020-11-29:
User error, you have connected the servo to a motor output. Please check the board PWM allocation  https://github.com/iNavFlight/inav/wiki/Boards,-Targets-and-PWM-allocations#target-omnibusf4v3

uldariel, 2020-11-29:
The connection is correct. The motor is on M1 on the board and the servos on M3 and M4. 
[image]

stronnag, 2020-11-29:
Following a telegram chat the following is known:
* Servos are connected to M3 / M4
* radio is Futaba T12FG
* map is ETRA
* stick movement show correctly in the configurator RX tab.
* Mix looks OK
```
smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1
```

closed 2021-04-07 by DzikuVx