iNavFlight/inav #6350: Matek hex files missing F411wse board and matek f411 wing - DJI variants softserial

Romulusg, 2020-11-29:
2 matek inav targets missing

inav_2.5.1_MATEKF411_S5S6_SS2.hex

inav_2.5.1_MATEKF411SE_S6_SR1.hex

These allow for DJI on softserial and are missing in the 2.6.0 RC3 list and having warnings implemented as the notes described.

Thanks

closed 2021-05-08 by DzikuVx

alexvazquez, 2021-05-13:
@DzikuVx Hi Pawel, I flashed my Matek F411WSE with the latest INAV 2.6.1 and Caddx Vista Softserial stopped working. The only way to make it work is to go back and flashed 2.5.1.  And I want to have 2.6.1 because want to use the new dji_use_name_for_messages to get warnings in the craft name.

alexvazquez, 2021-05-13:
> 2 matek inav targets missing
> 
> inav_2.5.1_MATEKF411_S5S6_SS2.hex
> 
> inav_2.5.1_MATEKF411SE_S6_SR1.hex
> 
> These allow for DJI on softserial and are missing in the 2.6.0 RC3 list and having warnings implemented as the notes described.
> 
> Thanks

@Romulusg I tried to flash to my Matek F411WSE firmware version 2.6.1 and I can confirm that DJI on softserial doesn't work. The only way I can make it work is using version 2.5.1. Any clue?