iNavFlight/inav #6364: Configuration tab wont open in Configurator 2.5

CMFischer, 2020-12-03:
Matek 405SE board flashed with firmware V2.5.2.  Configurator (V2.5) was working fine, but suddenly wont open the Configuration tab...shows "waiting for data" but never opens.  It opens all other tabs.  I have tried the Chrome App 2.5 version as well and it does same thing. 

Tried closing configurator, unplugging usb and start all over.....same thing happened. 

I saw someone else had this issue, but it was closed after they reflashed FC.  I dont want to start all over though. 

****
Here's info from Configurator upon connecting board:

2020-12-03 @ 11:36:58 -- Running - OS: MacOS, Chrome: 78.0.3904.87, Configurator: 2.5.0

2020-12-03 @ 11:36:59 -- Loaded release information from GitHub.

2020-12-03 @ 11:37:09 -- Serial port successfully opened with ID: 1

2020-12-03 @ 11:37:09 -- MultiWii API version received - 2.4.0

2020-12-03 @ 11:37:09 -- Flight controller info, identifier: INAV, version: 2.5.2

2020-12-03 @ 11:37:09 -- Running firmware released on: Aug 4 2020 10:24:57

2020-12-03 @ 11:37:09 -- Board: MF4S, version: 0

2020-12-03 @ 11:37:09 -- Unique device ID received - 0x25001f524b500320383548

CMFischer, 2020-12-04:
Issue closed.    When I tried to arm, it would not.....due to prearm condition  'Settings not Validated'.  My gyro_hardware_lpf was not an acceptable value. I set it at 188HZ and rebooted.  FC now arms AND Configuration tab opens in the Configurator.  Thanks to Pawel's youtube video I was able to diagnose and fix.  Thank you Pawel!  BTW, you could play a Russian gangster in the movies.

closed 2020-12-04 by CMFischer

agniel, 2021-01-03:
Had the same issue, thanks for the fix.