iNavFlight/inav #6380: iNav 2.6: RunCam protocol - in flight camera reset

Pairan, 2020-12-05:
## Camera on RunCam Protocol seam to reboot randomly in flight! iNav 2.6 RC3 and 2.6.0
I have a RunCam Hybrid connected to a UART on a Matek F722SE Board. Configuring the camera by a QR code and START/STOP recording HD is the desired feature which works pretty well in iNav v.2.5.x

With v2.6RC3 the camera suddenly reboots in flight! HD gets turned of and this ruines the recording ;) aside of being blind for 
some seconds wherever you currently are! There must be something causing this in v2.6x

### Video of behaviour in iNav v2.6 RC3
https://youtu.be/zBbMEK2RSzY

### Video of behaviour in iNav v2.6.0
https://youtu.be/zBbMEK2RSzY

## Steps to Reproduce
1. Launch
2. start the recording
3. fly! ... at some point it will happen

## Expected behavior
The camera should not reboot

## Additional context / DIFF of the Default 7" Multirotor Preset
`# reset configuration to default settings
defaults noreboot

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

smix 0 1 18 100 0 -1

# logic
logic 0 0 -1 3 2 1 0 100 0

# feature
feature -TX_PROF_SEL
feature BAT_PROF_AUTOSWITCH
feature GPS
feature PWM_OUTPUT_ENABLE

# map
map ERTA

# serial
serial 0 2 115200 115200 0 115200
serial 2 2048 115200 115200 0 115200
serial 3 1024 115200 115200 0 115200

# aux
aux 0 0 1 1500 2100
aux 1 2 0 1450 2100
aux 2 11 0 1150 1250
aux 3 10 0 1075 1150
aux 4 13 4 1800 2100
aux 5 40 6 1800 2100
aux 6 41 6 900 1200
aux 7 42 2 1100 1325
aux 8 43 2 1325 1675
aux 9 44 2 1675 2100

# adjrange
adjrange 0 0 5 900 1050 49 3
adjrange 1 0 5 1050 1150 15 3
adjrange 2 0 5 1150 1250 16 3
adjrange 3 0 5 1250 1350 17 3
adjrange 4 0 5 1350 1450 18 3
adjrange 5 0 5 1450 1550 19 3
adjrange 6 0 5 1550 1650 20 3

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 26 2 V
osd_layout 0 1 1 9 V
osd_layout 0 4 8 6 H
osd_layout 0 7 21 1 V
osd_layout 0 8 9 0 V
osd_layout 0 9 23 14 H
osd_layout 0 10 1 1 H
osd_layout 0 11 2 11 V
osd_layout 0 12 1 10 V
osd_layout 0 13 2 7 H
osd_layout 0 14 25 1 V
osd_layout 0 15 23 6 H
osd_layout 0 22 6 1 V
osd_layout 0 23 1 1 V
osd_layout 0 24 12 2 V
osd_layout 0 26 24 8 V
osd_layout 0 28 23 10 V
osd_layout 0 29 23 9 H
osd_layout 0 30 0 12 V
osd_layout 0 31 25 3 V
osd_layout 0 33 24 11 V
osd_layout 0 34 10 1 V
osd_layout 0 38 1 8 V
osd_layout 0 40 1 2 V
osd_layout 0 46 22 12 H
osd_layout 0 47 22 9 H
osd_layout 0 85 2 7 V
osd_layout 0 87 1 3 V
osd_layout 0 96 24 7 V
osd_layout 0 97 8 14 V
osd_layout 0 105 24 2 V
osd_layout 1 1 1 14 V
osd_layout 1 11 2 13 V
osd_layout 1 15 24 2 V
osd_layout 1 22 6 1 V
osd_layout 1 23 1 1 V
osd_layout 1 30 1 10 V
osd_layout 1 31 1 14 H
osd_layout 1 35 22 13 H
osd_layout 1 45 0 0 V
osd_layout 1 48 1 11 H
osd_layout 1 49 1 12 H
osd_layout 1 98 23 1 V
osd_layout 1 99 25 3 H
osd_layout 2 2 0 0 V
osd_layout 2 3 8 6 V
osd_layout 2 7 24 1 V
osd_layout 2 15 1 1 V
osd_layout 2 16 1 1 H
osd_layout 2 22 23 1 V
osd_layout 2 26 6 1 V
osd_layout 2 41 1 2 H
osd_layout 2 45 0 0 V
osd_layout 2 63 1 2 H
osd_layout 2 98 0 0 V
osd_layout 2 99 1 3 V
osd_layout 3 1 2 13 V
osd_layout 3 10 2 13 H
osd_layout 3 12 2 14 V
osd_layout 3 16 2 4 V
osd_layout 3 17 2 5 V
osd_layout 3 18 2 6 V
osd_layout 3 20 13 3 H
osd_layout 3 21 13 4 H
osd_layout 3 22 7 1 V
osd_layout 3 23 2 1 V
osd_layout 3 37 9 12 H
osd_layout 3 41 23 3 V
osd_layout 3 42 23 2 V
osd_layout 3 46 22 13 H
osd_layout 3 47 22 11 H
osd_layout 3 56 2 7 V
osd_layout 3 58 2 8 V
osd_layout 3 63 18 12 H
osd_layout 3 97 13 1 V
osd_layout 3 106 2 10 V

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 100
set gyro_lpf_type = PT1
set gyro_stage2_lowpass_hz = 160
set gyro_stage2_lowpass_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 200
set dynamic_gyro_notch_min_hz = 80
set acc_hardware = MPU6000
set acczero_x = 58
set acczero_y = -1
set acczero_z = -123
set accgain_x = 4065
set accgain_y = 4108
set accgain_z = 4032
set align_mag = CW90FLIP
set mag_hardware = HMC5883
set magzero_x = -52
set magzero_y = -14
set magzero_z = 39
set maggain_x = 690
set maggain_y = 613
set maggain_z = 498
set baro_hardware = BMP280
set pitot_hardware = ADC
set min_check = 1050
set serialrx_provider = CRSF
set blackbox_rate_denom = 2
set motor_pwm_protocol = DSHOT600
set throttle_idle =  7.000
set failsafe_procedure = RTH
set model_preview_type = 3
set applied_defaults = 2
set gps_ublox_use_galileo = ON
set mc_airmode_type = THROTTLE_THRESHOLD
set nav_rth_alt_mode = AT_LEAST_LINEAR_DESCENT
set nav_rth_altitude = 80
set osd_video_system = PAL
set i2c_speed = 800KHZ
set tz_offset = 2
set tz_automatic_dst = EU
set vtx_power = 5
set vtx_low_power_disarm = UNTIL_FIRST_ARM

# profile
profile 1

set mc_p_pitch = 44
set mc_i_pitch = 60
set mc_d_pitch = 25
set mc_i_roll = 50
set mc_d_roll = 25
set mc_p_yaw = 45
set mc_i_yaw = 70
set dterm_lpf_hz = 90
set dterm_lpf_type = PT1
set dterm_lpf2_type = PT1
set yaw_lpf_hz = 30
set mc_iterm_relax = RPY
set d_boost_factor =  1.500
set d_boost_max_at_acceleration =  5000.000
set d_boost_gyro_delta_lpf_hz = 65
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set setpoint_kalman_enabled = ON
set setpoint_kalman_q = 200
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# battery_profile
battery_profile 1

set bat_cells = 6
set vbat_min_cell_voltage = 290
set vbat_warning_cell_voltage = 310
set battery_capacity = 3100

# restore original profile selection
profile 1
battery_profile 1`
---
- FC Board name and vendor: MATEKF722SE / Mateksys
- INAV version string:  2.6.0 Nov 20 2020 / 23:58:58 (27266e34)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Pairan, 2020-12-10:
could somebody confirm this?

closed 2021-01-22 by Pairan

Jetrell, 2021-02-04:
@Pairan how did you go with this? Did you get it working?
I was thinking of getting a Hybrid 2.
I remembered this issue of yours and tested an old RC split with 2.6. It worked fine. Would start/stop recording and change modes etc.

Pairan, 2021-02-05:
I closed this issue due to two reaseons :)

1) nobody seems to have the problem (or care)
2) I found to have the same issue later on iNav 2.5.2 which was fine before.

So I can't go here and claim this to be a iNav v2.6 issue ... It's more likely a issue with my camera that I haven't resolved so far!

I would assume that the most readers of the issue would had the same FIRST idea about it and then simply went ;)

So ... I can't claim that the RunCam Hybrid has a problem with iNav (v2.6 or other) at this point ;) ...

Jetrell, 2021-02-06:
@Pairan thanks for the run down. 
Could it be a fault with the interconnection cable or headers, between the module and board?
I read of many having similar issues with the RC Splits, years back.

Pairan, 2021-02-09:
Didnt have the time to fix it so far- yet it p#### me off plenty :D I assume it to be the wiring somehow so .. yeah, that could be a problem