iNavFlight/inav #6381: During the FS the wing turned and flew away in an unknown direction.

MikroDarthVader, 2020-12-05:
During the FS wing flew in my direction at an altitude of 70m, but at a distance of 750m from me suddenly turned left and went down. Perhaps the gps signal is missing - this has already happened once during this flight. I went into the manual mode and started to correct the course, but the wing got into the radio shadow and I lost the whole signal. I found some eyewitnesses of my wing who claimed that it flew further on the course at which the loss of communication occurred. I don't know if this can be connected somehow, but this course approximately coincides with the direction of my actual home and the place of my first flights where I set up the inav.

What could have happened to the inav? I searched all the surroundings within a kilometer, but I did not find my wing.

digitalentity, 2020-12-05:
On GPS loss during RTH INAV will fly as it was flying for `nav_position_timeout` seconds (default 5 seconds). If GPS will not recover, INAV will enter an emergency landing procedure and set `failsafe_throttle`, `failsafe_fw_roll_angle` and `failsafe_fw_pitch_angle` (by default 0% throttle, 20 deg left bank, 10 deg dive).

By your description it seems either an emergency landing due to GPS loss or sensor failure.

You can set `nav_position_timeout` to allow INAV to maintain course indefinitely in case of GPS loss (which also could be risky)

MikroDarthVader, 2020-12-05:
Сould the wing due to something after the transition of FS->manual->FS, for example, lose its home point and fly away at the current heading?

MikroDarthVader, 2020-12-05:
I'm using matek f405 wing fc, if it's important

digitalentity, 2020-12-05:
Failsafe without GPS lock will always end up spiraling down, it will not lose it's home point. I don't think it's possible for a wing to end up flying forward in failsafe, but there might be bugs lurking somewhere. Do you have at least goggles recording of the event?

MikroDarthVader, 2020-12-05:
Unfortunately no, I did not make a record.

0crap, 2020-12-07:
@MikroDarthVader At least give us the "diff all" and iNav version you used.

MikroDarthVader, 2020-12-08:
Unfortunately, I did not have a diff all, but I found a diff
`
# version
# INAV/MATEKF405SE 2.5.1 Jun 16 2020 / 12:50:26 (098a574b3)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 4 0 50 0 -1
smix 1 4 1 50 0 -1
smix 2 3 0 -50 0 -1
smix 3 3 1 50 0 -1

# servo
servo 3 1000 2000 1516 100
servo 4 1000 2000 1562 -100

# logic
logic 0 1 -1 3 1 8 0 1190 0
logic 1 1 -1 7 4 4 4 5 0
logic 2 1 -1 7 4 6 4 7 0
logic 3 1 -1 2 1 8 0 1810 0
logic 4 1 -1 2 1 8 0 1210 0
logic 5 1 -1 3 1 8 0 1490 0
logic 6 1 -1 2 1 8 0 1510 0
logic 7 1 -1 3 1 8 0 1790 0

# gvar

# gf
gf 0 1 0 3 0 1 0
gf 1 1 1 3 0 2 0
gf 2 1 2 3 0 3 0
gf 3 1 3 3 0 4 0

# feature
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH

# beeper

# map
map TAER

# serial
serial 2 2048 115200 115200 0 115200
serial 3 2 115200 115200 0 115200
serial 30 0 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1500 2100
aux 1 1 1 1300 1700
aux 2 3 1 1700 2100
aux 3 11 2 1700 2100
aux 4 45 2 1300 1700
aux 5 12 6 1300 1700
aux 6 37 5 1300 1700
aux 7 21 5 1700 2100
aux 8 13 7 1500 2100
aux 9 27 6 1700 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 24 1 V
osd_layout 0 1 2 4 V
osd_layout 0 2 0 0 V
osd_layout 0 9 1 2 H
osd_layout 0 11 3 5 V
osd_layout 0 12 23 5 V
osd_layout 0 13 23 8 H
osd_layout 0 14 23 2 V
osd_layout 0 15 3 8 V
osd_layout 0 19 3 6 V
osd_layout 0 23 17 3 V
osd_layout 0 24 13 4 V
osd_layout 0 26 3 9 V
osd_layout 0 28 2 1 V
osd_layout 0 32 13 1 V
osd_layout 0 33 23 10 V
osd_layout 0 35 2 12 V
osd_layout 0 36 24 4 V
osd_layout 0 39 2 11 V
osd_layout 0 40 23 6 V
osd_layout 0 50 9 3 V
osd_layout 0 85 23 8 V
osd_layout 0 97 1 14 V
osd_layout 0 105 2 2 V

# master
set looptime = 2000
set acc_hardware = MPU6000
set acczero_x = 52
set acczero_y = -19
set acczero_z = -348
set accgain_x = 4087
set accgain_y = 4103
set accgain_z = 4017
set mag_hardware = NONE
set magzero_x = 58
set magzero_y = 309
set magzero_z = -185
set baro_hardware = BMP280
set pitot_hardware = NONE
set motor_pwm_rate = 4000
set motor_pwm_protocol = DSHOT150
set failsafe_delay = 3
set failsafe_procedure = RTH
set align_board_roll = -84
set align_board_yaw = 900
set current_meter_scale = 275
set model_preview_type = 8
set servo_pwm_rate = 60
set small_angle = 180
set applied_defaults = 3
set gps_sbas_mode = AUTO
set nav_rth_allow_landing = NEVER
set nav_rth_alt_mode = AT_LEAST_LINEAR_DESCENT
set nav_rth_altitude = 7000
set nav_fw_max_thr = 1650
set nav_fw_loiter_radius = 4000
set nav_fw_launch_velocity = 150
set nav_fw_launch_max_angle = 180
set nav_fw_launch_thr = 1730
set nav_fw_launch_idle_thr = 1365
set nav_fw_launch_motor_delay = 0
set nav_fw_launch_spinup_time = 150
set nav_fw_launch_min_time = 300
set nav_fw_launch_timeout = 10000
set nav_fw_launch_climb_angle = 25
set osd_time_alarm = 20
set osd_alt_alarm = 1500
set osd_dist_alarm = 6000
set osd_neg_alt_alarm = 2
set osd_current_alarm = 30
set osd_crosshairs_style = AIRCRAFT
set osd_left_sidebar_scroll = ALTITUDE
set osd_right_sidebar_scroll = GROUND_SPEED
set osd_sidebar_scroll_arrows = ON
set i2c_speed = 800KHZ
set name = Reptile s800 v2
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 4

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 24
set fw_ff_pitch = 110
set fw_p_roll = 6
set fw_i_roll = 10
set fw_ff_roll = 45
set fw_p_yaw = 8
set rc_yaw_expo = 70
set roll_rate = 40
set pitch_rate = 25
set yaw_rate = 10

# battery_profile
battery_profile 1

# end the command batch
batch end
`

closed 2020-12-11 by DzikuVx