iNavFlight/inav #6382: WINGFC doesn't boot in master (likely H7 regression)

alteman, 2020-12-06:
## Current Behavior
WINGFC target doesn't boot after commit e72e70209 (according to git bisect)

## Steps to Reproduce
1. git clone 
2. ./build.sh WINGFC
3. Flash the firmware
4. Reboot the board

## Expected behavior
The board boots, USB is detected, LEDs flash

## Suggested solution(s)
USB not detected, LEDs don't flash

- FC Board name and vendor:
Wing FC 10DOF (Furious FPV F-35 Lightning clone)
- INAV version string:
e72e70209

ATTN @digitalentity

stronnag, 2020-12-06:
This is not entirely unexpected; #5827 states "Broken stuff and missing features will be addressed in future PRs". H7 support was merged early to avoid complex future merges, not because everything was working.

I have two F405 boards, one of which boots, one doesn't. The one that boots passes the "hover in the garden" test though I'm not convinced I'd fly it further that than.

As you noted, reverting the iniital H7 commit facilitates booting again.

digitalentity, 2020-12-06:
My debug rig with `OMNIBUSF4PRO` doesn't boot either. Looking into this

digitalentity, 2020-12-06:
Hardfaulting in `readEEPROM()`. Weird.

digitalentity, 2020-12-06:
Ok, not `readEEPROM()` but `findEEPROM()`. Now let's see why it's reading beyond valid flash area.

digitalentity, 2020-12-06:
This is unrelated to H7 code itself, but to the way we read config and was merely revealed by a H7 debugging leftover code.

https://github.com/iNavFlight/inav/blob/master/src/main/config/config_eeprom.c#L155-L158:

```
        if (record->size == 0
            || p + record->size >= &__config_end
            || record->size < sizeof(*record))
            break;
```

Here we check that `p + record->size` doesn't span beyond config memory area, but what happens if `record->size` is beyond the end of config memory area? Nothing harmful, unless the config memory area is the last page of the flash and there is nothing beyond that. In that case reading `record->size` will cause a hardfault, which is exactly what happened here.

In pre-H7 code we did a call to `ensureEEPROMContainsValidData()` to make sure EEPROM makes sense, but this was accidentally left out when debugging config read/write on H7. This caused `readEEPROM()` to be executed on a potentially invalid config memory contents which revealed this bug.

Interestingly, `ensureEEPROMContainsValidData()` may suffer from the same bug as it also doesn't check that `record->size` is within valid config memory area.

Note that F7, H7 and some beefy F4 chips don't suffer from this bug as there is valid memory beyond `__config_end`.

https://github.com/iNavFlight/inav/pull/6384 fixes this

closed 2020-12-06 by digitalentity

alteman, 2020-12-06:
Thanks, working now!

PR #6384 (2020-12-06): Fix EEPROM loading code
Fixes potential hardfault when `findEEPROM()` attempts to read beyond `__config_end`. Restores a missing check to `ensureEEPROMContainsValidData()` to do a pre-check of EEPROM before attempting to load it.

Fixes #6382