iNavFlight/inav #6392: INAV, airplane, Alt Hold, Pos Hold, RTH problem, goes into spin.

KaptenAndreas, 2020-12-07:
****

## Current Behavior

## Steps to Reproduce

1. See decription above.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

avsaase, 2020-12-07:
I have had this exact same problem, with the same airplane so I think I can help. I will assume you have correctly setup the rates, also for yaw (I know, hard to measure correctly) and that auto tune has given you reasonable P, I and FF gains. Also check that your rudder throws in manual mode are sensible, if they are too large it is best to reduce them mechanically. Alternatively you can reduce the yaw FF gain. If these things are not setup correctly you should fix those things first. There are some great guides on https://inavfixedwinggroup.com/.

Having said that, the problem is that the rudder on this airplane is incredibly effective, and INAV is overusing the rudder when making turns in the navigation modes. On this airplane rudder and ailerons are very much cross-controlled, but INAV doesn't understand this. So when INAV tries to use rudder to create a yaw rotation to make a nice coordinated turn, it is actually adding a lot of roll. This causes the plane to tip into a spin and also explains why switching to any non-navigation mode immediately recovers from the spin; these modes do not automatically apply rudder. You can reduce the amount of rudder used for coordinated turns by decreasing `fw_turn_assist_yaw_gain` from the default of 1. If you set it to zero it will not use any rudder, but then the rudder might actually work the other way to fight the turn (something I am trying to solve in #6387). 

TL;DR: if everything else is **really** setup correctly, reduce `fw_turn_assist_yaw_gain`.

KaptenAndreas, 2020-12-07:
> I have had this exact same problem, with the same airplane so I think I can help. I will assume you have correctly setup the rates, also for yaw (I know, hard to measure correctly) and that auto [...]

Thanks for the tips I will double check my rudder travel motion and if necessary maybe reduce them a little. And then test to change `fw_turn_assist_yaw_gain`.

KaptenAndreas, 2020-12-08:
> I have had this exact same problem, with the same airplane so I think I can help. I will assume you have correctly setup the rates, also for yaw (I know, hard to measure correctly) and that auto [...]

Hi again!

One more question to your answer.
Should I set the correct rates for the rudders in Manual Mode or in Angle/Horizon mode?
Because there is a big difference between them.

/KaptenAndreas

closed 2020-12-11 by DzikuVx