iNavFlight/inav #6399: Noxe V1-INAV/FLYWOOF411 2.6.0

Finnigan1971, 2020-12-11:
Noxe V1, # INAV/FLYWOOF411 2.6.0 Dec  4 2020 / 11:03:18 (cada68e4)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

With Dshot (all three) motor 3 don't spin (even in the configurator), with Oneshot (both versions) the four motors spin weird.

vaicekas, 2021-01-04:
I want to confirm that the motor No. 3 is operating abnormally. 
In the configurator, all four motors spinning only when the master slider is moved. 
With slider motor no. 3 - the motor does spinning, but ESC beeps like a restarting.
When the quadcopter armed on the ground all engines spinning, but switching to hold altitude mode- shuts down the third engine.

rickgode, 2021-01-07:
I "think" have this same issue.  On a stock flywoo explorer with the flywoof411 board only motor 1 spins.  When I move the master slider only motor 1 spins the other 3 do nothing.  I've tried dshot 150,300/multishot/oneshot.  I only get anything to work with dshot, but again, only one motor spins.

I have verified that all 4 motors work as intended with the betaflight version it came with.

Ricky

stronnag, 2021-01-12:
Following some discussion on RCG, @rickgode was able to get information from the vendor as to the V2 changes. They seem to have gone all "Omnibus" on us, with a V2 that has only superficial resemblance to V1.

Attached (Zip file) 
[inav_FLYWOOF411_V2.hex.zip](https://github.com/iNavFlight/inav/files/5800286/inav_FLYWOOF411_V2.hex.zip)
2.6.0 and 2.7.0 V2 hex files, bench tested by @rickgode as working. 

PR to follow RSN.

stronnag, 2021-01-13:
Fixed by #6495

closed 2021-01-13 by stronnag

s2kaka, 2021-04-20:
can i use this FC for a Wing?