iNavFlight/inav #6402: TURN ASSIST logic error?

avsaase, 2020-12-11:
Sorry for opening an issue about this, but I think I found a mistake in the TURN ASSIST math:
https://github.com/iNavFlight/inav/blob/ab051b18c7024aeb33d902eb48db7d34f8d426b2/src/main/flight/pid.c#L877

If i understand correctly here it says that the rotation around the vertical axis in the plane reference frame is equal to the rotation around the vertical axis in the earth reference frame. But if the bank angle is non-zero then this is not correct.  Adding `RADIANS_TO_DEGREES(coordinatedTurnRateEarthFrame)` to the yaw target rate then gives a too high target rate. 

I have to admit that I do not fully understand the quaternion math so if this is already accounted for in `imuTransformVectorEarthToBody(&targetRates)` then please close the issue.

digitalentity, 2020-12-12:
`targetRates.z` is still in the Earth frame of reference. Before being added back to PID yaw rate target the `taretRates` vector is rotated back to the UAV frame of reference here: https://github.com/iNavFlight/inav/blob/ab051b18c7024aeb33d902eb48db7d34f8d426b2/src/main/flight/pid.c#L889

closed 2020-12-12 by digitalentity

avsaase, 2020-12-12:
Okay thanks, that was the part I missed.