iNavFlight/inav #6412: [2.6] GPS - Ground Assistance Type

0crap, 2020-12-14:
When I set GPS Ground Assistance Type to "Disabled" it seems the GPS still reports SBAS as Enabled and on Auto-Scan.
I would expect Disabled.
[image]

When I set GPS Ground Assistance Type to EGNOS it is not using the correct sat numbers if I'm not mistaken.
According to the EGNOS website it should be PRN 123,136.
[image]
[image]

I tested this by using the configurator to change settings and used the CLI "gpspassthrough" command to connect ublox u-center to read back the settings from my GPS. (Beitian BN-220)

Hope the devs can have a look to see if this is correct behaviour, or need some improvement.
Thx!

digitalentity, 2020-12-14:
Looks like UBX8 and above have deprecated the `enabled` field in the `UBX-CFG-SBAS` message. We'll have to implement `UBX-CFG-GNSS` message to enable/disable SBAS properly on Ublox8 and Ublox9 receivers.

0crap, 2020-12-14:
Thx for having a look.

One more screenshot, in case you need it, to show the complete GPS configuration I normally set in the GUI.
(Ground Assistance Type set to European EGNOS is of course the one I play with, because I live in the EU.)
No other changes done on CLI.

[image]

0crap, 2020-12-15:
For the fun of it I also checked BF on this issue.
They seem to have it working correctly.
When disabled it does read back from the GPS as SBAS Subsystem Disabled.
When enabled with EGNOS, I read back the correct sats.

[image]

stronnag, 2020-12-17:
* If Galileo is configured, we use  `UBX-CFG-GNSS` to enable 3 SBAS channels.
* The UBLOX M8N documentation states (or at stated when we implemented this)  (5.1 SBAS (Satellite Based Augmentation Systems)) we should use 120,124,126. We set this mask (+131) (0x851).
*  we  would need to change the `UBX-CFG-SBAS` mask if the PRNs have indeed changed.

stronnag, 2020-12-17:
ESA announcement https://egnos-user-support.essp-sas.eu/new_egnos_ops/sites/default/files/library/official_docs/service_notice_15.pdf

So we're only 3 years late ... let me try implementing that.

0crap, 2020-12-17:
> ESA announcement https://egnos-user-support.essp-sas.eu/new_egnos_ops/sites/default/files/library/official_docs/service_notice_15.pdf
> 
> So we're only 3 years late ... let me try implementing that.

After 3 years that pdf seems outdated as well. :-)
Here you can find the current operational sats.

https://egnos-user-support.essp-sas.eu/new_egnos_ops/egnos_system_realtime

That is: 123,136 and 126.

stronnag, 2020-12-17:
OK, now (hack, hack) if I enable Galileo, I see :
[image]

As per the latest ESA informaton

stronnag, 2020-12-17:
However as @digitalentity noted, we still need UBX-CFG-GNSS for the non-Galileo case, otherwise we won't have SBAS on other augmentation systems.

0crap, 2020-12-17:
> OK, now (hack, hack) if I enable Galileo, I see :

I guess you mean "if I enable Ground Assistance Type is EGNOS".
Galileo is something else, that is the EU counterpart of GPS from the USA.

stronnag, 2020-12-17:
I know well what Galileo is (I happened to have implemented in inav). We only enable SBAS now when Galileo is set (using `UBX-CFG-GNSS`). The PR will use `UBX-CFG-GNSS` to correctly enable SBAS as required regardless of SBAS.

0crap, 2020-12-17:
Ahh, thx, I see. I have Galileo always enabled indeed.
I didn't test Ground Assistance Type with Galileo switched off.
(It really helps, often 20 sats in flight.)

stronnag, 2020-12-17:
The whole of our `UBX-CFG-GNSS` needs cleaning up, it does stuff that it shouldn't necessarily do (like set SBAS unconditionally if Galileo is enabled). So a revisit is long overdue.

0crap, 2020-12-17:
Thx for helping out!
SBAS switched off will help speeding up initial fix (I hope), great if it can be decoupled from the Galileo setting.

stronnag, 2020-12-17:
It will be decoupled. And part of #6431 will be to also verify that we current PRNs for the other systems.

commit e90c76ae (2020-12-18): update EGNOS PRN mask to latest EGSA definition, fix SBAS for non-Galileo usage. (#6431)

[GPS] update SBAS and Galileo

* Update GNSS / SBAS handling
* Improve readabily of SBAS scanmask1 / PRN setting
* Use current PRNs for all SBAS options
* Decouple Galileo dependency for UBX-CFG-GNSS
* Only set required GNSS options
* Set GNSS options dynamically
* Replace magic numbers with #define where appropriate

Fixes #6412
closed 2020-12-18 by stronnag