iNavFlight/inav #6413: Altitude switches between home and gps when crsf and mavlink telemetry run simultaneously

iblackymb, 2020-12-14:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

https://photos.app.goo.gl/3AFYEz9vywPS42WZ8
Same behavior is seen sending the telemetry through 
the internal bluetooth of the crossfire module, so i think that a opentx bug is not probable.

The problem disappears using Sbus instead on crsf as serial rx protocol.

Same behavior on 3 planes, current and  default settings tested.

Steps to Reproduce

1. Crsf serial rx protocol 
2. Mavlink telemetry enabled
3. INAV 2.5.2 
4. 

## Expected behavior

Stable alt data through telemetry to be used with an antenna tracker.

## Suggested solution(s)

Maybe giving the option to disable crsf telemetry output or be able to choose between home and gps Altitude.

## Additional context

---

- FC Board name and vendor: matek f765 and f405
- INAV version string: 2.5.2

OptimusTi, 2020-12-14:
Have you tried to delete and rediscover the telemetry sensors? I can't reproduce the problem here.

iblackymb, 2020-12-14:
Hi Optimus.

Yes deleted them several time, tried with a new opentx model too.

iblackymb, 2020-12-15:
Just to add some more data.

I just set up a new f765 with inav 2.6, only with the essential settings to make work the gps and rc, and I'm getting the same behavior.

iblackymb, 2020-12-18:
> Have you tried to delete and rediscover the telemetry sensors? I can't reproduce the problem here.

Hi Optimus

Can you tell me your setup? I dont know what else to try, the only way I make it work together is disconnecting the CRSF RX cable.

iblackymb, 2020-12-26:
Hi,

I just tried to disabling crsf, using sbus to send the rc channels and receive the telemetry by mavlink. Just realized that testing each one separately, crsf telemetry send the relative altitude, and mavlink send the gps telemetry. This is why I have this variances when both telemetry runs simultaneously.

Problem is that I need the mavlink telemetry to use my tracker, and now with inav I found useful the crsf state in the osd info.

HazzaHFPV, 2021-04-07:
I actually have the exact same problem. Crossfire Nano Rx on latest Firmware (4.11) with an F4 flight Controller with Baro onboard and a BN880 GPS including a Compass.
Only Crossfire connected: Height above Ground received via telemetry.
Mavlink Telemetry connected: Height above sea-level shown.
Both connected: value jumps between the two as in @iblackymb video.

So far I was not able to pinpoint the problem or solve it.

Maxeemo, 2022-03-19:
Hi, I have the exact same problem, have you solved in some way?

b14ckyy, 2024-03-20:
if both use the same sensor ID, there is nothing we can do about it. Disable one of the sensors with the altitude you don't want in the sensors list of OTX/ETX.

closed 2024-03-20 by b14ckyy