iNavFlight/inav #6423: Blackbox log improvements for autonomous modes

avsaase, 2020-12-15:
I have a couple suggestions for improving the blackbox logging that will help troubleshoot issues with the autonomous modes.

## Current Behavior
- `setpointRate[roll/pitch/yaw]` doesn't actually log the setpoint rates for the PID/PIFF controller, but instead mirrors `rcCommand[roll/pitch/yaw]`. Is this a bug? I think `axisRate[0/1/2]` does measure the setpoint rates, but I'm not sure. 
- Target attitude is not not logged. In the autonomous modes `rcCommand[roll/pitch/yaw]` gives an indication of what the autopilot is trying to do but the displayed units are confusing.
- There are variables for attitude, `attitude[0/1/2]`, but I don't see how I can get pitch and roll angles from these.

## Suggested Solution
- Improve the labeling of the target rates variables
- A new variable that logs the target target pitch and roll angles. I think target attitude can be calculated as:
```
targetAttitude[roll] = pidRcCommandToAngle(rcCommand[FD_ROLL], pidProfile()->max_angle_inclination[FD_ROLL])
targetAttitude[pitch] = pidRcCommandToAngle(rcCommand[FD_PITCH], pidProfile()->max_angle_inclination[FD_PITCH])
```
Obviously there is no target attitude in ACRO. 

- Log the actual attitude as pitch and roll angles. This can also help with diagnosing horizon drift issues. 

## Who does this impact? Who is this for?
Everyone analyzing blackbox logs.

## Additional context
The Blackbox Explorer probably needs to be updated to accommodate these changes.

breadoven, 2020-12-15:
You're right about the target attitude log values. As for the actual attitude values paste the following into flightlog_fields_presenter.js in the Blackbox Explorer "js" folder:

Around line 300 perhaps after the entry for BaroAlt (doesn't matter where exactly since it's a select case).

	          case 'attitude[0]':
		            return (value / 10).toFixed(1) + " degs";

	          case 'attitude[1]':
		            return (value / 10).toFixed(1) + " degs";

	          case 'attitude[2]':
		            return (value / 10).toFixed(1) + " degs";

You can do the same for other fields that don't return a user "Friendly" output (assuming you can work out the units).

avsaase, 2020-12-15:
Thanks, this works like a charm. To be sure, negative pitch value indicate pitching up, right?