iNavFlight/inav #6424: F722-miniSE DPS310 barometer readings

cjmv, 2020-12-15:
## Current Behavior
Firmware: INAVMATEKF722MINI 2.6

Very strange altitudes readings from the flight controller

- Before arming: Alt = -1
- After arming, it jumped to -8, -9.
- As I was gaining altitude the value actually got lower, for instance, -11, -15.

While not armed and connected to INAV, the altitude value drops to 0 when stable at any altitude. But I'be been told by Matek that this is quite normal.

Furthermore I've noticed that when I look into the INAV sensor values "baro" and "Alt" is like somehow these 2 values are switched. The Baro value in INAV is really stable and decreases and increases as height gets lower or higher (respectively) and... that can't be barometric pressure, since pressure actually decreases as you get higher and not the opposite. So unless that Baro value in INAV is supposed to represent something else, my best guess is that at least it is switched with the Alt value.

I've installed your BF firmware version 4.2.3 (custom firmware provided by Matek since BF firmware 4.3 is not available yet), and just by looking at the sensor it seems that the _baro_ is working perfectly fine (without arming the motors).

## Steps to Reproduce

1. Flash INAVMATEKF722MINI 2.6 on a F722-miniSE
2. Configure and arm the quad-copter,
3. Check telemetry Alt values variation as altitude gets higher or lower.
4.

## Expected behavior
Altitude to steadily increase and decrease as quad gets higher or lower.

## Suggested solution(s)
N/A

## Additional context

[PasteBin](https://pastebin.com/8F2LHXis)

---

- FC Board name and vendor: F722-miniSE DPS310
- INAV version string: INAVMATEKF722MINI 2.6

Jetrell, 2020-12-15:
@cjmv I'm using the same hardware and firmware you are. 
As you have noted. Mine too has some drift on the bench. Which doesn't seem any worse to me than my BMP280 sensors.

While its in the air, sensor fusion between Baro and GPS altitudes, works well in my application. 
It holds altitude precisely in `Altitude_hold` and `Position_hold` flight modes and I haven't seen any altitude drift to speak of while looking back through my DVR footage and logs.

cjmv, 2020-12-15:
Hi @Jetrell,
Thanks for you comment.
"edit: I also have a piece of high density foam over the sensor."

I'll try the high density foam. Maybe that will do the trick!

closed 2020-12-16 by DzikuVx