iNavFlight/inav #6425: Pilot Heading Arrow

buzkill420, 2020-12-16:
## Current Behavior
Currently inav only has ability to place HomePoint and aircraft heading on osd display.

## Desired behavior

Having the aircraft heading and "direction to home" arrow on the osd is a necessary for any long range flight. Flying my DJI, i also find it very handy having the direction I'm facing (the pilot)  onscreen as well, allowing you to position yourself for optimal reception..ie goggles and transmitter.
Was thinking how inav might be able to accomplish the same thing.

## Suggested Solution

Using a headtracker (I'm using the internal fatshark module) you should be able to use the gyroscope to track your position by feeding the correct axis through on a channel to your fc. If only using the internal module, it lacks a compass, so auto aligning is out (and tbh idk if the compass data is even output....far over my head) so a "heading set" mode would have to be created- just like headless mode and headajust.

## Who does this impact? Who is this for?
Long range pilots or anyone using directional antennas

What do you think? Worth wild or just a waste of time? Im curious what others think of the idea, good or bad.  
 
Happy flying!

ABLomas, 2020-12-16:
While i understand what you want and think that this is very usefull thing - i don't think this is doable in INAV, at least for now.
I also opened similar issue on OpenHD tracker - but OpenHD OSD is drawn on ground side, not air, while INAV does this on air side.
So, several things here:
 - good, no drift 3D position tracker on ground (very good headtracker OR commercial sensor fusion with mag/gyro/acc (starting from 35$ on ali) OR some new code with smth like MPU9150 or so). Hard to do with long-term drift but doable
 - ability to get aircraft position relative to yours. Requires telemetry (and here OpenHD already wins - as everything is integrated already)
 - ability to draw additional arrow on OSD (possible with cheap FC already or separate OSD chip) - again, OpenHD wins here as it draws OSD on ground side by design.

I think #1 and #3 can be done with some FC which has mag (old NAZE32, spracingf3 -both have no integrated OSD - i don't know any up-to-date FC with mag onboard and OSD) onboard, then only telemetry and code remains (and if 3D orientation code can be taken from current INAV - then drawing arrow is relatively simple).

buzkill420, 2020-12-16:
> While i understand what you want and think that this is very usefull thing - i don't think this is doable in INAV, at least for now.
> I also opened similar issue on OpenHD tracker - but OpenHD OSD is [...]

Thanks for the reply!  As i said, implementation is quite over my head,  but the thought was there and iwas hoping to hear exactly your reply.  Gives me some keythings to look into and learn. 

As for the few things i did understand,

From what I've built,  and have learned,  most of my boards have osd built in (not sure why you'd need a second one as you suggested,  but curious)  and the recommended matek f405 wing also has an integrated compass (also not sure why you'd need that on drone side as well) 

My basic thought would be that given the fight controller (inav) already has the home point coordinates and can draw it onscreen relative to drone, as well as drones heading (given drone has compass and/or gps- as wings don't need compass for inav to calculate heading.)
All that would be needed to be implemented is the ability to 1) connect a reliable motion tracker(as you mentioned)  and 2) somehow map a channel to that along with ability to calibrate. (Like headless mode)

Assuming the internal fatshark head tracker was good enough (I've used it on a few of my drones and haven't found drift an issue), making a channel from that and getting it to fc is trivial.

As for needing telemetry from ground station,  not sure why you would need that either,  though most people,  myself included, flying long range or use a gps definitely  have and use a telemetry down-link.

buzkill420, 2020-12-16:
also another note,  i wouldn't think another arrow would be drawn,  it would be better(though idk about easier)  to  just "rotate" the map.

ABLomas, 2020-12-16:
Oh, so your idea is to send "ground heading" using spare RC channel to craft and display that as arrow?
Should be easier than just draw arrow on ground side, but restricted to heading only (no altitude, no other params).
Interesting idea ;-)

buzkill420, 2020-12-16:
> Oh, so your idea is to send "ground heading" using spare RC channel to craft and display that as arrow?
> Should be easier than just draw arrow on ground side, but restricted to heading only (no altitude, no other params).
> Interesting idea ;-)

Yup! Exactly! I don't have a ground station (yet!😄) and no other params needed really... K.I.S.S method! I'm not interested in my elevation and I'm definitely not moving far after goggles go on.

Jetrell, 2020-12-16:
At present in 2.6, it has `Azimuth `. Which gives you the cardinal degrees of your models location with respect to the home location, The only thing you need to know, is which direction is north, at were your standing. Then make sure you face north.
You can then tell whether the model is 90° to your right East. 180° behind you South etc.

buzkill420, 2020-12-16:
> At present in 2.6, it has `Azimuth `. Which gives you the cardinal degrees of your models location with respect to the home location, The only thing you need to know, is which direction is north, at [...]

Thanks for the tip on "azimuth" but your idea sounds an awfully like "just remembering what direction you're facing" which is helpful for sure,  and obviously something you should never forget when flying, just isn't QUITE what I'm trying to get at.

ABLomas, 2020-12-20:
Very interesting implementation idea. 
I do not care about integrated headtracker (fits only one type of goggles, can't be easily mounted on antenna, etc), so i ordered few sensors.
While they are on slow mail i did some experiments with MPU9150 and arduino - possible to get PWM out on one direction (with many bugs but this can be fixed i'm sure). Sensor is tiny and cheap and can be mounted directly on directional antenna reflector (antenna direction is what matters, not direction where you are looking).
I think i can interface it with TX using trainer port.

Now bigger problem - need to calculate angle from between home heading and airplane position. This is not so hard and will try to do, but displaying numeric stuff on OSD is beyond my skills at this moment.

closed 2022-02-03 by DzikuVx

MrD-RC, 2022-02-03:
For this to work, you would need a compass on your transmitter and/or goggles. Then send that to iNav, probably via an RC channel. Without the compass, there’s no real way of knowing which way you’re facing. The head tracker wouldn’t really be a good solution, as they can drift.