iNavFlight/inav #6426: Specktrum Radio issues with esc calibrations

SkyChief101, 2020-12-16:
I cant get the motor to work properly on INAV airplane.  I have tried ESC calibrations over and over.  I have changed ESC. I have reloaded the software.  I have uploaded 2.6.0 R3.  However when I uploaded the BetaFlight configurator everything worked as it should.  The ESC calibrated, the motor started properly and ran smoothly.  What is going on?  If I can do this in Betaflight I should not have a problem in INAV.  I believe there is a problem in the firmware can someone look into this.   I have a Speedbee F4 AIO connected to an Orange RX connected to a Spektrum DX8 second gen.  I have set up INAV in past and have not had this problem.  Here is the problem no matter how many time i calibrate the ESC or switch the protocol the motor only engages after about half throttle until then it beeps.  the tabs in the radio of INAV all move correctly, all the servos work fine.

SkyChief101, 2020-12-18:
Switched over to 2.4.0 motor calibrated perfectly, went back to 2.6.0 nothing and least nothing good enough to fly with.  I would stay with 2.4.0 however it does not have the benefits of so many new changes that 2.6.0 has especially new nav flight modes.  Can someone help.  Thanks  Also when I switched back everything worked on Standard, when I moved over Standard did not work.  On the ESC.

digitalentity, 2020-12-19:
How do you calibrate? Please describe the process or record a video

closed 2020-12-19 by digitalentity