iNavFlight/inav #6427: Flight timer doesn't start until auto launch sequence ends

avsaase, 2020-12-16:
## Current Behavior
Basically the title. I have launch mode permanently enabled and use the the combined on/flight timer. When arming the fc, the timer switches from the on timer to the flight timer and resets to zero, but it doesn't start counting until the launch sequence ends.

## Steps to Reproduce

1. Launch plane with auto launch
2. Flight timer doesn't start until launch sequence ends

## Expected behavior
Flight timer starts when the fc is armed or even better, when the launch is detected.

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: Matek F405-Wing
- INAV version string: # INAV/MATEKF405SE 2.6.0 Dec 15 2020 / 17:15:38 (8eb39862)

breadoven, 2020-12-16:
I noticed this. The flight timer only runs on launch detection and launchfinishedoraborted states which means it only really runs continuously after the latter since Launch detection lasts 1 loop ? Needs all the launch states between detection and Launch finish to better account for the flight time during launch.

breadoven, 2020-12-17:
Actually looking at this again it's a bug introduced by the revamped Launch code. isFixedWingLaunchDetected previously used a boolean value which kept its state once set true on launch detection. The revamped current launch code determines isFixedWingLaunchDetected based on the current launch state and sets it true on `return fwLaunch.currentState == FW_LAUNCH_STATE_DETECTED;` which means it doesn't hold true once the launch detection state is no longer current. I think changing the "==" to ">=" should fix it without breaking anything else.

avsaase, 2020-12-17:
Thanks, this solves the problem with the flight timer. I don't have the RTH estimator setup, so I don't know if this bug broke that feature and if this fixes it again. I can make a PR, but maybe someone who's more familiar with the launch code should have a look first so that this doesn't introduce new bugs. [DVR](https://i.imgur.com/LhziDJp.mp4)

breadoven, 2020-12-17:
Actually the change above probably (haven't tested it) doesn't fix the RTH estimator, not beyond the end of launch anyway, that needs another addition. Needs a bit of thought to work out the best fix.

closed 2021-04-12 by avsaase