iNavFlight/inav #6438: Improving autoland after failsafe-RTH for fixedwing aircraft

Flight-instructor, 2020-12-22:
## Current Behavior

## Desired Behavior

The airfield has a direction that normally does not change ( as noby turns it during the night )
Normally, we do not know the exact direction.
And also normally, we try more or less taking of against the wind, so pending on it, either in the one, or the other direction.

Wanted parameters:
Aircraft learns, in what direction and out of what altitude and position a straight landing shall be performed ( or at least a straight approach.

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

closed 2022-02-03 by DzikuVx

MoriXTV, 2023-12-12:
Did this get Implemented yet? Just came across it and I must say the Idea is really cool.

Maybe a simpler implementation with just two points like P1: "land here" and P2: "approach from here" that can be set by walking there with the model and hitting a switch would do.
From these two GPS points and the already existing vertical landing velocity, cruise speed and RTH altitude the approach vector and approach point elevation should be easy enough to calculate.

Then if the landing mode is activated (after RTH or via switch) the pane could just fly to the calculated approach point at RTH altitude and begin to descend.

It would be really nice to have this and not having to worry about landing especially for something like the AR-Pro with 4s6p  18650 because we all know that landings with small and heavy planes are either fast and tricky or short and expensive ;)

Love your work guys keep that up.

MrD-RC, 2023-12-12:
A much improved autoland is currently scheduled for 7.1.