iNavFlight/inav #6439: Save trims to iNav subtrims

MrD-RC, 2020-12-22:
## Overview
This is really a new feature for LOS plane pilots. Coming from LOS, servo autotrim is pretty unintuitive, especially if you're not using iNav via FPV. These pilots have been trimming their models manually using the transmitter, and are comfortable doing so. The problem is, other than fudges with servo autotrim, there is no way to get these trims stored in to iNav.

### Desired Behaviour
The pilot launches their model, flys in manual, and trims their plane with the transmitter trims. The would then land, perform the _trims to iNav subtrims_ task, and finally zero their transmitter trims.

## Suggested Solution
I am suggesting a new _trims to iNav subtrims_ function that can be accessed from the OSD menu or via a stick command. The actual function its self could be based on servo autotrim. It would be much simpler though. As this is being performed disarmed and on the ground, it would just need to instantly grab the stick positions and save them.

## Who does this impact? Who is this for?
This would be useful for all plane pilots who prefer trimming manually. This could also make it simpler for more traditional pilots to use iNav.

## Additional Caveats
This would be a great feature. However, I think we should recommend a couple of things to pilots who use this

### Excessive trimming should still be performed physically with the linkages
If a lot of trimming is needed, it would be better to fix them physically. I think the trims should still be stored, but perhaps give feedback via the beeper and OSD system message if the trim is excessive. Perhaps, the system message could be added globally, as I'm sure there are instances when performing a regular servo autotrim results in excessive trim levels.

### Unlink trims
The first recommendation would be to unlink trims in all modes other than manual on the transmitter. This can be performed very simply in OpenTX. This would allow them to trim in manual, but if they did move to another flight mode, it would not see the trims as stick inputs.

### Set up an automated task for saving the trims
There are a couple of requirements for this to work.
1. They must be in manual when they perform the _trims to iNav subtrims_ feature
2. They must reset the trims on the transmitter to 0 after performing the feature
Therefore, I think it would be a good idea to recommend using transmitter logic to enter manual, call the _trims to iNav subtrims_ function, reset the trims to 0, then release manual mode. Again, this should be pretty simple in OpenTX

##Notes
There have been other requests for similar features, so I'll link them below.
https://github.com/iNavFlight/inav/issues/4655
https://github.com/iNavFlight/inav/issues/4132

Other requests that may have an interest in this
https://github.com/iNavFlight/inav/issues/5966
https://github.com/iNavFlight/inav/issues/5750
https://github.com/iNavFlight/inav/issues/5033

MrLAtt, 2020-12-22:
Sounds like a good solution 👍

avsaase, 2020-12-22:
Can't you do this already using servo auto trim?

1. Fly in manual
2. Trim using transmitter trims
3. Land
4. Do not disarm
5. Enable auto trim
6. Disarm
7. Reset transmitter trims
8. Done

I think the fact that auto trim takes two seconds to transfer the channel trim to the servo midpoints instead of doing it instantaneously is hardly a problem. The only benefit of this new feature would be that this works when disarmed. Using the osd menu for a feature that benefits LOS flyers also is a bit of a contradiction. IMO an in-flight adjustment for the servo center as two of the issues you linked asked for would make more sense, but this requires three spare channels if you want to map each of the trims to an adjustment. 

Honestly, the best way to trim your plane in INAV is to fly in acro and enable autotrim. If you want you can then switch back to manual.

MrD-RC, 2020-12-22:
The 8 steps that you mentioned are the servo autotrim fudge that I discussed in the overview. The 2 seconds may not seem like an issue, but it would be a guess, there is no feedback. Though it could be automated by the Tx. 

However, this also requires staying armed; which could go against the rules of some flying clubs. You would need to land, get you model off the strip asap, which means going in to the pits, where a disarmed state could be a club requirement on safety grounds. You have to remember that some of these people could be in a club environment, which is totally different to most FPV flying in a random field alone or with a couple of mates.

The OSD menu feature could seem like a contradiction. But I am not just talking about pilots who are using iNav LOS; but also pilots who have come from LOS and are flying iNav FPV. I don't see that a little bit of replication like that is a big deal, as it exists for other things. This may also be helpful for pilots who are coming over from Vector, seeing as EagleTree are no more. They are again used to trimming manually, then accessing an OSD menu option to save trims to subtrims.

Pilots coming from LOS, whether flying iNav LOS or FPV will be more used to manual tuning with the Tx trims. The most natural direct route, as you pointed out, would be the adjustments. I thought at one stage this would be a good idea, but 3 channels is a lot to lose. which is why I'm looking at alternatives.

I disagree about that being the _best_ way to trim. It is **a** way to trim in iNav. But, what is best is what the individual pilots prefer. It really isn't the best way for most LOS pilots. I can tell you 100% that I never use servo autotrim. I tried it, it didn't perform as well as expected, so went back to manually tuning and adjusting the linkages. I know many pilots who fell the same way and ignore that feature completely. Servo autotrim is near on useless when flying LOS, as you are changing how the model reacts, you can't just hold it level.

ShikOfTheRa, 2020-12-23:
Hey Mr D - Re “The first recommendation would be to unlink trims in all modes other than manual on the transmitter. This can be performed very simply in OpenTX”
Just bear in mind not everyone uses  OpenTX or a system that can do this.

Definitely agree that pilots converting from LOS struggle with concept of not using TX trims – and also setting them up to be centred correctly. I guess part of the issue is using a FC adds a lot of complexity that needs some research and learning. I read the manual when can’t figure it out myself 😊

I fly iNav on planes that are pure LOS and also FPV and tbh not really had any issues with autotrim. Find it really simple and easy. Autotrim was a great addition to iNav
1.	Launch / Fly
2.	Put into level mode at a steady cruise speed
3.	Enable auto trim
4.	Check in manual mode to see if trims are good, if not disable autotrim and repeat steps 2+3 
5.	Land and Disarm to save

OlivierC-FR, 2020-12-23:
The whole idea of trimming is wrong from the very beginning in INAV, there should not be any trims anyway. Put the FC level, center the linkages and servos so they have the same amount of travel each side, and that's done. Then if the aircraft does not fly level in self-leveling modes, fix the board angles. We let the damn flight controller do its job.

MrLAtt, 2020-12-23:
@OlivierC-FR 

As I understand it, the problem with that is that to get the most out of a PID or in this case a PIFF controller, it needs the response of the plane to be proportional (and known) in both directions. For this reason, to get best performance it needs to know the centre position of the servo to achieve a no angular acceleration/rotation state. 

Of course this could be done with a constantly updating centre position, much like is done with the compass/accelerometer/gyro for attitude, however this type of approach is not fool proof (as is shown with the horizon drift problem(although with this it would likely be easier)) and is at the very least another point of failure.

The other problem is manual mode. Without having any sort of trimming system it would be difficult to fly thr plane in manual mode as many people choose to do quite often (myself included), and at the very least, should always be there (and working well and trimmed) as a fallback for when something goes wrong. For this reason, I also don't think manual mode should be subject to any constantly reupdating centre position values, meaning that some sort of trimming feature will always have to be around in INAV IMHO

OlivierC-FR, 2020-12-24:
I get what you mean, but manual mode as a fallback is a myth, it never happens, if/when the FC fails, it's all or nothing, you don't get to keep the mixing for the servos. When the FC goes down, all outputs goes down. Then having the servos perfectly centered for leveled flight is irrelevant. And, in such  case, having the plane perfectly trimmed is not good, you don't want it to go and glide forever out of control. The sooner it spirals to the ground, the better for the recovery.

avsaase, 2020-12-24:
The only situation where manual mode can save you from a crash is if the fc mount somehow breaks and the fc can move around in the model. I've never had this happen to me but I think this is still a reason to have trimmed servo centers.

@MrD-RC, I didn't consider club flyers since I am not familiar with that side of the hobby. However in the situation you describe I still think my procedure could work: enable auto trim directly after landing and disarm a few seconds later. Perhaps a beeper after two seconds to confirm the setting of the new servo centers would help here.

OlivierC-FR, 2020-12-24:
Good point with the FC moving free, but yeah, this happening is ultra unlikely, and if that's because of crash damage, then the plane has other worries than centered servos.

Altho had such case on a 5" quad, all 4 M3 nylons had broken after multiple crashes ("stunt basher") and the FC was just hanging by the 8 thick wires to the ESCs. Because flying 100% of  the time in acro mode, it took a surprinsigly long time for me to notice that...

MrD-RC, 2020-12-26:
> The whole idea of trimming is wrong from the very beginning in INAV, IMHO there should not be any trims anyway. Put the FC level, center the linkages and servos so they have the same amount of travel each side, include the reflex on fixed wing if needed, so the "zeros" are reasonably close, and that's done. Then if the aircraft does not fly level in self-levelling modes, adjust the board angles. We let the flight controller do its job. If not, why even put a FC in the first place.

@OlivierC-FR This logic is flawed. Firstly with this you may as well say remove manual too by doing this. Everything that you are stating here will mean that manual mode will not fly well. For that, the model needs to be properly trimmed. You can't just rely on the FC as it is bypassed. **A lot of people actually like flying planes in manual.** It's not just a backup flight mode for if the FC is not setup correctly. Manual mode is not just a fall-back. However, if it is used as a fall-back, it's not because the FC has failed or come loose; but because the stabilisation modes are not working as expected. Most cases this would be in early flights due to some kind of setup error, which I and I'm sure many others have encountered. But more often is that people just want to fly in manual. Quad guys may not get this, but plane guys do.

Secondly, you mention reflex. How would you know the exact amount of reflex needed for your model? Again this is where trimming would come in to play. You can't rely on manuals. If they even state the reflex, it won't always be correct for your model. I've certainly had models where the reflex is different to that stated in the manual, if its even stated at all. What about motor torque correction? All things that are trimmed out.

Thirdly, if the model is trimmed out and flying great in manual, the flight controller will have an easier time. So why not set up the model correctly?

> I didn't consider club flyers since I am not familiar with that side of the hobby. However in the situation you describe I still think my procedure could work: enable auto trim directly after [...]

@avsaase As I mentioned, it would depend on the rules of the club. These can vary greatly. If they are not allowed to hang around on the strip or be armed in the pits, it won't be an acceptable method. 

The beeper thing is a good idea, for people with beepers fitted. However, there's not much need for it if servo autotrim is being used in the manner it was designed for. The model will be in the air beeping. I guess it wouldn't hurt, so long as the beep only occurred once after 2 seconds. Otherwise it could get annoying xD.

But I honestly don't see what issues it would cause by having a simple function that is accessible when disarmed. As I mentioned, it would also be more intuitive for people coming from Vector, as well as simpler for traditional LOS pilots. It wouldn't impact people who are currently happy with servo autotrim. It would just make things simple for others. It's just making iNav more inclusive.

avsaase, 2020-12-26:
Personally I do not see why you would go through all the hassle of setting up INAV to then only fly in manual mode. It is very well possible that I am missing something here but this is what motivates my position in this issue. If you dislike the wind stabilization you can just set the P-gains to zero and still have the automatic trimming from the I-term. I've also never had issues with auto trim. 

> Secondly, you mention reflex. How would you know the exact amount of reflex needed for your model? Again this is where trimming would come in to play. You can't rely on manuals. If they even state [...]

IMO all this can be taken care of by the stabilization.

> As I mentioned, it would depend on the rules of the club. These can vary greatly. If they are not allowed to hang around on the strip or be armed in the pits, it won't be an acceptable method.

I highly doubt club rules make it impossible to leave the model on the strip for the three seconds needed to perform the auto trim procedure I described. There is no need to take the model from the strip in the armed state.

Another reason not to trim using the transmitter trims is that you need to make sure these trims do not carry over into the other modes. This is easy in OpenTX but may not be on other transmitters (I think, never tried anything else myself). Also, this is yet another thing that pilots new to INAV would need to setup so it creates the possibility for new problems. In general, I think it is best to leave all the flight dynamics stuff to the FC and keep the radio setup as easy as possible. 

To be clear, I am not opposed to other ways to set the servo centers. There can be multiple ways to do things and we don't all have to agree  on which is the best one. The neatest solution would be a stick command but the problem is that these move the control surfaces so it would need to use the servo positions from a couple seconds before the command is triggered.

MrD-RC, 2020-12-26:
> Personally I do not see why you would go through all the hassle of setting up INAV to then only fly in manual mode. It is very well possible that I am missing something here but this is what [...]

Some people just like to fly manual, its purer. Some people just use iNav as an OSD with the added benefit of RTH, and fly in manual. Some pilots just point the plane and use 3D cruise or waypoint missions. Some could argue that's not even flying the plane at all. But, that is the beauty of iNav. There are many different types of pilots, all with different preferences, and all can be supported.

> IMO all this can be taken care of by the stabilization.

Not if you're not using the stabilisation ;) I like flying a mix, so will use manual and stabilised modes. With all my models, I like to know that they are going to fly well in manual. The flight controller just sits on top of this. A plane flying well in manual can only help the flight controller after all. The FC shouldn't be there to compensate for a badly set up plane.

> I highly doubt club rules make it impossible to leave the model on the strip for the three seconds needed to perform the auto trim procedure I described. There is no need to take the model from the strip in the armed state.

There are clubs where you must leave the strip immediately after landing. This is so the strip is clear for others to land. For example an IC plane suffers a dead stick. 

> Another reason not to trim using the transmitter trims is that you need to make sure these trims do not carry over into the other modes. This is easy in OpenTX but may not be on other transmitters [...]

Fair point. This is really simple to do in OpenTX. But there are pilots who still use other systems.

> To be clear, I am not opposed to other ways to set the servo centers. There can be multiple ways to do things and we don't all have to agree on which is the best one. The neatest solution would be a [...]

That is a very good point about the stick commands. There would need to be something in place to ensure the stick command position is not the centre point that is recorded.

wx4cb, 2021-03-22:
@MrD-RC you just want vectors transmitter trim thingy :D :D that said, i do agree, i currently do the silly auto trim crap in the ground.

I personally use opentx's "instant trim" feature on my momentary switch, makes it so much easier... fly s/l hit the button, land auto trim on ground and save/disarm

closed 2022-02-03 by DzikuVx