iNavFlight/inav #6441: Mamba F722 Stack

dbarvik, 2020-12-22:
## Current Behavior
I2C is not working on dedicated pins

## Desired Behavior
Remap I2C to dedicated I2C pins, not to UART3 pins

## Suggested Solution
Add target board for MAMBAF722I2C stack

## Who does this impact? Who is this for?
For all users which have bought FC based on your board recommendation (https://inavflight.com/shop/s/bg/1490805)

## Additional context
It should be this new target added to inav, because you recommended this board but there is no target for it. Now I2C works only when connected to SERIAL 3 pins. But in my application I need all serial ports.
I just add the official config file from diatone
[MAMBA_F722_I2C_0716.txt](https://github.com/iNavFlight/inav/files/5731188/MAMBA_F722_I2C_0716.txt)

djc0ntra, 2021-01-27:
I'm also interested in including this target into INAV. Affecting both Mamba F722S (bluetooth) and Mamba F722APP (wifi)

Animetronics, 2021-01-29:
This looks related to my potential issue (but in my case, it's the F405).
[https://github.com/iNavFlight/inav/issues/6535](https://github.com/iNavFlight/inav/issues/6535)

Maybe the assignments can be ported from here ?
[https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/DIAT-MAMBAF722_I2C.config](https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/DIAT-MAMBAF722_I2C.config)

Animetronics, 2021-01-31:
Ok, 

After looking at the target source, I see that the F722 and F405 are near identical (only difference I see is the MCU designation ).

As per [Spychalski's Guide](https://www.youtube.com/watch?v=5wlr4WYpv1w) , I modded the MAMBA's target.h... target.c is unchanged so as not to tamper with timers. However, PB9 is attached to a PPM-in timer there. That might not effect the I2C port though, but worth remembering. 

This attached hex file _should_ leave TX3/RX3 free for UART use, while enabling the newly freed SDA/SCL pins for use. One can flash it using the local file option. 
**Just bear in mind it's experimental.**

Originally:
Line 49 `#define I2C2_SCL        PB10        // SCL pad TX3`
Line 50 `#define I2C2_SDA         PB11        // SDA pad RX3`

Modded:
Line 49 `#define I2C2_SCL        PB8      // was PB10        // SCL pad TX3`
Line 50 `#define I2C2_SDA        PB9     // was PB11        // SDA pad RX3`

[TEST_inav_2.7.0_MAMBAF722_I2C.zip](https://github.com/iNavFlight/inav/files/5900510/TEST_inav_2.7.0_MAMBAF722_I2C.zip)

Animetronics, 2021-02-11:
Update:

I attempted to flash my Flight controller with the above modification. INAV did not recognize the board. Recovering the FC to it's original firmware (Betaflight in my case) was easy. I've removed the link to the HEX file for this reason.

I'll try to contact the MAMBA_405 and 722 HEX author for clarification soon.

dbarvik, 2021-02-11:
I've made these changes earlier with the board name change. And now I have flashed my custom FW in that board which works well. 
This wont be the stable release but I didn't have any problem with my quad since I flashed it.

[inav_2.7.0_MATEKF722.zip](https://github.com/iNavFlight/inav/files/5963455/inav_2.7.0_MATEKF722.zip)

Animetronics, 2021-02-11:
Great to know ! So the new I2c pins worked with your custom hex file (Baro
detected, etc)?

Mine is a MAMBA_405, it looks like the code and pin layout are the same...
But it would have to be recompiled since the F405 is a different MCU.

What did you change, exactly ? Did target .H and .C
 both require modification?

dbarvik, 2021-02-11:
The I2C pins PB8 and PB9 are not the I2C2 bus, but the I2C1 based on datasheet
[image]

Here I attach the target folder. And I use the I2C for the BMP280 baro and compas from Matek M8Q-5883 module.

[MAMBAF722I2C.zip](https://github.com/iNavFlight/inav/files/5963961/MAMBAF722I2C.zip)

Animetronics, 2021-02-11:
I see, so I _was_ using an incorrect timer/interrupt then. Ok, I've recompiled it and have gotten further along with connecting.
However, I'm now denied access to the configuratior, saying "You need to upgrade your firmware before you can use the [whatever] tab."

The [version] command in CLI says it's 2.7.0. My configurator is 2.6.1 is there a development configurator?

dbarvik, 2021-02-11:
I somewhere found the 2.7.0 inav configurator. I am sorry, but i forgot where it was :(.

stronnag, 2021-02-11:
http://seyrsnys.myzen.co.uk/inav-configurator-next/

Animetronics, 2021-02-11:
Thank you! Its reading now. 
I had to go through all Barometers until I got to MSP... strange. I'm guessing that this must be a serial barometer, then ?
Secondly, it keeps zeroing out, but I see that this is due to me not being armed.

I'll attach the modded hex to my issue.

positivcheg94, 2021-03-15:
@dbarvik Hey, are you gonna make a pull request? I just happened to like this FC but tied to use Betaflight since there is no target in INAV and even though I'm a programmer I have 0 understanding of how builds work in this C+CMake repo

coulsonrobert, 2021-03-24:
Have we got any closer to an INAV target for this board? I've got the Mk3 Mamba F722 Basic and I can't run INAV due to the lack of I2C target and hence baro+mag. I'd be happy to lose UART3 however there's no TX3 pad, and it still doesn't solve the baro issue. Help!

Or....can anyone describe how to convert the zip folder above  by dbarvik into a working .hex?

Thanks! back to Betaflight for now :-(

positivcheg94, 2021-03-24:
> Have we got any closer to an INAV target for this board? I've got the Mk3 Mamba F722 Basic and I can't run INAV due to the lack of I2C target and hence baro+mag. I'd be happy to lose UART3 however [...]

Unless one of the INAV devs gets interested on this board and make working firmware we have no chances. I tried building it with the folder provided by @dbarvik, but I2C doesn't work. 
However, you can easily fly inav with it by flashing F722 target. The only problem will be that I2C won't work and Rx3+Tx3 will work as I2C.

coulsonrobert, 2021-03-24:
> Unless one of the INAV devs gets interested on this board and make working firmware we have no chances. I tried building it with the folder provided by @dbarvik, but I2C doesn't work.
> However, you [...]

I don’t have both TX3 and RX3 pads on my Mk3 though? Also it doesn’t get around the baro issue which isn’t recognised. Any thoughts?

[image]

trudeaudm, 2021-03-25:
> Have we got any closer to an INAV target for this board? I've got the Mk3 Mamba F722 Basic and I can't run INAV due to the lack of I2C target and hence baro+mag. I'd be happy to lose UART3 however [...]

Dealing with the same issue! No TX3 pad 😑 I guess I should have done more digging before expecting INAV to be this functional.

coulsonrobert, 2021-04-01:
> Thank you! Its reading now.
> I had to go through all Barometers until I got to MSP... strange. I'm guessing that this must be a serial barometer, then ?
> Secondly, it keeps zeroing out, but I see that this is due to me not being armed.
> 
> I'll attach the modded hex to my issue.

Any chance you could post the modified hex file?

coulsonrobert, 2021-04-01:
> I've made these changes earlier with the board name change. And now I have flashed my custom FW in that board which works well.
> This wont be the stable release but I didn't have any problem with my [...]

Is this a typo (is it a matek or mamba target?)

thanks!

dbarvik, 2021-04-02:
> Is this a typo (is it a matek or mamba target?)
>
> thanks!

Thats not typo, I just changed this targer to MambaF722 I2C and I forget rename it.

coulsonrobert, 2021-04-02:
> Thats not typo, I just changed this targer to MambaF722 I2C and I forget rename it.

Great, so it should flash to my mamba f722i2C ok?

many thanks for your work!

positivcheg94, 2021-04-02:
> Great, so it should flash to my mamba f722i2C ok?
>
> many thanks for your work!

For me his target folder didn't work though. Betaflight sees both i2c external baro + mag but inav doesn't =(

Inspektor7, 2021-04-02:
I have the same problem. Betaflight sees the MS5611 barometer and the RadioLink TS100 M8N GPS module. Inav no.

djc0ntra, 2021-04-02:
> The I2C pins PB8 and PB9 are not the I2C2 bus, but the I2C1 based on datasheet
> [image]
> 
> Here I attach the target folder. And I use the I2C for the BMP280 baro and compas from Matek M8Q-5883 module.
> 
> [MAMBAF722I2C.zip](https://github.com/iNavFlight/inav/files/5963961/MAMBAF722I2C.zip)

This one almost worked, but my compass was showing up as red in inav; MAG: UNAVAILABLE in status.

I made the following changes in target.h and now my manometer is working on my Mamba F722 APP flight controller:

**Original:**

// *************** Baro **************************
#define USE_I2C
#define USE_I2C_DEVICE_2

#define I2C2_SCL                        PB10        // SCL pad TX3
#define I2C2_SDA                        PB11        // SDA pad RX3
#define DEFAULT_I2C_BUS                 BUS_I2C2

#define USE_BARO
#define BARO_I2C_BUS                    DEFAULT_I2C_BUS

#define USE_BARO_BMP280
#define USE_BARO_MS5611

//*********** Magnetometer / Compass *************
#define USE_MAG
#define MAG_I2C_BUS                     DEFAULT_I2C_BUS

#define USE_MAG_HMC5883
#define USE_MAG_QMC5883
#define USE_MAG_LIS3MDL

**Modified:**

// *************** I2C **************************
#define USE_I2C
#define USE_I2C_DEVICE_1

#define I2C1_SCL                        PB8        
#define I2C1_SDA                        PB9        
#define DEFAULT_I2C_BUS                 BUS_I2C1

// *************** Baro **************************
#define USE_BARO
#define BARO_I2C_BUS                    DEFAULT_I2C_BUS

#define USE_BARO_BMP280
#define USE_BARO_MS5611

//*********** Magnetometer / Compass *************
#define USE_MAG
#define MAG_I2C_BUS                     DEFAULT_I2C_BUS

#define USE_MAG_HMC5883
#define USE_MAG_QMC5883
#define USE_MAG_LIS3MDL

The main difference is that I had to change everything referencing I2C Device 2 to I2C Device 1, keeping reference to pins PB8 and PB9. I also decided to break out the I2C configuration into it's own section, but you can keep it all under the barometer part if that's the inav standard. I just thought it looked neater. 

Attached is the compiled firmware with these changes. Use at your own risk. Maybe someone can comment if their barometer is working too since I haven't tried connecting my BME280 yet (which I'm not sure is even compatible).

[inav_2.7.0_MAMBAF722I2C.zip](https://github.com/iNavFlight/inav/files/6249662/inav_2.7.0_MAMBAF722I2C.zip)

coulsonrobert, 2021-04-02:
> > The I2C pins PB8 and PB9 are not the I2C2 bus, but the I2C1 based on datasheet
> > [image]
> > Here I attach the target folder. And I use the I2C for the BMP280 baro and compas from Matek M8Q-5883 module.
> > [MAMBAF722I2C.zip](https://github.com/iNavFlight/inav/files/5963961/MAMBAF722I2C.zip)
> 
> This one almost worked, but my compass was showing up as red in inav; MAG: UNAVAILABLE in status.
> 
> I made the following changes in target.h and now the compass is working on my Mamba F722 APP flight controller:
> 
> **Original:**
> 
> // *************** Baro **************************
> #define USE_I2C
> #define USE_I2C_DEVICE_2
> 
> #define I2C2_SCL PB10 // SCL pad TX3
> #define I2C2_SDA PB11 // SDA pad RX3
> #define DEFAULT_I2C_BUS BUS_I2C2
> 
> #define USE_BARO
> #define BARO_I2C_BUS DEFAULT_I2C_BUS
> 
> #define USE_BARO_BMP280
> #define USE_BARO_MS5611
> 
> //*********** Magnetometer / Compass *************
> #define USE_MAG
> #define MAG_I2C_BUS DEFAULT_I2C_BUS
> 
> #define USE_MAG_HMC5883
> #define USE_MAG_QMC5883
> #define USE_MAG_LIS3MDL
> 
> **Modified:**
> 
> // *************** I2C **************************
> #define USE_I2C
> #define USE_I2C_DEVICE_1
> 
> #define I2C1_SCL PB8
> #define I2C1_SDA PB9
> #define DEFAULT_I2C_BUS BUS_I2C1
> 
> // *************** Baro **************************
> #define USE_BARO
> #define BARO_I2C_BUS DEFAULT_I2C_BUS
> 
> #define USE_BARO_BMP280
> #define USE_BARO_MS5611
> 
> //*********** Magnetometer / Compass *************
> #define USE_MAG
> #define MAG_I2C_BUS DEFAULT_I2C_BUS
> 
> #define USE_MAG_HMC5883
> #define USE_MAG_QMC5883
> #define USE_MAG_LIS3MDL
> 
> The main difference is that I had to change everything referencing I2C Device 2 to I2C Device 1, keeping reference to ports PB8 and PB9. I also decided to break out the I2C configuration into it's own section, but you can keep it all under the barometer part if that's the inav standard. I just thought it looked neater.
> 
> Attached is the compiled firmware with these changes. Use at your own risk. Maybe someone can comment if their barometer is working too since I haven't tried connecting my BME280 yet (which I'm not sure is even compatible).
> 
> [inav_2.7.0_MAMBAF722I2C.zip](https://github.com/iNavFlight/inav/files/6249662/inav_2.7.0_MAMBAF722I2C.zip)

@djc0ntra legend, thanks I'll try this tonight 👍

coulsonrobert, 2021-04-02:
@djc0ntra works fine! I2C seems to be online as it’s picked up the onboard baro and my magnetometer straight away. 👍 

🤞 for a test flight tomorrow.

ivirtex, 2021-04-12:
How is progress going? I bought my Mamba F722I2C from iNav's recommendation page and can't flash it due to lack of target in the iNav configurator.
Do devs accept PR from users?

DzikuVx, 2021-04-12:
F722I2C is scheduled to be supported in INAV 3.0

coulsonrobert, 2021-04-12:
> F722I2C is scheduled to be supported in INAV 3.0

Legend, thanks Pawel 👍

zeesh1979, 2021-04-18:
any rough idea when inav3.0 will be available for use?

zeesh1979, 2021-04-18:
> This one almost worked, but my compass was showing up as red in inav; MAG: UNAVAILABLE in status.
>
> I made the following changes in target.h and now the compass is working on my Mamba F722 APP flight [...]

i used this hex file to flash my mamba f722 mini mk3...its done successfully but when i connect, it doesnt let me go to any tab and stays in CLI. when i click any tab, it says i need to upgrade the firmware

robertha64, 2021-04-21:
You need to use preliminary Configurator version (2.7.0).

zeesh1979, 2021-04-21:
Can you send me the link to download that version?

Inspektor7, 2021-04-21:
[image]
I have a Roma F5 6S with a Diatone Mamba F722APP MK1 WiFi 50A 6S flight. connected via I2C bus, barometer MS5611 and GPS module RadioLink TS100 M8N. and this result did not work, inav does not determine the compass and baro. everything works in Betaflight. and now in inav. THANKS)) sorry, I use a translator from Ukrainian)

robertha64, 2021-04-21:
> Can you send me the link to download that version? Get Outlook for Android<https://aka.ms/ghei36>
> […](#)
> 
http://seyrsnys.myzen.co.uk/inav-configurator-next/

robertha64, 2021-04-21:
Diatone Mamba 722APP and Diatone Mamba Basic 722 MK3 discussed here are 2 slightly different targets.
You need to check resources of your board in BF by using command "resource" in CLI and paste result here. 
The I2C pins could be on different mapping than software expects.

zeesh1979, 2021-04-21:
when i click on Load Firmware (Local), nothing happens

http://seyrsnys.myzen.co.uk/inav-configurator-next/

zeesh1979, 2021-04-22:
I tried betaflight to update fw as inav3.0 wasn't opening window to load local file. After updating via BF i connected inav3.0 but it didn't let me goto setup tab... I used ur mamba f722i2c to flash mamba stack f722 mini mk3

http://seyrsnys.myzen.co.uk/inav-configurator-next/

mecke96, 2021-05-06:
Hi! I have a mamba F722 MK3 basic. Was going to flash the hex that "djc0ntra" provided. But now when the devs have updated the Configurator version to 3.0.0 im stuck at CLI. Is there anyone that can send me the 2.7.0 configurator or if "djc0ntra" is willing to update his hex file for 3.0.0 version. 

Kind regards// Melker S

Inspektor7, 2021-05-06:
https://github.com/iNavFlight/inav/pull/6866

on my Mamaba F722_I2C works

mecke96, 2021-05-06:
> #6866
> 
> on my Mamaba F722_I2C works

Did you flash it with version "mamba f722" 2.6.1

For me the inbuilt baro won't work. It's just red in the top bar in the configurator 🤔

ivirtex, 2021-05-06:
> > #6866
> > on my Mamaba F722_I2C works
> 
> Did you flash it with version "mamba f722" 2.6.1
> 
> For me the inbuilt baro won't work. It's just red in the top bar in the configurator 🤔

Your baro doesn't work because you need to flash the F722_I2C version. For me this version works fine, all sensors are functional.

[inav_2.6.1_MAMBAF722_I2C.hex.zip](https://github.com/iNavFlight/inav/files/6437695/inav_2.6.1_MAMBAF722_I2C.hex.zip)

mecke96, 2021-05-06:
> Your baro doesn't work because you need to flash the F722_I2C version. For me this version works fine, all sensors are functional. [...]

Oh thank you! Finally i can get my build going 😃

DzikuVx, 2021-05-08:
2.6.1 avaliable via Configurator

closed 2021-05-08 by DzikuVx

TOME00721, 2021-05-26:
Thank you guys this has saved me !!! Long range time!

Badfrog99, 2022-04-28:
Is it safe to assume this will also work for the Mamba Basic F722 mini mk3?

NiceFpv, 2022-05-01:
> Is it safe to assume this will also work for the Mamba Basic F722 mini mk3?

Sombody knows?