iNavFlight/inav #6444: Flywoo F7 Dual Mini wild compass calibration values unusable on 2.6.0 stable

kelleysislander, 2020-12-23:
I went to calibrate the magnetometer on the Matek M8Q-5883 and am getting wild out of range values and of course it does not work correctly so this build is a non-op as currently configured.  

set magzero_x = -5848
set magzero_y = -18286
set magzero_z = 11515
set maggain_x = 17095
set maggain_y = -23349
set maggain_z = 4100 

## Current Behavior
Run through normal calibration procedure and get wildly invalid results and the quad just jerks around in the setup tab.  Cannot use iNav 2.6.0 on this board due to bad compass calibration.

## Steps to Reproduce
Calibrate magnetometer, observe resultant values.

## Expected behavior
Should calibrate this Matek M8Q-5883 and get valid results, please see the diff all posted below and the pastebin.

## Suggested solution(s)
self explanatory

## Additional context

Full dump here:
[https://pastebin.com/WaSwDxP3](url)

---
# version
# INAV/FLYWOOF7DUAL 2.6.0 Dec  4 2020 / 11:05:11 (cada68e4)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

- Flywoo F7 Dual Mini

Here is the diff all output

# diff all

# version
# INAV/FLYWOOF7DUAL 2.6.0 Dec  4 2020 / 11:05:11 (cada68e4)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# safehome

# logic

# gvar

# feature
feature -AIRMODE
feature VBAT
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature LED_STRIP
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper

# map
map TAER

# serial
serial 1 32 115200 115200 0 115200
serial 2 2 115200 115200 0 115200
serial 3 2097152 115200 115200 0 115200
serial 5 262144 115200 115200 0 115200

# led
led 0 6,15::AB:0
led 1 7,15::AO:0
led 2 8,15::AO:0
led 3 9,15::AB:0

# color

# mode_color

# aux
aux 0 0 3 1300 1700
aux 1 2 0 1225 1975
aux 2 29 0 1125 1950
aux 3 13 3 1700 2100
aux 4 15 1 1300 2100
aux 5 19 1 1700 2100
aux 6 26 4 1700 2100
aux 7 42 2 1075 1300
aux 8 43 2 1300 1700
aux 9 44 2 1700 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 20 15 V
osd_layout 0 1 16 15 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 1 6 H
osd_layout 0 8 0 0 H
osd_layout 0 9 23 15 V
osd_layout 0 11 8 15 V
osd_layout 0 12 4 15 V
osd_layout 0 13 9 14 V
osd_layout 0 14 6 14 V
osd_layout 0 15 15 13 V
osd_layout 0 20 15 14 V
osd_layout 0 21 11 14 V
osd_layout 0 22 14 13 V
osd_layout 0 23 12 14 V
osd_layout 0 24 25 14 V
osd_layout 0 28 0 14 V
osd_layout 0 30 0 6 V
osd_layout 0 32 12 15 V
osd_layout 0 34 3 13 H
osd_layout 0 40 22 0 H
osd_layout 0 99 14 13 H
osd_layout 0 106 23 1 H
osd_layout 1 0 20 15 V
osd_layout 1 1 16 15 V
osd_layout 1 7 1 6 H
osd_layout 1 8 18 0 H
osd_layout 1 9 23 15 V
osd_layout 1 11 8 15 V
osd_layout 1 12 4 15 V
osd_layout 1 13 5 0 V
osd_layout 1 14 10 0 V
osd_layout 1 15 1 0 V
osd_layout 1 20 12 1 V
osd_layout 1 21 1 1 V
osd_layout 1 22 14 13 V
osd_layout 1 23 16 0 V
osd_layout 1 28 0 14 V
osd_layout 1 30 0 6 V
osd_layout 1 32 12 15 V
osd_layout 1 34 10 14 V
osd_layout 1 40 22 0 V
osd_layout 1 99 14 13 H
osd_layout 1 106 23 1 V
osd_layout 2 0 20 15 V
osd_layout 2 1 16 15 V
osd_layout 2 7 1 6 H
osd_layout 2 8 18 0 H
osd_layout 2 9 23 15 V
osd_layout 2 11 8 15 V
osd_layout 2 12 4 15 V
osd_layout 2 13 5 0 V
osd_layout 2 14 10 0 V
osd_layout 2 15 1 0 V
osd_layout 2 20 12 1 V
osd_layout 2 21 1 1 V
osd_layout 2 22 14 13 V
osd_layout 2 23 16 0 V
osd_layout 2 28 0 14 V
osd_layout 2 30 0 6 V
osd_layout 2 32 12 15 V
osd_layout 2 34 10 14 V
osd_layout 2 40 22 0 V
osd_layout 2 99 14 13 H
osd_layout 2 106 23 1 V
osd_layout 3 0 20 15 V
osd_layout 3 1 16 15 V
osd_layout 3 7 1 6 H
osd_layout 3 8 18 0 H
osd_layout 3 9 23 15 V
osd_layout 3 11 8 15 V
osd_layout 3 12 4 15 V
osd_layout 3 13 5 0 V
osd_layout 3 14 10 0 V
osd_layout 3 15 1 0 V
osd_layout 3 20 12 1 V
osd_layout 3 21 1 1 V
osd_layout 3 22 14 13 V
osd_layout 3 23 16 0 V
osd_layout 3 28 0 14 V
osd_layout 3 30 0 6 V
osd_layout 3 32 12 15 V
osd_layout 3 34 10 14 V
osd_layout 3 40 22 0 V
osd_layout 3 99 14 13 H
osd_layout 3 106 23 1 V

# master
set looptime = 250
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 110
set gyro_lpf_type = PT1
set gyro_stage2_lowpass_hz = 250
set gyro_stage2_lowpass_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 200
set acc_hardware = MPU6000
set acczero_x = 27
set acczero_y = 3
set acczero_z = 88
set accgain_x = 4071
set accgain_y = 4112
set accgain_z = 4070
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set magzero_x = -5848
set magzero_y = -18286
set magzero_z = 11515
set maggain_x = 17095
set maggain_y = -23349
set maggain_z = 4100
set baro_hardware = BMP280
set pitot_hardware = NONE
set min_check = 1050
set rssi_channel = 16
set serialrx_halfduplex = OFF
set blackbox_rate_denom = 4
set motor_pwm_rate = 16000
set motor_pwm_protocol = DSHOT600
set throttle_idle =  5.000
set motor_poles = 12
set failsafe_procedure = DROP
set align_board_yaw = 1800
set model_preview_type = 3
set applied_defaults = 2
set rpm_gyro_filter_enabled = ON
set gps_provider = UBLOX7
set deadband = 3
set yaw_deadband = 3
set mc_airmode_type = THROTTLE_THRESHOLD
set nav_auto_speed = 894
set nav_rth_climb_first = OFF
set nav_rth_altitude = 3810
set i2c_speed = 800KHZ
set name = Fouride5

# profile
profile 1

set mc_p_pitch = 44
set mc_i_pitch = 60
set mc_d_pitch = 25
set mc_i_roll = 50
set mc_p_yaw = 45
set mc_i_yaw = 70
set dterm_lpf_hz = 100
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 150
set dterm_lpf2_type = PT1
set yaw_lpf_hz = 30
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set setpoint_kalman_enabled = ON
set setpoint_kalman_q = 200
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

set vbat_warning_cell_voltage = 340
set battery_capacity = 1300
set battery_capacity_warning = 429
set battery_capacity_critical = 286

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

kelleysislander, 2020-12-24:
Went back to 2.5.2 to see if that made a difference - just more of the same, and the compass calibration made no values for any gains, just the x,y,z values:

set magzero_x = -4687
set magzero_y = -18410
set magzero_z = 12963

DzikuVx, 2021-04-07:
That mean that MAG is either faulty or you have it very close to string source of magnetic field. Calibration will not fix that

closed 2021-04-07 by DzikuVx