iNavFlight/inav #6447: Add DURATION and DELAY to Logic Conditions

Jetrell, 2020-12-25:
I believe these features would be useful to determine how long a logic condition will stay active after there is a state change. Or after how long the next state change would take effect.

DzikuVx, 2020-12-26:
We mightr do it for boolean values, or rather as a new Logic Condition.
Delay and Duration for int would take too much RAM

Jetrell, 2020-12-26:
> 
> 
> We might do it for boolean values, or rather as a new Logic Condition.
> Delay and Duration for int would take too much RAM

@DzikuVx Having it as a new logic condition might give the largest amount of flexibility?
After using the Logic Conditions frequently. I have found  - Greater than, Less than, Equal too and XOR. To be more useful than general boolean.

Mateyhv, 2020-12-29:
I would also consider adding an inverted option for the available operands and logic condition. Much better than using another slot just for the NOT.

DzikuVx, 2020-12-29:
Personally I'd prefer keeping it closer to RISC than CISC concept. Inverted and so on might looks like a good idea to simplify the process, but adds complexity and additional flags and logic to drive it

closed 2022-07-05 by Jetrell