iNavFlight/inav #6457: ghost protocol failsafe after ca. 4300s ontime

Christoph-Hofmann, 2020-12-29:
## Current Behavior
After ca 4300s ontime the flightcontroller goes into failsafe and does not comeback unless you reboot it.
I checked this several times on the bench.

In BF this does not happen, but bf resets the ontime to 0 after 4300s.
So i assume there is an overflow in a time relevated variable.

config dump: https://pastebin.com/rq3hWiSZ

# version
# INAV/MATEKF405 2.7.0 Dec 27 2020 / 16:20:50 (04e10aef)
# GCC-9.3.1 20200408 (release)

digitalentity, 2020-12-29:
Does it happen with S.Bus or other RX drivers?

Christoph-Hofmann, 2020-12-29:
@digitalentity 
the receiver is connected to the tx pad.
The settings in the transmitter are:
RX ProtoGHST

RF Mode Normal
Tx Mode Auto
Region EU (LBT)

DzikuVx, 2020-12-29:
@digitalentity I have a hunch it's due to overflow in the lost frame detection in GHST implementation.

```
typedef struct ghstFailsafeTracker_s {
    unsigned onTimePacketCounter;
    timeMs_t lastSeenMs;
} ghstFailsafeTracker_t;
```

my guess is  `onTimePacketCounter` overflow

digitalentity, 2020-12-29:
@Christoph-Hofmann does it happen with other type of receivers?

@DzikuVx `onTimePacketCounter` can not overflow, it's incrementing is capped by `GHST_RC_FRAME_COUNT_THRESHOLD`. 

4300s seems to be a 32-bit overflow for `micros()`, I suspect we have a bug in `rx.c` code again where we truncate `timeUs_t` to 32-bits.

Another theory, especially if this happens only with `GHST` is that the GHST RX stops sending some of the packets after 32-bit microtime overflow happens inside the RX firmware. Since we're watching for all channels in GHST driver - we detect this issue, while BF is only concerned about total loss of the datastream.

Christoph-Hofmann, 2020-12-29:
@digitalentity i cross checked with fport , which does run for serveral hours without problem.

Christoph-Hofmann, 2020-12-29:
@digitalentity to find out if there is a bug in tx or rx firmware, i reset the time to 0 after 4450s and then the receiver work fine again.

drivers/time.c

timeMs_t millis(void)
{
  if (sysTickUptime>4450*1000) {
    sysTickUptime=0;
  }
    return sysTickUptime;
}
so after 32bit overflow the ghost packets are lost.
reset the time to 0 ghost packets are fine.

DzikuVx, 2020-12-29:
so it looks like a problem in GHST driver after all.

digitalentity, 2020-12-29:
I tested with S.Bus for 1.5 hours - no issue, so this is indeed Ghost-specific. I'm curious what's so special about forcing overflow manually (before 32-bit overflow).

digitalentity, 2020-12-29:
@Christoph-Hofmann let's check a few things:

1. Let's try triggering this manually by starting at non-zero time. Can you please change the initial value here `STATIC_UNIT_TESTED volatile timeMs_t sysTickUptime = 0;` from `0` to `4294907`? This will cause microtime to cross 32-bit boundary in 60 seconds.

2. Let's see if we continue receiving GHST packets after the overflow happens.

Can you change `ghst.c` and add `debug[0]++` after line 170? You should end up with something like this:

```
    if (ghstFrameIdx < fullFrameLength) {
        ghstIncomingFrame.bytes[ghstFrameIdx++] = (uint8_t)c;
        if (ghstFrameIdx >= fullFrameLength) {
            ghstFrameIdx = 0;

            // NOTE: this data is not yet CRC checked, nor do we know whether we are the correct recipient, this is
            // handled in ghstFrameStatus
            memcpy(&ghstValidatedFrame, &ghstIncomingFrame, sizeof(ghstIncomingFrame));
            ghstFrameAvailable = true;
            debug[0]++;

            // remember what time the incoming (Rx) packet ended, so that we can ensure a quite bus before sending telemetry
            ghstRxFrameEndAtUs = microsISR();
        }
    }
```

And let's see if the debug[0] continues to increase when we hit the problem.

Christoph-Hofmann, 2020-12-29:
I have to change the time related stuff to 
bool timeshiftset = false;
// Return system uptime in milliseconds (rollover in 49 days)
timeMs_t millis(void)
{
  if (sysTickUptime>25000 && !timeshiftset) {
    timeshiftset=true;
    sysTickUptime=4220*1000;
  }
    return sysTickUptime;
}

because the board will not initialize, if you start at other then zero.
So after 25 seconds there is a fast forword to 4220s.

because i have no hardware debugging tool, i use Serial printf style debugging as follow

STATIC_UNIT_TESTED void ghstDataReceive(uint16_t c, void *data)
{
static uint32_t packetCounter = 0;
const timeMs_t currentTimeMs = millis();
...

if (ghstFrameIdx < fullFrameLength) {
        ghstIncomingFrame.bytes[ghstFrameIdx++] = (uint8_t)c;
        if (ghstFrameIdx >= fullFrameLength) {
            ghstFrameIdx = 0;

            // NOTE: this data is not yet CRC checked, nor do we know whether we are the correct recipient, this is
            // handled in ghstFrameStatus
            memcpy(&ghstValidatedFrame, &ghstIncomingFrame, sizeof(ghstIncomingFrame));
            ghstFrameAvailable = true;
            packetCounter++;
            if ((currentTimeMs - lastSendMsgLong) >= MSGUPDATEINTERVALL) {
              LOG_E(SYSTEM, "packetcount :%lu",packetCounter);
              lastSendMsgLong = millis();
            }

            // remember what time the incoming (Rx) packet ended, so that we can ensure a quite bus before sending telemetry
            ghstRxFrameEndAtUs = microsISR();
        }
    }

the output of the debug can be found here:

https://pastebin.com/xJdjLqzC

the failsafe start around 4296s

digitalentity, 2020-12-29:
I can't see anything wrong with the GHST code, but apparently I must be missing something.

I've tested that `micros()` is consistent when crossing 32-bit boundary and found no issue. Time monotonically increments without jumps or going backwards so this is either GHST code or the way GHST interacts with RX.

I wonder what condition in `ghstDetectFailsafe()` is triggered when failsafe gets stuck.

DzikuVx, 2020-12-29:
I might play with it tomorrow, should have everything to do so

Christoph-Hofmann, 2020-12-29:
before the failsafe starts you can see in the inav configurators receiver tab that the values are jumping around

digitalentity, 2020-12-29:
Seems like the protocol sync is being lost. 

One thing that may amplify the issue is https://github.com/iNavFlight/inav/blob/master/src/main/rx/ghst.c#L163 - we don't do any bondary checking on `ghstFrameIdx` so it's possible to overrun the `ghstIncomingFrame` and write garbage into adjacent variables. While this is orthogonal to timing problem, this may contribute to severity of the issue.

DzikuVx, 2020-12-30:
Interesting, the process of FS is not a rapid change to FS mode. It's more like gradual process with signal resteored from time to time. 

When FS occures, it's triggered by

```
if ((currentTimeMs - ghstFsTracker[pktIdx].lastSeenMs) >= GHST_RC_FRAME_TIMEOUT_MS) 
```

DzikuVx, 2020-12-30:
Interestingly, the frame that causes the failsafe is `GHST_UL_RC_CHANS_HS4_9TO12`. It's not registered anymore

DzikuVx, 2020-12-30:
OK, I found the issue. It's the telemetry frame. After going above threshold it's triggered in wrong moments and interrupts RC frames. Do not know why yet

DzikuVx, 2020-12-30:
OK, it's caused by `shouldSendTelemetryFrame` returning `true` in wrong moments. I still have no idea why yet

DzikuVx, 2020-12-30:
One step deeper. It's caused by `handleGhstTelemetry` starting to produce new telemetry frames every 2ms instead on 100ms

PR #6462 (2020-12-31): Fix GHST telemetry
Fixes #6457

After overflow, telemetry was generated every 2ms instead of 100ms. That was slowly clogging serial port so failsafe was triggered.

@tonycake you might be interested in this one
closed 2020-12-31 by digitalentity