iNavFlight/inav #6461: iNav 2.6 Copter RTH Crash

ralphwu, 2020-12-30:
I switch to RTH after ALT+POS Hold Mode in about 5 meters with 15 Sat Count. Then the copter become uncontrollable and crash.

stronnag, 2020-12-30:
Please provide a blackbox log.

ralphwu, 2020-12-30:
Blackbox log
[https://drive.google.com/file/d/1O6_WvRk8MCn1c_75qsSnLq5V7Nj5TR1W/view?usp=sharing
](url)

stronnag, 2020-12-30:
The log is rather corrupt; however it indicates that the craft was not in an automated navigation mode when it crashed, it was in Horizon mode.

In the image:
* Craft launched in piloted mode (cyan track)
* Craft flown NE in PosHold (green track), to about 22m range,..
*  RTH initiated (yellow)
* Craft is switched to piloted about 8m NE of home position
*  Craft crashes about 8m SW of home location, in Horizon mode.

Hard to see how this is an inav problem. There is no evidence of magnetic interference (good correlation between comapss and GPS CoG).

[image]

Pairan, 2020-12-31:
mine crashed too twice now. I didn't use RTH though!

My trouble came when flying back and hitting PosHold to be able to safely pull the goggles off. Then taking over made the quad controls kinda inverted or uncontrollable so it went down.
Flashed back to 2.5.2 for now and have no log. So for now its just a heads up for "I could think, getting out of PosHold causes trouble"

closed 2021-03-02 by DzikuVx