iNavFlight/inav #6466: GPS failure after arming isn't shown anywhere

skorokithakis, 2021-01-02:
If the GPS fails after arming, no error is shown and INAV acts as if everything is fine. Distances, speeds, etc do not update, but there is no "GPS FAILURE" error message or anything.

Jetrell, 2021-01-02:
@skorokithakis There are multiple forms of GPS failure that would be nice to know, quickly. 
* Total loss of GPS or electrical failure.
* If the lock drops below 6

I have had a plane fly the wrong way in RTH, after the GPS satellite count dropped to 4. Even though the Sat count did recover within a few seconds during that test, But it still kept fly the wrong way.

skorokithakis, 2021-01-02:
That's a good point, temporary loss of satellites is especially good to hear about, as I wouldn't want my craft to be unable to RTH without me having any idea what's going on. Your plane kept flying the wrong way even after satellites recovered? That's very odd.

breadoven, 2021-01-02:
How do you know the GPS failed ?

skorokithakis, 2021-01-02:
Because it was on the ground, cables still attached.

avsaase, 2021-01-02:
Cables becoming loose could also be a form of gps failure.

Jetrell, 2021-01-02:
> 
> 
> Cables becoming loose could also be a form of gps failure.

The system should be able to detect no packets being received from the GPS. Which should give a good indication of GPS module electrical failure .

breadoven, 2021-01-04:
After finally checking this there are indications the GPS has failed after arming, the beeper sounds and the OSD satellite number flashes. It's understandable why there is no message if the GPS fails after arming, it would clog up the message display for the duration of the failure unless if was time limited in which case you may miss it if you weren't looking at the OSD at the time. However, a permanent indication of GPS failure could be set on one of the GPS displays, Satellite count being the most obvious. Perhaps rather than just showing 0 as suggested in https://github.com/iNavFlight/inav/issues/6474, Satellite count could display some symbol, e.g. F! or X! maybe, if the GPS feed is lost completely.

skorokithakis, 2021-01-04:
I don't have the GPS satellite count permanently in the display, as it's something that would only be useful in an emergency, and I can't hear the beeper more than 5 meters away. There really should be an error system message for this.

breadoven, 2021-01-05:
I made a change which sets the satellite count to 0 and displays "X!" in the OSD satellite number field if the GPS communication is lost. Don't know if it's worth doing a PR, devs to decide if this is wanted or not. I'll keep it because it's useful to know that the GPS connection is lost rather than the GPS just failing to find any/enough satellites (not that that is likely in reality unless there's poor reception). Currently it doesn't seem obvious once armed if the GPS satellite field is blinking due to lost communication or a poor fix (< sats than specified in the settings for a valid fix).

No System Message though. I'd have thought the satellite number field would be a basic requirement if you're using INav with a GPS ... and it's only 4 characters long so doesn't take up much space.

skorokithakis, 2021-01-05:
Personally, I don't see the GPS field as useful after it's seen enough satellites, as the only time where I want to be notified is if it somehow fails. If a GPS failure isn't worthy of a system message, I don't know what is.

avsaase, 2021-01-06:
@breadoven I think this would be great, but I agree with @skorokithakis that a system message `GPS FAILURE!` would be nice to have as well.

breadoven, 2021-01-06:
@avsaase Just thinking that would have to be decided by the main devs because I assume the fact you get the warnings pre arming but not post means it was decided to do it that way previously.

MrD-RC, 2021-01-06:
@breadoven you should 100% make a PR for this. Its really useful functionality. It will also make sat count in programming a useful sanity/error check. 

If you’re not sure about the system message on failure, maybe make that a separate PR if possible? But I think this is a great idea too. Perhaps the devs didn’t see a benefit, maybe it was an oversight? I don’t see how informing on the condition if the model’s system can be bad in any case. Knowing the GPS has failed would allow the model to be flown back safely in a non-navigation flight mode. 

I can’t see how this is anything other than a benefit to all iNav pilots.

avsaase, 2021-01-07:
Would it be possible to apply the same X! indication to elements that rely on gps speed? I'd say that gps speed and derived values like efficiency are immediately useless if the fix is lost or a gps failure occurs and I'd rather have a clear indication of this than a frozen OSD element. Not sure if the internal value used by the LC's should also be changed (zero also doen't make sense), but if the LC's can use the GPS fix explicitly it doesn't make a huge difference. Values that rely on position rather than speed, such as trip distance and distance to home, should stay the same but making them blink would make it clear that they are no longer being updated.

avsaase, 2021-01-07:
I thought this was a good opportunity to see how all the gps related OSD elements behave when satelites are lost, recovered, the gps unit gets disconnected, and reconnected. Here is the dvr from this test, references are in the description and the timeline: https://www.youtube.com/watch?v=mRG-AvdEDdA

My observations:
- The lat lon elements start changing before the gps has locked even one satellite. The same happens in the configurator. These coordinates are completely off until four satellites are locked, at which point the configurator says there is a 3D-fix.
- The plus code only appear when six satellites (`gps_min_sats`) are locked and goes back to dashes when the number of satellites locked drops  below 6. The regular coordinates just start jumping around again.
- When sats < 6 the gps speed (and 3D speed) jumps around so much that it is completely useless and should be discarded.
- Physically disconnecting the gps unit indeed just lets the number of sats blink but keeps the value the same. After reconnecting it takes a while for the satellite count to go to zero and then it start acquiring satellites until it has a fix again.
- After acquiring 6 sats it took a while before I as able to arm. Not sure what that was about.

As you can see there are a couple inconsistencies. I think we can distinguish four gps states and I propose the following behavior for each osd element for each state. When the gps has not acquired a lock since boot then last known values would be replaced with `---`.

| OSD elements                                                | State 1 | State 2                       | State 3                       | State 4                                                  |
|-------------------------------------------------------------|---------|-------------------------------|-------------------------------|----------------------------------------------------------|
| `OSD_GPS_SATS`<br>`OSD_GOS_HDOP`                            |         | blinking value                | blinking value                | blinking `--` or `X!` & <br>`GPS FAILURE` system message |
| `OSD_GPS_SPEED`<br>`OSD_3D_SPEED`                           |         | `---`                         | `---`                         | `---`                                                    |
| `OSD_GPS_LAT`<br>`OSD_GPS_LON`<br>`OSD_PLUS_CODE`           |         | blinking last <br>known value | blinking last <br>known value | blinking last <br>known value                            |
| `OSD_HOME_DIST`<br>`OSD_TRIP_DIST`                          |         | blinking last <br>known value | blinking last <br>known value | blinking last <br>know value                             |
| `OSD_HOME_DIR`<br>`OSD_HOME_HEADING_ERROR`<br>`OSD_HEADING` |         | `-`                           | `-`                           | `-`                                                      |
| other elements                                              |         | `-`                           | `-`                           | `-`                                                      |

WDYT?

MrD-RC, 2021-01-07:
That makes a lot of sense. The testing also backs up what I saw with the data being unreliable quite quickly as the sats decrease.

With the 5th block (OSD_HOME_DIR etc). Perhaps it would be handy do blink between the last known value and **?**. I think this may be more useful for state 3. But it would at least give an indication of the last known home dir, which could be useful if the pilot is flying in a straight line. It would at least give some approximate idea of which way to go.

skorokithakis, 2021-01-07:
I very much agree with your table, I think blinking the last known value will very quickly be more confusing than helpful, since it will be entirely wrong after turning, and you'll have no idea what your original heading was. I'd rather it not show anything than keep showing "northwest" when that's more quickly wrong than right.

avsaase, 2021-01-07:
In theory the gyro should be able to figure out the heading for a little bit after the gps heading is lost. Not sure how long it takes to to drift too much to be of any use.

skorokithakis, 2021-01-07:
If it does, it might be good to show the heading, but I don't know if the gyro is currently used for that.

MrD-RC, 2021-01-07:
That’s a fair comment @skorokithakis. I was just looking for a way to give some kind of assistance. 

For example, if someone is flying at a new site, or is not 100% aware of their surroundings. I just thought it would be helpful in the first instance after the GPS was lost for the pilot to have something to check against. Even if this had a timeout or disappeared when a turn was detected. If they haven’t checked the home dir in a while, then it just vanishes without warning; they could be screwed anyway.

avsaase, 2021-01-07:
Linked above my first attempt at implementing this. I'm not yet able to detect a hardware failure and use is for the osd elements. Feedback welcome.

closed 2022-02-03 by DzikuVx