iNavFlight/inav #6472: flywoo nano BT LE module sending iNav FC to DFU mode

helicocrasher, 2021-01-03:
## Current Behavior
Installed a "flywoo BT nano V2"  (Bluetooth LE module very small) in my 100g wing build to use with the Speedybee app for entering WPs.
As soon as the BT module is connected the matek 722se mini FC, it is going into DFU mode.

Captured the the startup output of the BT nano V2 module on a serial terminal:  BT nano V2 sends the string "+READY"
Sending the same string with the terminal to the serial MSP port used for Bluetoth also sends the FC to DFU.

The issue https://github.com/iNavFlight/inav/issues/1111 in 2017 was dealing with the same behavior.

## Steps to Reproduce

1. Connect the flywoo nano BT V2 module to a serial port with MSP and the FC is going to DFU before finishing bootup
2. FC used MateK F722SE mini. On a Matek F722SE "not mini"- "full size", the effect is not observed.  
3. iNav Version 2.6 and iNav V2.5.2 tested
2.
3.
4.

## Expected behavior

Boot up normally not to DFU 

## Suggested solution(s)

There might be already be a ignore time, to ignore "+Rxxx"  strings on MSP during and shortly after boot up, as on the Full Size Matek F722SE the startup is working. Add such an ignore time or if already there increase it slightly.  

---

- FC Board name and vendor: Matek F722 SE Mini
- INAV version string: 
     # version
     # INAV/MATEKF722SE 2.5.2 Aug  4 2020 / 10:22:04 (faaedc7cb)
    # GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

helicocrasher, 2021-01-11:
Found another solution:
Flywoo tech support sent an AT command to suppress the "+Ready" bootup message of the BT module.

AT+ENLOG0

alexrjohnston, 2021-01-24:
for anyone who wants to use helicocrasher's solution, here is a guide for a similar BT unit:

https://howtomechatronics.com/tutorials/arduino/how-to-configure-pair-two-hc-05-bluetooth-module-master-slave-commands/

RX goes to RX and TX goes to TX!!!!! not the usual RX - TX // TX - RX

Make sure you connect to the flywoo BT at 115200 baud on the arduino serial monitor. and enter AT, the unit will reply with OK to show its working. then enter AT+ENLOG0 and it will reply with OK again. after this it wont put your FC into DFU mode anymore

Flywoo love to send out eletronics that only half work. then you have to jump through hoops to make it useable. how did they not test this? I had the same issue on an F405 board.

closed 2021-04-07 by DzikuVx

d-to-the-ennis, 2021-04-27:
I got in touch with flywoo about this issue, after unsuccessfully trying to send commands via USB TTL with putty. They provided a software and a short instruction to send the required command. This worked for me. Contact flywoofpv@outlook.com for the software.
This is the response that included the software and instructions. I like how they just flat out tell me to buy a new one, when there's no indication on even their page that there's been a revision to the unit.
[image]

Hello there,
Since we mainly focus on the use of Betaflight, this problem will occur in the early BT module under INAV. We have adjusted the BT module and this problem has been solved. I recommend you two solutions to solve the problem.
1. Purchase a new Bluetooth module, which will be more convenient for your use.
2. Connect the Bluetooth module to the USB-TTL module, use the tools in the accessories, and then follow the steps in the picture to solve the problem.
Thank you.