iNavFlight/inav #6474: Default number of sats to 0 if GPS lost

MrD-RC, 2021-01-04:
## Current Behaviour
Currently, if the GPS is lost (hardware/wiring failure) the number of satellites remain at the current value.

## Desired Behaviour
If the GPS is lost, set the number of satellites to 0.

## Who does this impact? Who is this for?
Anyone using the programming functions who use the sats as a sanity check or feature enabler. 

## Additional context
While keeping all the other data from the GPS (coords etc) is extremely useful if the GPS is lost. The same is not true for the satellite count. In fact, if used in the programming to enable switches, it could actually cause issues.

I believe that the sat count should always be kept live, including setting to 0 if the GPS is lost. There is no benefit from keeping this value persistent.

rb1205, 2021-01-04:
Seconded. Locked sat number is used as generic GPS fix validation in logical conditions, radio telemetry, OSD etc. Forcing it to zero when GPS fix is no longer valid would make lots of sense for all those uses.

Christoph-Hofmann, 2021-01-06:
Could someone of the devs check if this https://github.com/Christoph-Hofmann/inav/commit/2b6b2794e192ef30da913d82c880c9a57f77c896
fixes the issue.

PR #6750 (2021-03-29): Add GPS valid fix to programming
This adds a boolean for when there is a valid GPS 3D fix in the programming. Useful for sanity checks. This also sets the number of satellites to 0 if there is a hardware problem, which is more useful than a false value.

Requires https://github.com/iNavFlight/inav-configurator/pull/1206
Fixes #6474
closed 2021-03-29 by DzikuVx