iNavFlight/inav #6476: Support for Emax F4 AIO (Tinyhawk II/S) "F411RX"

HerpDerpula, 2021-01-04:
## Current Behavior
This board is currently supported by BF, but not iNav. Due to how popular this drone is, I believe it should be added. I'd like to see it support full iNav functionality (within reason) including the use of a compass/gps.

## Desired Behavior
Consider the feasibility of porting from BF > iNav and if possible produce a beta firmware for testing.

## Suggested Solution
Depending on the hours needed to make this happen, there **may be a possibility for a "sponsor" for the iNav team**,
 whether it be with boards or funding. 

## Who does this impact? Who is this for?
There's a huge market for these boards. They're great entry level boards with a lot of capability used on a variety of drones such as the Emax Tinyhawk II and Tinyhawk S. Adding support for this board will help a lot of users.

## Additional context

[image]

HerpDerpula, 2021-01-08:
Turns out there's _two_ UART's open, not just one. This allows more GPS/Compass options provided I'm not off base with the boards capabilities. I have a BN-180 connected to UART1 right now, leaving UART2 open. This assumes the user is using SBUS as an input. This thing is tiny, the use case for something like crossfire protocol is low, I'd use SBUS if I changed from FrSky > Crossfire on it.

closed 2021-11-14 by DzikuVx

TIMO88D, 2022-09-16:
Hello,
Did you get this working? 
Id love to try it.

legop3, 2025-02-13:
Hey, did this ever happen? would be awesome.