iNavFlight/inav #6478: About Throttle display

HiroshiKosaka, 2021-01-06:
It may be a problem already mentioned, but I thought that the OSD display of iNAV2.6 was strange, so I will describe it. The Autolaunch mode is set to permanent, and the throttle setting is 80%. It lasts for 20 seconds.
The first PIC was Armed before the flight and raised the throttle to about 60%, but the THRO display remains 0 (iNAV2.5 shows the actual amount of stick raised). The second PIC is running Autolaunch, but the THRO display is '93', which is also different from the actual amount of stick raised (about 60%). Then the timer is not counting up in Autolaunch mode (iNAV2.5 is counting up). The third PIC is the state after transitioning to 3D CRUSE after the Autolaunch mode is over. The timer started counting, but I don't know why the THRO display shows a value close to the Autothro setting. Moreover, it moves freely even though I haven't touched the stick. In iNAV2.5 it was of course constant unless I touched the stick.
[image]
[image]
[image]

breadoven, 2021-01-06:
The flight timer issue is a bug https://github.com/iNavFlight/inav/issues/6427.

Not sure about the throttle.

avsaase, 2021-01-06:
I have not looked at the code yet but I think #6162 may have caused this.

avsaase, 2021-01-11:
I did some testing with a PWM meter and found the following results.
|                              | OSD  | Configurator | PWM  |
|------------------------------|------|--------------|------|
| Armed (throttle down)                       | 4%   | 0%           | 998  |
| Raise throttle               | 20%  | 26%          | 1215 |
| Launch                       | 65%  | 85%          | 1815 |
| Flight (throttle stick down) | 0%   | 0%           | 998  |
| Flight (throttle mid)        | 40%  | 48%          | 1404 |
| Flight (throttle up)         | 100% | 100%         | 1845 |

For the last row the Configurator indicated 100% throttle before the stick was fully up. Relevant settings are:
```
max_throttle = 1850
nav_fw_launch_idle_thr =1200
nav_fw_launch_thr = 1650
throttle_idle =  5.000
```
If I understand correctly, when the esc is calibrated it will take the 1850us as full throttle. But that means auto launch is using almost 100% throttle instead of the 65% indicated on the osd, and what I thought nav_fw_launch_thr = 1650 meant. Is this a bug or do I just misunderstand?

avsaase, 2021-01-11:
Now I have another weird thing: I have the throttle stick around the middle, but the configurator says 100% and I measure 1850us ont he esc signal wire. Something very strange is going on here...

avsaase, 2021-01-11:
Repeating my test from above on 2.5 with the same diff gives these results:
|                              | OSD  | Configurator | PWM  |
|------------------------------|------|--------------|------|
| Armed                        | 0%   | 0%           | 998  |
| Raise throttle               | 20%  | 42%          | 1350 |
| Launch                       | 65%  | 100%         | 1850 |
| Flight (throttle stick down) | 0%   | 0%           | 998  |
| Flight (throttle mid)        | 50%  | 71%          | 1599 |
| Flight (throttle up)         | 100% | 100%         | 1845 |

I have no clue what's going on.

HiroshiKosaka, 2021-01-13:
Since max throttle is 1850 instead of 2000, is 80% of the launch throttle I set calculated as 80/85 and displayed as 93% on the OSD?

MrD-RC, 2021-02-03:
@avsaase  Are your transmitter outputs correctly mapped to 1000 - 2000uS?

From my calculations, they're all wrong.
 PWM | Percentage 
-------| ------------
1215 | 22%
1350 | 35%
1404 | 40%
1599 | 60%
1815 | 82%
1845 | 84%
1850 | 85%

avsaase, 2021-02-03:
@MrD-RC My transmitter has the range 988-2012 which I mapped to 1000-2000 using the `rxrange` command. Can you share your calculation?

MrD-RC, 2021-02-03:
If the range is 1000-2000
PWM - 1000 / 10 = %

The problem here is that you’re reading the PWM from the Tx, but that is mapped to a different range. So the PW value on the Tx isn’t what iNav is using. We’d need to convert that using:

iNavPWM = (((TxPWM - 988) * 1000) / 1024) + 1000
So your 1215 on the Tx (988-2012) would be 1222 in iNav (1000-2000). 1222uS would still be 22%, so maybe it doesn’t make that much of a difference 😄

avsaase, 2021-02-03:
The PWM values I measured are what INAV sends to the ESC, not what it received from the receiver. I should have made that more clear.

MrD-RC, 2021-02-04:
Fair enough. Still, the percentages seem wrong. Unless it is being based on 1000-1850?  

Just tried the calculation. 1215 (1000-1850) would be 1253 (1000-2000), which is 25%. That’s pretty close to the configurator result in your 2.6 tests. 1815 becomes 1959, or 96%. So the numbers seem out there again. Unless there’s a min throttle setting? I will update my online calculator to allow a specified uS range, rather than being set to 1000-2000, which it is currently. 

Anyone know why the max throttle defaults to 1850 btw? Seems odd to me. I always change mine to 2000, then calibrate the ESC.

avsaase, 2021-02-04:
I think it is based on 1000-1850, and everything is just scaled to this range. If you calibrate the esc with this range the esc will also see this as 0-100% so changing it to 2000 doesn't make a difference. 

Anyway, I always use throttle voltage compensation which would effect the PWM signal sent to the esc but _should_ logically leave the osd and configurator throttle percentages unaffected.

MrD-RC, 2021-02-04:
Maybe its the voltage compensation then. As even scaling to 1000-1850, the percentages seem way of in some cases. I've updated the calculator this morning, so it may help.

[image]

https://www.mrd-rc.com/tutorials-tools-and-testing/useful-tools/a-simple-rc-channel-value-conversion-tool/

b14ckyy, 2021-09-10:
I have the same issue on my X-UAV Clouds. And not only autolaunch is affected but also autonomous modes in general. 
Now while I am cutting a new video about the same plane I realized that also my Cruise throttle is always higher in the OSD than it should be. At perfectly level flight the cruise throttle is at 43-45% although it is set to 40% in INAV. So it seems all autothrottle values are boosted by around 15% in my case. And I have no idea why.

Could it be that this is ESC protocol related? need to check other DVRs

closed 2022-02-03 by DzikuVx

b14ckyy, 2022-05-02:
There are still people reporting this issue in 4.0 and 4.1. @DzikuVx 

[image]

DzikuVx, 2022-05-02:
But why anybody says that 93% is wrong? By default PWM width is up to 1850us. So 93% seems to be a correct value

breadoven, 2022-05-02:
It doesn't start at 1000, it starts at the idle value that's been set (default = 15% of 1000 to 1850 = 1128). So % range is 1128 to 1850 = 722. 1800 = 672 / 722 = 93%.

b14ckyy, 2022-05-03:
> But why anybody says that 93% is wrong? By default PWM width is up to 1850us. So 93% seems to be a correct value

But that still makes no sense to me. What happens when I apply manually 2000 throttle with the throttle stick? It will be scaled to 1850 but shown as 100%? But then When I set launch throttle or Cruiose throttle to 2000, the throttle in the OSD shows 115%? 

I don't get the whole 1850us anyway. Why not just use 1000-2000pwm range for receiver like any receiver does when I fly without FC? 

> It doesn't start at 1000, it starts at the idle value that's been set (default = 15% of 1000 to 1850 = 1128). So % range is 1128 to 1850 = 722. 1800 = 672 / 722 = 93%.

that's even worse imho. 

When I set up my plane and I want an idle throttle of 30% for autolaunch and 80% for launch and 40% cruise, then I set the values to 1300, 1800 and 1400. In this case I expect that my ESC get a throttle signal of 1300, 1800 or 1400 when requested and not some random values that are different based on the ESC Protocol or idle throttle or whatever. 

when I fine tune my cruise throttle right now I never know if the throttle value sent to the ESC is what is shown in the OSD or if it is the value set in the cruise throttle variable. And without knowing that I cannot even distinguish in flight if INAV has changed the throttle higher because there is some pitch angle setpoint to climb or if it is just the weird conversion and bad scaling. This totally overcomplicates everything.

DzikuVx, 2022-05-04:
There are 2 THR ranges: `Radio->FC` and `FC->ESC`. All NAV settings are in the second one. To be honest, we should change them to percents or fractions to finally resolve this misunderstanding

MrD-RC, 2022-05-04:
> It doesn't start at 1000, it starts at the idle value that's been set (default = 15% of 1000 to 1850 = 1128). So % range is 1128 to 1850 = 722. 1800 = 672 / 722 = 93%.

But that’s still not scaled. That is just transposed. If it were scaled, it should still be 80%, which is 1706uS in the 1128 - 1850 range.

breadoven, 2022-05-04:
Would make sense to make the settings consistent without mixing pwm values and percentages.

Out of interest where does the 1850 come from ? I thought it was related to the fact that most ESC's are at max output at 1850 so higher PWM's makes no difference. And ESCs also seem to have a dead band at the low end so may not actually start until >1100 perhaps. Did read they have factory default calibrations of typically 1200 - 1800 which will work OK for all RC systems but to fine tune you need to recalibrate it to your RC system using the usual method (stick HIGH on power on then move stick LOW).

It's a bit confusing as to what RC systems actually output for given endpoint settings, Some info seems to say you only get 1000 - 2000 with max endpoints set and something within that range with 100% endpoints set (1150 - 1850 ?). Possibly this is only for Spektrum systems (https://www.rcgroups.com/forums/showpost.php?p=48401983&postcount=34) ? This isn't true of my RC gear. The Radiomaster Zorro outputs pretty much 1000 - 2000 with 100% endpoints and 750 - 2250 at max endpoints. Is this where the 1850 comes from, covering RC systems with strange pwm outputs ?

> But that still makes no sense to me. What happens when I apply manually 2000 throttle with the throttle stick? It will be scaled to 1850 but shown as 100%? But then When I set launch throttle or Cruiose throttle to 2000, the throttle in the OSD shows 115% ? >

The throttle is always constrained to `max_throttle ` (default 1850) regardless of the actual value sent by the Tx. The reason there was a problem with the Launch throttle going over 100% was because this constraint had been omitted in the Launch code (now fixed).
 
> When I set up my plane and I want an idle throttle of 30% for autolaunch and 80% for launch and 40% cruise, then I set the values to 1300, 1800 and 1400. In this case I expect that my ESC get a [...]

The ESC will get the pwm values set regardless. It's the throttle % that will vary depending on the throttle settings used for `max_throttle`, `throttle_idle `and `min_command`.

> _It doesn't start at 1000, it starts at the idle value that's been set (default = 15% of 1000 to 1850 = 1128). So % range is 1128 to 1850 = 722. 1800 = 672 / 722 = 93%._
>
> But that’s still not [...]

This is just what it does in the code to calculate the %: https://github.com/iNavFlight/inav/blob/7cd66149880f127149b28c09b6fa2d221a5ef3fe/src/main/flight/mixer.c#L614

MrD-RC, 2022-05-04:
I'll have to check that out at home. It's impossible to work out what that's doing just on GitHub. I need to see the variables are returning. But, my immediate thought is that `getThrottleIdleValue()` should not be deducted from the rcCommand side of the equation. `PWM_RANGE_MIN` should be idle. So `getThrottleIdleValue` is irrelevant.

breadoven, 2022-05-04:
The current throttle % just shows the % of the throttle stick movement after mixing. Idle throttle needs to be taken into account because it idles even if the throttle stick is low (zero throttle from a user point of view). So I guess it depends on whether or not you think throttle % should show this or the actual pwm output to the ESC, i.e. show 15% throttle with stick low rather than 0% as it does now.

MrD-RC, 2022-05-04:
Sorry, the * 100 in a weird place threw me. I had to think BODMAS. Anyway:

`int16_t thr = (constrain(rcCommand[THROTTLE], PWM_RANGE_MIN, PWM_RANGE_MAX ) - getThrottleIdleValue()) * 100 / (motorConfig()->maxthrottle - getThrottleIdleValue());`

With real values, that would be:
```
  thr = ((constrain(1850, 1000, 2000) - 1128) / (1850 - 1128)) * 100;
∴ thr = ((1800 - 1128) / (1850 - 1128)) * 100
∴ thr = (672 / 722) * 100;
∴ thr = 0.930 * 100
∴ thr = 93
```
The problem is, that is still not scaling. That is transposing. If it were scaling, the % of range A would be the same as the % of range B. You would also never get a % greater than 100%, as the constrain limits the input to within 0-100%. If you do the above equation with 1950 (95%) as the rcCommand, the result is 114%, so it is not being scaled. The RC side of the equation has no reference to the range at all.

The idle side of it, I don't think should be taken on the RC input side. At the bottom of the stick range, it will either be motor off, or idle throttle (for multirotors). That is dealt with on the ESC side of the scaling. There shouldn't be a gap at the bottom of the stick for those 128uS. The whole point should be that the entire range of the throttle stick translated to 0% to 100% at the ESC. Adding the idle throttle loses 128uS from the bottom end, and transposing loses 150us from the top end. So 28% of the stick resolution is lost.

To actually scale, rather than just transpose the value over, it should do something like this. I've taken the idle out of the RC side for simplicity. So to get the percentage
```
  thr = ((constrain(rcCommand[THROTTLE], PWM_RANGE_MIN, PWM_RANGE_MAX) - PWM_RANGE_MIN) / (PWM_RANGE_MAX - PWM_RANGE_MIN)) * 100;
  thr = ((constrain(1850, 1000, 2000) - 1000) / (2000 - 1000)) * 100
∴ thr = ((1800 - 1000) / 1000) * 100
∴ thr = (800 / 1000) * 100
∴ thr = 0.800 * 100
∴ thr = 80
```

Then to get the microsecond value for the output range it would be
```
uint16_t uS = ((motorConfig()->maxthrottle - getThrottleIdleValue()) * (thr / 100)) + getThrottleIdleValue();
∴ uS = ((1850 - 1128) * (80 / 100)) + 1128
∴ uS = (722 * 0.8) + 1128
∴ uS = 577.6 + 1128
∴ uS = 1706
```

1706 (1128-1850) is the scaled microsecond value of 1800 (1000-2000). If we do this calculation again with 1950, the result would be 1814. In both instances the percentage is always 80% or 95% respectively.

b14ckyy, 2022-05-04:
@breadoven 

> Out of interest where does the 1850 come from ? I thought it was related to the fact that most ESC's are at max output at 1850 so higher PWM's makes no difference. And ESCs also seem to have a dead band at the low end so may not actually start until >1100 perhaps.

This is exactly what I don't get. First of all if I calibrate the ESC  then the new range of 1000-1850 will be the new 0-100% power range within the ESC and the "no extra power" in the last 10% will still appear as this happens inside the esc. 

The next thing is, some of my planes have BLHeli ESCs using dshot. There is no calibration and there is no PWM signal but pure percentage. But nevertheless I get a 113% power output on full throttle autolaunch in the OSD. Because still this 1850 default ESC scaling applies although it is totally useless in this case (with dshot). 

remove this 1850us max throttle limit. It has zero advantage imho and just causes trouble. 1000-2000 and the min throttle value just skips the lower values but is not included in the scaling. So when raising the throttle it just jumps to 5% if min throttle is 5%. This makes the most sense, is simple and does not confuse anyone.

breadoven, 2022-05-05:
> The next thing is, some of my planes have BLHeli ESCs using dshot. There is no calibration and there is no PWM signal but pure percentage. But nevertheless I get a 113% power output on full throttle [...]

So you're saying the Launch code fix didn't work when using BLHeli ESCs, i.e. throttle during launch still shows > 100% if max_throttle is set at 1850 and Launch throttle set higher ? Where do these ESCs get their control signal from if not pwm ?

> remove this 1850us max throttle limit. It has zero advantage imho and just causes trouble. 1000-2000 and the min throttle value just skips the lower values but is not included in the scaling. So [...]

The 1850us can be set as required using max_throttle as can the other settings affecting throttle behaviour, throttle_idle, min_command. min_check also affects things because it's used to determine the Throttle Low position which triggers motor stop if set (default = 1100, I use values closer to 1000).

Have to admit I think I've lost track of what the issue is here other than mixing % and us and maybe the fact the current throttle display is confusing for fixed wing use. Perhaps it would be useful to clarify again what the problem is.