iNavFlight/inav #6481: Quadcopter not very stable in ina gps hold

goobisoft, 2021-01-07:
I m using latest inav firmware i think 2.6 and in gps hold mode it does pos hold but it Twitches and wobbles not stable 
Hardware using 
Omnibus f4prov2
Blheli_s 
2212 920kv motors with 10 inch props 
Gps ublox m8n neo 
I m getting good no of satellites more then 10 atleast 
Still wobbles there is less wobble in normal angle mode but when i switch pos hold there is more wobble and twitch

JuliooCesarMDM, 2021-01-07:
an acquaintance of mine has the same problem, the multirotor maintains the position, however when handling the sticks, the quad is unstable and also accelerates upwards.

goobisoft, 2021-01-08:
> an acquaintance of mine has the same problem, the multirotor maintains the position, however when handling the sticks, the quad is unstable and also accelerates upwards.

Can u tell me what position and velocity options does in gps pid tunning

JuliooCesarMDM, 2021-01-08:
default configuration that comes in inav

closed 2022-02-03 by DzikuVx