iNavFlight/inav #6489: New Target Request; HIFIONRC F7 AIO 25A/45A

ltwin8, 2021-01-09:
## Current Behavior
No fitting target found in inav

## Desired Behavior
Please add a target for the board https://www.banggood.com/25_5x25_5mm-HIFIONRC-AIO-F7-Flight-Controller-OSD-Built-in-45A-Blheli_S-2-6S-4-In-1-Brushless-ESC-for-FPV-Racing-Drone-p-1716359.html?cur_warehouse=CN&rmmds=search

I think this board has good potential for inav since it has many useful features, for example:
7 full UARTs
A Barometric pressure sensor
25A or 45A 2-6S capable ESC (with BB2 MCU, G-H-30) (depending on version)
5V 3A SBEC
whoop-sized, 25.5 x 25.5 mm

## Suggested Solution

the resource assignments can be found in here:

https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/HFOR-HIFIONRCF7.config

## Who does this impact? Who is this for?
likely big part of the hobby since sub 250g is more and more important, this board is light (7.5g) and also packs good features, ESCis not only capable for whoops, I rn it on 1900kv 2306 6S 5" "toothpick-style"

## Additional context
I think the above resources are suitable to add the target to the official iNav, if I understand correctly, the target should be very close to the already supported mamba F722 FC

ltwin8, 2021-01-09:
`/*
 * This file is part of INAV.
 *
 * INAV is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * INAV is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with INAV.  If not, see <http://www.gnu.org/licenses/>.
 */

//this is Part of INAV Sourcecode, more exactly target.h for MAMBAF722, modified by LTwin8 to fit HIFIONRCF7 AIO and HIFIONRCF7 PRO

//Used resources are the INAV-Code and the Betaflight unified target file https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/HFOR-HIFIONRCF7.config

#pragma once

#define USE_TARGET_CONFIG

#define TARGET_BOARD_IDENTIFIER         "HFOR"
#define USBD_PRODUCT_STRING             "HIFIONRCF7"

// ******** Board LEDs  **********************
#define LED0                            PA15                            //taken from unified target
//#define LED1                            PC14                          //No 2nd LED defined in unified target

// ******* Beeper ***********
#define BEEPER                          C15                             //taken from unified target
#define BEEPER_INVERTED

// ******* GYRO and ACC ********
#define USE_EXTI
#define GYRO_INT_EXTI                   PC4                             //taken from unified target, additional Interrupt available on C03
#define USE_MPU_DATA_READY_SIGNAL

#define MPU6000_CS_PIN                  SPI1_NSS_PIN                    //B02 & A04, taken from unified target
#define MPU6000_SPI_BUS                 BUS_SPI1

#define USE_IMU_MPU6000
#define IMU_MPU6000_ALIGN               CW90_DEG                        //taken from unified target

// *************** Baro **************************
#define USE_I2C
#define USE_I2C_DEVICE_2

#define I2C2_SCL                        PB6        // SCL pad TX3       //taken from unified target
#define I2C2_SDA                        PB7        // SDA pad RX3       //taken from unified target
#define DEFAULT_I2C_BUS                 BUS_I2C2

#define USE_BARO
#define BARO_I2C_BUS                    DEFAULT_I2C_BUS

#define USE_BARO_BMP280
#define USE_BARO_MS5611

//*********** Magnetometer / Compass *************
//#define USE_MAG
//#define MAG_I2C_BUS                     DEFAULT_I2C_BUS               //taken from unified target, no mag onboard

//#define USE_MAG_HMC5883
//#define USE_MAG_QMC5883
//#define USE_MAG_LIS3MDL

// ******* SERIAL ********
[... 82 lines cut ...]
#define WS2811_PIN                      PA8                             //taken from unified target

// ******* FEATURES ********
#define DEFAULT_RX_FEATURE              FEATURE_RX_SERIAL
#define SERIALRX_UART                   SERIAL_PORT_USART2              //taken from Product Page
#define SERIALRX_PROVIDER               SERIALRX_SBUS

#define DEFAULT_FEATURES                (FEATURE_OSD | FEATURE_TELEMETRY) //no SoftSerial

#define TARGET_IO_PORTA                 0xffff                          //not sure
#define TARGET_IO_PORTB                 0xffff                          //not sure
#define TARGET_IO_PORTC                 0xffff                          //not sure
#define TARGET_IO_PORTD                 (BIT(2))                        //not sure

#define MAX_PWM_OUTPUT_PORTS            8                               //not sure, would guess 4, because its aiming towards quadcopters
#define TARGET_MOTOR_COUNT              4                               //taken from unified target, Motors 1,2,3,4 are located on PB0, PB1, PB4, PB3

// ESC-related features
#define USE_DSHOT
#define USE_SERIALSHOT
#define USE_ESC_SENSOR
#define USE_SERIAL_4WAY_BLHELI_INTERFACE
`

Above is a modified target.h file (taken from MAMBAF722) and modified to fit the HIFIONRCF7 AIO and HIFIONRCF7 PRO

ltwin8, 2021-01-09:
`/*
 * This file is part of INAV.
 *
 * INAV is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * INAV is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with INAV.  If not, see <http://www.gnu.org/licenses/>.
 */

//this is Part of INAV Sourcecode, more exactly target.h for MAMBAF722, modified by LTwin8 to fit HIFIONRCF7 AIO and HIFIONRCF7 PRO

//Used resources are the INAV-Code and the Betaflight unified target file https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/HFOR-HIFIONRCF7.config

#include <stdbool.h>
#include <stdint.h>

#include <platform.h>

#include "common/axis.h"

#include "config/config_master.h"
#include "config/feature.h"

#include "drivers/sensor.h"
#include "drivers/pwm_esc_detect.h"
#include "drivers/pwm_output.h"
#include "drivers/serial.h"

#include "fc/rc_controls.h"

#include "flight/failsafe.h"
#include "flight/mixer.h"
#include "flight/pid.h"

#include "rx/rx.h"

#include "io/serial.h"

#include "sensors/battery.h"
#include "sensors/sensors.h"

#include "telemetry/telemetry.h"

void targetConfiguration(void)
{
   //no hardwired UARTs, no Softserial
}
`

Above is a modified config.c file (taken from MAMBAF722) and modified to fit the HIFIONRCF7 AIO and HIFIONRCF7 PRO

ltwin8, 2021-01-13:
[HIFIONRCF722.zip](https://github.com/iNavFlight/inav/files/5810055/HIFIONRCF722.zip)

Hello, I compiled the file and it seems to work, sadly still not able to test since the hardware is on its way to me from china.
@digitalentity @DzikuVx 

I hope you can Pull this piece into the main code if you approve it.

krizw, 2021-01-13:
> [HIFIONRCF722.zip](https://github.com/iNavFlight/inav/files/5810055/HIFIONRCF722.zip)
>
> Hello, I compiled the file and it seems to work, sadly still not able to test since the hardware is on its way [...]

Where is the hex file for me to try?

ltwin8, 2021-01-13:
> Where is the hex file for me to try?

hello, there you go.
but please, be careful! I have not really programmed iNav before, but I guess it should be okay, Please test the Gyro and motors before flying without props...

even if this does not work one should be able to flash back to Betaflight "Note: STM32 bootloader is stored in ROM, it cannot be bricked."

I am more into hardware than deep dive in iNav Software, but I think it should not harm your hardware (besides if it crashes or so)

maybe you can test without props and let me know

thanks for your help

[inav_2.7.0_HIFIONRCF722.hex.zip](https://github.com/iNavFlight/inav/files/5810234/inav_2.7.0_HIFIONRCF722.hex.zip)

krizw, 2021-01-13:
Hello, I had a very quick go with your Hex, on the bench, the barometer 
was unavailable (RED), the battery voltage read about 1.4v and the DJI 
OSD was not showing. Motors ran and were in the correct order, CRSF 
receiver worked and showed correctly in a tab, GPS worked, Accelerometer 
calibrated ok. Ports tab showed Uarts 1 to 6 ok, but then another Uart 1 
between 5 and 6.

Thank you for your work, I hope we can get an official build eventually. 
I will try any other versions you make.

Cheers

Chris

25A/45A (#6489)

ltwin8, 2021-01-13:
Thanks for testing.

motors ran on dshot?
which UART was the DJI OSD? have you tried all? maybe I mixed something up (for example inav uart2 is board uart5)
- sorry edit while writing: I skip this for now, I remember something that DJI OSD and Inav don't like each other.

Battery not working, OK I thought that would be so, have you tried and adjusted the scale? I had no idea and just set 110 if that not works, what does *Analog* RSSI and current show?
background: VBAT, RSSI, CURR are all ADC Channels, maybe I mixed them up and as soon as you plug the lipo for example RSSI shows strange behavior

CRSF Worked on a UART? numbering correct? 

Baro was unavailable that means the I2C is not working, looking into it.

second UART1 shown, okay, I look into that

ltwin8, 2021-01-13:
Update on Baro: I am looking into the type, I2C Bus should be okay, PB06 as SCL PB07 as SDA

I am looking into #define USE_I2C_PULLUP

attached a new hex, should address I2C

[inav_2.7.0_HIFIONRCF722.hex.zip](https://github.com/iNavFlight/inav/files/5810768/inav_2.7.0_HIFIONRCF722.hex.zip)

ltwin8, 2021-01-13:
Duplicated UART, my bad, sorry, soft serial was still in there...
updated 👍 

[inav_2.7.0_HIFIONRCF722.hex.zip](https://github.com/iNavFlight/inav/files/5810791/inav_2.7.0_HIFIONRCF722.hex.zip)

krizw, 2021-01-13:
Dshot 300 workedDJI osd was on UART 1Crsf was on UART 2ESC telemetry on UART 3BLUETOOTH on UART 4 (forgot to test) MSPGPS on UART 5All connections worked in BF 4.2.4

ltwin8, 2021-01-13:
Okay, shot working should mean timer correct, good thing

Maybe you can try again DJI osd, one UART1 was soft serial with wrong pins in disguise, so maybe in the last uploaded hex it works.

does the blackbox work?

Can you upload fonts to the analog osd? (I know you use digital just to see if the SPI Interface to it works, I think iNav errors out if it can not communicate with the OSD during uploading)

Does the Battery voltage show correct after "set vbat_scale = 1100" ?

so just to keep track the issues
-DJI OSD not working
-VBAT wrong Value
-duplicated UART 
-Baro not working

I will update List when issue is fixed

krizw, 2021-01-13:
DJI osd workingOsd analogue font loadsBatt voltage works with 1100 scaleBaro working ( on bench not flown )Data flash shows data saved ( not checked data content )Good work

ltwin8, 2021-01-13:
Okay, thanks for testing.

If any issues please update me

I can not guarantee, but I think flying should be okay.

krizw, 2021-01-23:
could you define the magnetometer, please

//*********** Magnetometer / Compass *************
//#define USE_MAG
//#define MAG_I2C_BUS DEFAULT_I2C_BUS //taken from unified target, no mag onboard

//#define USE_MAG_HMC5883
//#define USE_MAG_QMC5883
//#define USE_MAG_LIS3MDL

krizw, 2021-01-23:
Could you add the magnetometer, please?

25A/45A (#6489)

ltwin8, 2021-01-23:
Hello,

AFAIK there is no magnetometer on the board?

Maybe add one through I2C?

krizw, 2021-01-23:
I have added one on the I2C, i think drivers required, it's not working as it is.

ltwin8, 2021-01-23:
Okay, i will compile a version for it

ltwin8, 2021-01-23:
I hope this will help you

[inav_2.7.0_HIFIONRCF722.hex.zip](https://github.com/iNavFlight/inav/files/5861062/inav_2.7.0_HIFIONRCF722.hex.zip)

krizw, 2021-01-23:
Obviously, I have no idea, but this is how it is set up on my matek 
f405. Thanks in advance.

// *************** I2C ****************************// SLC clash with 
WS2812 LED#ifdef MATEKF405OSD    // OSD - no native I2C    #define 
USE_I2C    #define USE_I2C_DEVICE_EMULATED    #define 
I2C_DEVICE_EMULATED_SHARES_UART3    #define SOFT_I2C    #define 
SOFT_I2C_SCL            PC10 //TX3 pad    #define SOFT_I2C_SDA 
PC11 //RX3 pad    #define DEFAULT_I2C_BUS         BUS_I2C_EMULATED#else 
// AIO / CTR / STD    #define USE_I2C    #define USE_I2C_DEVICE_1 
#define I2C1_SCL                PB6    #define I2C1_SDA 
PB7    #define DEFAULT_I2C_BUS         BUS_I2C1#endif#define 
USE_BARO#define BARO_I2C_BUS                DEFAULT_I2C_BUS#define 
USE_BARO_BMP280#define USE_BARO_MS5611#define USE_BARO_BMP085#define 
USE_BARO_DPS310#define USE_MAG#define MAG_I2C_BUS 
DEFAULT_I2C_BUS#define USE_MAG_AK8963#define USE_MAG_AK8975#define 
USE_MAG_HMC5883#define USE_MAG_QMC5883#define USE_MAG_IST8310#define 
USE_MAG_IST8308#define USE_MAG_MAG3110#define USE_MAG_LIS3MDL#define 
TEMPERATURE_I2C_BUS         DEFAULT_I2C_BUS#define 
USE_RANGEFINDER#define USE_RANGEFINDER_MSP#define 
USE_RANGEFINDER_HCSR04_I2C#define RANGEFINDER_I2C_BUS 
DEFAULT_I2C_BUS#define PITOT_I2C_BUS               DEFAULT_I2C_BUS

25A/45A (#6489)

krizw, 2021-01-23:
You are fast, wow.

25A/45A (#6489)

ltwin8, 2021-01-23:
i have uncommented 

`
#define MAG_I2C_BUS DEFAULT_I2C_BUS //taken from unified target, no mag onboard

#define USE_MAG_HMC5883
#define USE_MAG_QMC5883
#define USE_MAG_LIS3MDL
`

that should do.

krizw, 2021-01-23:
> I hope this will help you
> 
> [inav_2.7.0_HIFIONRCF722.hex.zip](https://github.com/iNavFlight/inav/files/5861062/inav_2.7.0_HIFIONRCF722.hex.zip)

no improvement.

ltwin8, 2021-01-23:
Oh okay, could you try if your setup works on BF? (the baro itself?)

krizw, 2021-01-23:
> Oh okay, could you try if your setup works on BF? (the baro itself?)

BARO works on both BF and INAV
MAG does not work on either

ltwin8, 2021-01-23:
sorry, I mean the mag, okay, if it not works maybe is the wiring wrong?

krizw, 2021-01-23:
You cannot even get the MAG icon in RED (settings correct, but no data), 
even if my wiring were wrong (2 wires) the MAG icon would be red.

25A/45A (#6489)

ltwin8, 2021-01-23:
i will test it once I have the hardware and mill test where the Pads are connected.

so mag do not work in BF either?

Bolgor, 2021-01-26:
Hello. I had problem witch beitian BN 880 mag on Hifionrc f7 pro on betaflight. I am a beginner but I managed to turn on the magnitometr after resource remaping of pads and few changes in cliI can send the dump file witch my configuration but I dont know IT is correct. Maybe that will help you.

ltwin8, 2021-01-26:
I guess you used UART 3 for that?

Bolgor, 2021-01-26:
Probably. I check it tomorrow.

ltwin8, 2021-01-26:
https://www.st.com/resource/en/datasheet/stm32f723vc.pdf

Page 81/82, so you used i2c capable pins...

done that in my repo for the AIO where UART 3 should be i2c

but I do not know inav well enough to know how to use 2 i2c (internal baro, external mag for example) Busses

Bolgor, 2021-01-27:
After only remapping mag didn't work. I change few options in cli and mag started:

set align_mag = CW270FLIP
set mag_align_roll = 0
set mag_align_pitch = 1800
set mag_align_yaw = 2700
set mag_bustype = I2C
set mag_i2c_device = 2
set mag_i2c_address = 0
set mag_spi_device = 2
set mag_hardware = AUTO
set mag_declination = 0
set mag_calibration = 50,-40,-260
set baro_bustype = I2C
set baro_spi_device = 1
set baro_i2c_device = 1
set baro_i2c_address = 0
set baro_hardware = AUTO

But I don't know that parameters will help you.

krizw, 2021-01-27:
Did the mag show up in red after you remapped, and then blue after CLI changes?

Bolgor, 2021-01-27:
I don't remember about red blinking light (remapped that few months ago) after remapping but maybe this movie will be helpful:
https://youtu.be/coHFI4fsTes

krizw, 2021-01-27:
What remapping did you do, to get the mag working?

25A/45A (#6489)

  https://youtu.be/coHFI4fsTes <https://youtu.be/coHFI4fsTes>

ltwin8, 2021-01-27:
[inav_2.7.0_HIFIONRCF722.hex.zip](https://github.com/iNavFlight/inav/files/5882661/inav_2.7.0_HIFIONRCF722.hex.zip)
 maybe you can try this, SCL on on TX3 and SDA on RX3

LeonardovichD, 2021-01-27:
Hi! I have hifionrc board and want to work it with inav. Could Ihelp you with some tests? I'm  able to do some one by one steps, but i'm noob in programming

ltwin8, 2021-01-27:
can you measure where the SDA and SCL Pads are connected to? which cpu pins

LeonardovichD, 2021-01-27:
> can you measure where the SDA and SCL Pads are connected to? which cpu pins

If you write how to do this)

ltwin8, 2021-01-27:
you take a multimeter with very fine tips and measure in resistance mode where the pads are connected to, continuity testing

LeonardovichD, 2021-01-27:
[image]

LeonardovichD, 2021-01-27:
10pin scl
11pin sda

LeonardovichD, 2021-01-27:
https://drive.google.com/file/d/1Ace4vfIft7m2OiXP0cshKjpYQP5rGQcT/view?usp=drivesdk

Bolgor, 2021-01-27:
I made this changes:
resource MOTOR 5 C09 remapping to resource SERIAL_RX 3 C09 after that resource MOTOR 5 NONE
resource MOTOR 6 C08 remapping to resource SERIAL_TX 3 C08 after that resource MOTOR 6 NONE
resource I2C_SCL 2 NONE remapping to resource I2C_SCL 2 B10
resource I2C_SDA 2 NONE remapping to resource I2C_SDA 2 B11

LeonardovichD, 2021-01-27:
It was helpful?

ltwin8, 2021-01-27:
Pin 10 and pin 11 are needed the uart3 pins regarding to configuration. so an additional bus

ltwin8, 2021-01-27:
thank you, that was helpful

ltwin8, 2021-01-27:
just let me change something, I can not #define use_mag because error.

I hope it is enough to set external i2c bus default and then do it in configurator

ltwin8, 2021-01-27:
[inav_2.7.0_HIFIONRCF722.hex.zip](https://github.com/iNavFlight/inav/files/5883062/inav_2.7.0_HIFIONRCF722.hex.zip)

Default bus set to the Pads (Port B bit 10 and 11), UART3 NOT usable, that are the i2c2 pins
i2c1 (internal, connected to Baro) not any more default

when testing please report back if:

- baro working
- external i2c working (either SDA SCL Pads or uart3 pads (for example for the AIO))

Shothertz, 2021-01-27:
Hi! The pic https://drive.google.com/file/d/1Ace4vfIft7m2OiXP0cshKjpYQP5rGQcT/view?usp=drivesdk is 90° out.
SDA is on pin 59 (PB7) and SCL on pin 58 (PB6)
The internal baro and the external mag runs on the same i2c bus
resource I2C_SCL 1 B06
resource I2C_SDA 1 B07

Set the mag i2c on adresses 0

set mag_bustype = I2C
set mag_i2c_device = 1
**set mag_i2c_address = 0**
set mag_spi_device = 0
set mag_hardware = AUTO
set mag_declination = 0
set mag_calibration = 0,0,0
set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 1
set baro_i2c_address = 0
set baro_hardware = AUTO

Tested and working on mine HIFIONRC F7 PRO V2 + GPS BN880 + mag on BetaFlight

Shothertz, 2021-01-27:
[image]

krizw, 2021-01-27:
> Hi! The pic https://drive.google.com/file/d/1Ace4vfIft7m2OiXP0cshKjpYQP5rGQcT/view?usp=drivesdk is 90° out.
> SDA is on pin 59 (PB7) and SCL on pin 58 (PB6)
> The internal baro and the external mag runs [...]

Yes that's working for me also in Betaflight, now we just need to convert the working settings to INAV, great job.

LeonardovichD, 2021-01-28:
What's next? I need to install hex file?

LeonardovichD, 2021-01-28:
Board does not connected after flashing.
"No configuration received within 10 sec"

LeonardovichD, 2021-01-28:
Upgrade 30.5x30.5mm HIFIONRC F722 Pro V2 F7 OSD Bluetooth Flight Controller w/ 5V 10V BEC & 45A BL_32 3-6S 4in1 ESC Stack for DJI Air Unit / Caddx RC Drone
https://www.banggood.com/Upgrade-30_5x30_5mm-HIFIONRC-F722-Pro-V2-F7-OSD-Bluetooth-Flight-Controller-w-or-5V-10V-BEC-and-45A-BL_32-3-6S-4in1-ESC-Stack-for-DJI-Air-Unit-or-Caddx-RC-Drone-p-1674952.html?cur_warehouse=CN&rmmds=search
Maybe it's because I have another hifionrc. But it will be great if it works)

LeonardovichD, 2021-01-28:
I figured out what was happened: my fc burnt after flashing

ltwin8, 2021-01-28:
how did the FC burn? that should not happen only because of flashing.

Shothertz, 2021-01-28:
Leonardo, I experieced your same problem whit mine FC and I resolved by reconnect 5/6 times the USB ;)

ltwin8, 2021-01-28:
my hardware arrived at customs today, maybe I will be able to test on it within next week

[19 more replies]

closed 2022-02-03 by DzikuVx