iNavFlight/inav #6494: Start and stop recording logic conditions

avsaase, 2021-01-11:
## Current Behavior
When connected via UART, you can start and stop the recording of the Runcam Hybrid HD camera (probably others as well) by selected a channel range in the modes tab for `CAMERA POWER`. However, this is not very convenient because you both start and stop the recording by switching the channel from outside to inside the range. You therefore need to remember if you have already started the recording or not because you cannot see this from the switch position on your transmitter. It also makes it impossible to combine this feature with your arming channel so that you only record when armed. 

## Desired Behavior
Have `START RECORDING` and `STOP RECORDING` logic conditions. This would allow much more flexibility than the current solution via the modes.

## Who does this impact? Who is this for?
Everyone using Runcam cameras with UART control.

goodnerdboy, 2021-01-11:
I put these logic conditions in my taranis, with a "sticky". It should be fairly simple to achieve this in inav with the existing logic conditions if you can overide a channel to send the result to the Hybrid.

To go further you would have to have a feedback from the Hybrid Tx weihter it is recording or not, but i don't think it gives it but in the camera osd.

avsaase, 2021-01-11:
I actually want to combine this on one channel with a camera switcher, where one camera is on a servo that lowers the camera from the bottom of the fuselage. It would be nice if all this was possible in INAV programming instead of having to combine it with OpenTX. But you're right, it is possible to work around this. 

> To go further you would have to have a feedback from the Hybrid Tx weihter it is recording or not, but i don't think it gives it but in the camera osd.

I think a simple counter would suffice. Starting at 0, even numbers means not recording, odd numbers means recording.

closed 2022-02-03 by DzikuVx