iNavFlight/inav #6512: iNav to support out-of-the box presets for -8-10 inch props

Ravid824, 2021-01-15:
Currently the out-of-the box presets for INAV work well for quads with 3-7" props.
The current quad 9" profile behavior is very very bad, easily can results with crashes.
The request is to extend the profiles support for 8-10" props (quad F450, F550 frams).
If I will use 7" profile on F450 quad 9" prop, do you think the behavior will be better?

aa-calvo, 2021-02-02:
I agree, preset for large drones are pretty bad, there should at least be some documentation on tuning to help cope with larger drone

DzikuVx, 2021-02-02:
Such a thing will happen only if someone with such a multirotor will prepare such a preset. From what I know, none of the devs have anything bigger than 7"

closed 2021-02-02 by DzikuVx