iNavFlight/inav #6520: BN-220 GPS module is not working on my F411-WING board. Please help!

AeroStuffFPV, 2021-01-19:
## Current Behavior

Hi all, it's my first time building my FPV plane running iNAV but I encountered a stubborn GPS problem.

So my BN-220 GPS is connected to the UART2 ports on the F411 WING. in INAV 2.6.1, I set the UART2 ports for GPS. 
When I power the board on, it doesn't detect anything from the GPS. The GPS Icon in iNAV is red and the GPS module itself
doesn't flash at all.

I have connected it properly and nothing is wrong with the wires and the RX/TX wires are swapped of course.
(When I put the same GPS with it's wires on a ZOHD Kopilot, it does flash. But when I plug it back in the 411, it doesn't.

Of course, I enabled GPS in INAV and in the configuration tab, I've tried the NMEA, UBLOX & UBLOX7 protocols but they don't work at all to flighten up the GPS.

What is the problem? What can I do to get my GPS working in iNAV so that I see a blue icon and a blue flashing
light on the GPS itself?

[image]

breadoven, 2021-01-20:
I assume you're powering from the flight battery not just the USB connection ?

AeroStuffFPV, 2021-01-20:
> I assume you're powering from the flight battery not just the USB connection ?

- Yes. I tried it with and without flight battery onboard.

fikusir, 2021-01-30:
Can you post the Ports tab and wiring?

cubical23, 2021-02-12:
Check the "GPS" tab, especially, what update rate have? With UBLOX it should be 5hz. If it's 0, it's probably a wiring issue (try swapping RX & TX wires then). Also post the ports tab, as fikusir asked for.

Also BN220 should use UBLOX7 protocol, not UBLOX, so you will get a update rate of 10hz instead of 5hz.

closed 2021-03-02 by DzikuVx