iNavFlight/inav #6522: New Target Request; FLYWOOF745

JyriT, 2021-01-20:
### Current Behavior

No fitting target found in inav

### Desired Behavior

Please add a target for the board https://flywoo.net/collections/goku-16x16-stack/products/goku-hex-f745-fc-16mm-x-16mm-w-4pcs-ws2812-led

I think this board has good potential for inav since it has many useful features, for example:
7 full UARTs
A Barometric pressure sensor
8 Mb blackbox flash
16x16 size, excellent for micro long-range builds

### Suggested Solution

the resource assignments can be found in here:

https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/FLWO-FLYWOOF745.config

### Who does this impact? Who is this for?

likely big part of the hobby since sub 250g is more and more important, this board is light (2.9g) and also packs good features. 

### Additional context

This target enables easy and small implementation of GPS+Compass builds with full Crossfire + Mavlink functionality.

HerpDerpula, 2021-02-04:
Are you willing to test a hex file?

JyriT, 2021-02-04:
Naturally. Table tests are possible at moment, I'm waiting for rx to arrive any moment, then flight tests are also doable.

HerpDerpula, 2021-02-04:
[FLYWOOF745.zip](https://github.com/iNavFlight/inav/files/5927087/FLYWOOF745.zip)
I didn't produce this, I take no credit for it. It's entirely an alpha/beta and not endorsed by the dev team.

drdbuck, 2021-02-20:
I also need this. Thanks.

thezolon, 2021-02-24:
Got another tester here if/when this is done. :) I have the FC in hand and will be swapping it on my hexplorer this weekend. iNAV is one reason I got it.

jdweed, 2021-02-28:
The Flywoo.net web page has a target inav_2.5.2 for this board. I am currently using it with their Goku F745 2-6S 16x16 Stack/Combo ( F745 FC / 35A 4in1 ESC ) and it works fine.

thezolon, 2021-02-28:
Awesome. Will give it a try. Just wish it was in the companion. I get a little twitchy about using firmware that isn't. (I am still a newb and have never used inav yet.)

BenWThomas, 2021-03-11:
Using iNav 2.6.1, flashed HerpDerpula's nav_2.6.0_FlywooF745.hex on new Flywoo F745 Nano, the sweet little 16x16 with builtin programmable LED on each side.
Flashed successfully, selected the smallest prop default (3-5"?  3-7"?), Quad mixer.
After first flash, I reconnected easily without pulling the USB.

The 7 UARTs, MPU6000, and Baro as BMP280 are recognized.
Baro, gyros, accelerometer signals in sensors look pretty normal.

Naturally, first test I tried was the LED Strip feature, since the Nano has 4 built-in.
    These were accessible with FlyWoo's posted 2.5.2 build.
    The builtin LEDs do not seem to be accessible from nav_2.6.0_FlywooF745.hex.

Also, the first time I disconnected USB, when I re-connected, the board didn't boot; the green status LED looked weak and was pulsing slowly.  I pulled USB again, counted off a few seconds, and... it reconnected fine, and everything looks as it did.  Still no LEDs, of course.

Should I expect av_2.6.0_FlywooF745.hex to support the Nano features?
Seems to run hotter than my other one with FW's Nano 2.5.2 build.

Will test some more as time permits.  THANKS, HerpDerpula!

jdweed, 2021-03-11:
I believe the Flywoo F745 Nano has is own hex file - and is not compatible with the FlywooF754.hex. 

It appears that the Flywoo F745Nano is using two (2) separate I2C busses. One seems to be internal to the FC - maybe the barometer - the other is broken out on FC board pads labeled as SDA2 and SCL2.

Flywoo's BNF model Explorer LR 4 V2 HD Micro Long Range uses the F745Nano stack. If you review their CLI download for this model it shows the following:

resource I2C_SCL 1 B06
resource I2C_SDA 1 B07
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

The Betaflight unified target file for the FlywooF745 FC show resources for only one I2C buss.

BenWThomas, 2021-03-12:
@HerpDerpula:  Thanks again VERY much for the binaries.  They seem to behave alike.  What's their lineage?

@JyriT:  Thanks for writing the bug; my #6675 got duped to yours.  VERY happy it's the Nano that gets your attention.
It is so very complete and elegant!

@drdbuck:  We will all be interested in how it flies, and what peripherals may have attached, and how they behave.

@thezolon:  Yes, I totally relate.  I have the Flywoo-posted 2.5.2 on which to fall back, and can assure you that if you download that file, and use iNav "Load from File" and then just flash (don't use popup menus to select board or build, because by loading the file, you've done that ... it will work.  You can always re-flash from the official BetaFlight release, but... I am not a fan, and really, really admire the engineering in the F745Nano and its matching ESCs.  So, it's time to play, and if a little fiddling and feedback can lead to official iNav releases, I will be totally good with that.

@JDWeed:  Thanks very much for your observations about resources... comments/questions follow these INav outputs:

Startup log:   Board self-identifies as FWF7:

2021-03-11 @ 22:28:22 -- Serial port successfully opened with ID: 1
2021-03-11 @ 22:28:22 -- MultiWii API version received - 2.4.0
2021-03-11 @ 22:28:22 -- Flight controller info, identifier: INAV, version: 2.6.0
2021-03-11 @ 22:28:22 -- Running firmware released on: Feb 2 2021 19:30:26
2021-03-11 @ 22:28:22 -- Board: FWF7, version: 0

CLI Resources show neither I2C for 2.6.0 as I have it from :

# Resource
IO:
----------------------
A00: FREE 
A01: FREE 
A02: LED1 OUT
A03: FREE 
A04: FLASH CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: FREE 
A09: FREE 
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: FREE 
B00: MOTOR1 OUT
B01: MOTOR2 OUT
B02: FREE 
B03: FREE 
B04: FREE 
B05: FREE 
B06: I2C1 SCL
B07: I2C1 SDA
B08: FREE 
B09: FREE 
B10: FREE 
B11: FREE 
B12: OSD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: FREE 
C03: ADC CH2
C04: FREE 
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: FREE 
C11: FREE 
C12: FREE 
C13: FREE 
C14: FREE 
C15: FREE 
D00: FREE 
D01: FREE 
D02: FREE 
D03: FREE 
D04: FREE 
D05: FREE 
D06: FREE 
D07: FREE 
D08: FREE 
D09: FREE 
D10: FREE 
D11: FREE 
D12: LED STRIP OUT
D13: FREE 
D14: FREE 
D15: BEEPER OUT
E00: FREE 
E01: MPU EXTI
E02: SPI4 SCK
E03: FREE 
E04: MPU CS
E05: SPI4 MISO
E06: SPI4 MOSI
E07: FREE 
E08: FREE 
E09: MOTOR3 OUT
E10: FREE 
E11: MOTOR4 OUT
E12: FREE 
E13: FREE 
E14: FREE 
E15: FREE 

=>    It will be interesting (once I have chosen peripherals) to see if there's no I2C at all with this firmware.
I may just use one of the itty bitty I2C compass modules I have in my parts box, for now, just to see what works.
(I shouldn't have been surprised that the onboard LEDs work, as they use the same signal as any off-board ones.  They are currently doing nice work as Port and Starboard colors in Larsen scanner mode: very flashy!)
Stacked and mounted on the matched 20A 4-in-1, the FCU doesn't seem any warmer than it should.  Just running USB so far.

=>    But I am in unfamiliar territory with declaring resources, although CLI is not scary.  (And I can always re-flash!)

=>    Can I just apply one (or both) pair(s) of resource commands to this Nano FWF7, and anticipate that I2C-1 and/or I2C-2 might then be accessible?

=>    Peripherals will include:

Bluetooth   Flywoo BT Nano: so small it's almost negative mass, runs at 115000 by default, not my first use of these.

ESC   Definitely the slick 4-in-1 direct-connect Flywoo.  No ribbon, plus 4 more programmable LEDs, and build quality visibly as slick as the FCU.  Not all stacks look as made-for-each-other as these.

GPS-5883   TBD; probably HolyBro Nano just acquired, altho I've not used these before. I do want a compass!

Ranging   Maybe not first priority, but if I2C will work, have generic VL53L0X modules, and an uncommitted Matek Optical Flow on hand.  So far, have VL53L0X has been pretty limited; have to be within 1m or less over grass in shade before it seems to be reliable. 

Comment:  very glad to see FCUs with Baro on board re-appearing.  Wish onboard compass would re-emerge; my best compass implementations to date are Sparky (original) and SPRacingF3Mini, which defy logic by appearing to work better than external GPS/Mag, for compass function.

=>   Actual soldering may not begin for a couple weeks -- daughter just today delivered her first baby/our first grandchild a few hours' drive away.   Strange territory; new life component installed...

jcink, 2021-03-19:
I would also like to express interest in an official iNav target for this board. Flywoo ships these with their V2 digital setup and with an official target it would pave the path to use iNav with it. :)

surfek, 2021-03-22:
Also looking for the INAV target for this FC (got this with Flywoo Explorer V2). Would be great top use INAV :)
Betaflight target is called: FLYWOO F745 NANO.

Is there any test INAV target I can use?

BenWThomas, 2021-03-24:
See HerpDerpula's comment, Feb 4.  Hex file is for F745 not F745Nano.
As it is, it works except for I2C, which is broken out (as I understand it) on the F745 Nano as I2C2, rather than I2C1.
Not sure if this can be fiddled with CLI resource-feature-whatever or not...

BenWThomas, 2021-03-24:
And... if iNav 2.5.2 will do for you, Flywoo has iNav 2.5.2 hex file posted on their Support/ Downloads page.
This seems to work fine.

Herpdula's F745 not-Nano files are 2.6 and 2.7 - not sure if there's anything apart from I2C that isn't compatible, but it burns and runs.  F745 and F745 hardware are similar, but not identical.

BenWThomas, 2021-03-24:
Flywoo iNav 2.5.2 is last item on the list, here:   https://flywoo.net/pages/manual

surfek, 2021-03-24:
Thanks. will use 2.5.2, hope Goku F745 Nano will be included into INAV target lists.
It is default on Flywoo Explorer, and believe it makes sense to use INAV instead of Betaflight there...

thivi87, 2021-03-29:
@HerpDerpula - is this hex for the F745, or the F745nano specifically?  I am trying to run a GOKU GN 745 40A AIO with GPS and mag and would rather run iNAV on this one.

Bibbleq, 2021-03-31:
Flashed the 2.6.0 version from above ZIP successfully on the GOKU HD F745 AIO target, not got motors and things to test with but gyro and baro reporting in okay at the moment

HerpDerpula, 2021-03-31:
> @HerpDerpula - is this hex for the F745, or the F745nano specifically? I am trying to run a GOKU GN 745 40A AIO with GPS and mag and would rather run iNAV on this one.

F745 I believe. I did not create this hex file. It was compiled by a dev who wanted to help out the community. It's entirely up to them if they want to speak up and offer more information. I'm just a messenger - no need to thank me. Just someone who appreciates the time the community puts into moving the ball forward for the little guys like me who enjoy flying.

stronnag, 2021-03-31:
> F745 I believe. I did not create this hex file. It was compiled by a dev who wanted to help out the community. It's entirely up to them if they want to speak up and offer more information. I'm just [...]

Specifically, the target files came from the manufacturer via a RCG user. They are attached below.
[flywoof745.tar.gz](https://github.com/iNavFlight/inav/files/6238881/flywoof745.tar.gz)

BenWThomas, 2021-04-06:
@stronnag. Although I haven't flown with the F745 2.6/2.7 hex files, and have F745Nano stacks, I've done a good bit on the bench.  As has been pointed out, on Nano it's I2C2 that's pinned out on the Nano hardware, and this hex file for 'regular' F745 expects peripherals on I2C1 (did I get that right, fellas?).   Serial ports (7 on the Nano!) seem to work for the several configs I've built up.  Soooo...  the question may be I2C on your board.  You should be able to figure out which it uses by burning released BetaFlight, and collecting status (and a DIFF to look at).  Status will show which I2C bus is enabled on your board.

BenWThomas, 2021-04-06:
Matek has/had a GPS with MSP MN9 GPS/compass -and- has a Baro on board - so you could use it via UART and get your compass and a Baro, too without I2C.  Matek discontinued it, stating on website that it was because they couldn't get sufficient volume of a component.  It is kinda big, as it has CAN and 2 other connectors.  Some may still be available.  I have not tested this device, but iNav offers MSP all sensors, so ... with a well-behaved UART, you won't need I2C for a full config, and you'd be getting a lot of mileage out of one UART.

BenWThomas, 2021-04-18:
Flywoo.net has just posted iNav 3.0.0.hex for their entire lineup.
flywoo.net/support/downloads

jimsynz, 2021-04-18:
It'd be nice if their devs would open a PR with their changes.

harry1453, 2021-04-26:
They are hosting INAV 3.0 builds here: https://cdn.shopifycdn.net/s/files/1/0010/7410/2324/files/INAV_3_0_0_Firmware.rar?v=1618564404, no clear indication of what "3.0.0" it is, if you run it you could probably find a commit hash in the CLI.

They seem to be hosting files for targets that INAV doesn't have, including FLYWOOF745 and FLYWOOF411HEX. I can't find source code anywhere.

**This is an extremely clear GPL violation.** Might be worth contacting them to point this out.

jimsynz, 2021-04-26:
It no one else has done so already I'm happy to reach out to their support channel and request it.

jdweed, 2021-04-27:
The flywoo.net website has the Inav 3.0 hex files listed in their support download section. Looks like these Inav 3.0 hex files referenced are from the manufacturer.

BenWThomas, 2021-04-28:
Flywoo.net iNav 3.0.0.hex for F745Nano functions completely as expected on bench, including I2C (I2C2) compass on GPS, which would not work with Flywoo's 2.5.2 F745 Nano firmware. Great news!

Not ready to fly yet, but -everything- I've wired and configured seems to work... except, weirdly,  Flywoo's own BTNano.
Tiny 192kbps BT board sets works great, until you remove and reapply power.  F745N powers up and drops to DFU.  Have repeated on 2 F745 Nano stacks on various serial ports, at 3.3v and 5v.   Generic JDY-18 (9600 bps) works fine.  The FW BT Nano works fine at default 192k with iNav 2.6.1 or 3.0.0 pre-release on another mfg's FCU, altho I forget which one I tested...

Anybody flown F745 Nano with this Flywoo 3.0.0 firmware yet?

JyriT, 2021-05-01:
Started testing the Flywoo inav3.0 hex file with F745 fc. 
Seems that compass connected to SDA2/SCL2 doesn't turn green. GPS & Baro is working fine. 
# version 
# INAV/FLYWOOF745 3.0.0 Apr 15 2021 / 21:10:19 (7b50b059)
# GCC-9.3.1 20200408 (release)

# status
System Uptime: 55 seconds
Current Time: 2041-06-28T02:04:00.000+01:00
Voltage: 12.24V (3S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=UNAVAILABLE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
Stack size: 6144, Stack address: 0x20010000, Heap available: 1820
I2C Errors: 60, config size: 7059, max available config: 32768
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = none
  AIRSPEED : configured = none, used = none
System load: 3, cycle time: 539, PID rate: 1855, RX rate: 49, System rate: 9
Arming disabled flags: ACC HWFAIL RX CLI
VTX: not detected

JyriT, 2021-05-01:
Update. 
Changed Mag to auto. Now it's detected.

kingasa47, 2021-05-07:
Just got my flywoo f745 will the 3.0 flywoo firmware work on inav 2.6.1? sorry I'm new to this thanks and bear with me

harry1453, 2021-05-07:
@kingasa47 The 3.0 flywoo firmware is "INAV 3.0" (in quotes because INAV 3.0 isn't actually released yet so it's actually just a development build), so it won't be 2.6.1 at all. No idea if it'll work, others say it does.

kingasa47, 2021-05-13:
Ready to flash the firmware confused as to which one to flash for my flywoo explorer lr v1
FLYWOO745.hex
FLYWOO745.HD.hex
or FYWOO745.NANO.hex
thanks for the help!

harry1453, 2021-05-16:
Unfortunately it's hard to know, as Flywoo haven't released details of which target does what. What FC do you have? Do you have a link to the Flywoo store for that FC?

jimsynz, 2021-05-16:
FYI I reached out to Flywoo support asking for them to honour the GPL license, but have not yet heard anything back.

harry1453, 2021-05-17:
@jimsynz Thanks, hopefully they do something about it...

jimsynz, 2021-05-18:
https://github.com/iNavFlight/inav/pull/6985

BenWThomas, 2021-05-18:
@kingasa47. --  If your Flywoo Explorer has a 16x16 FCU, it is the Nano.  (I am in process of building up a 4" Explorer LR - beautifully engineered frame!). Flywoo has a link to an iNav 3.0.0 Configurator on their Support page, and if you use iNav app to run iNav Configurator, you will see that pre-releases now include official 3.0.0-rc1, which is newer than Flywoo's posted version.

Also ordered several of Flywoo's tiny new I2C compass-baro modules, for use with certain FCUs that don't have either.

kingasa47, 2021-05-19:
> @kingasa47. -- If your Flywoo Explorer has a 16x16 FCU, it is the Nano. (I am in process of building up a 4" Explorer LR - beautifully engineered frame!). Flywoo has a link to an iNav 3.0.0 [...]
Where do I download 3.0.0-rc1?

BenWThomas, 2021-05-20:
@kingasa47 --  

If you use a Mac, the best answer is really cool:  https://inavapp.com.  iNav.app is a curating front-end for iNav users; it can launch -any- release of iNav from 1.7.3 to 2.6.1 plus prereleases through 3.0.0 rc-1.  This is incredibly useful if you have a mix of older and newer FCUs.  iNav.app will also launch BLheli, and Black Box Log Viewer.  (The same author also has a similar front-end app for BetaFlight.)

iNav.app may be a Mac-only thing; its developer is working with Apple to get it into the App Store.

Also, Flywoo.net lists a link for iNav 3.0.0 Configurator on its Support page.  The link points to a server in UK that hosts iNav rc-1 for all platforms.  3.0.0 rc-1 for Mac and Windows are unsigned (unofficial); the Linux 32 and 64-bit downloads here are are signed by the developer.

Flywoo.net's post may (or may not) be identical to the rc-1 available via iNav.app; I haven't checked that.
iNav.app always offers the entire range of releases.  (I have not checked to see if the Mac rc-1 via iNav.app is signed, or not.)

For Windows or Linux, you can (and may prefer) to download iNav Configurator 3.0.0 RC1 directly from Github:  https://github.com/iNavFlight/inav-configurator/releases/tag/3.0.0-RC1.
The Mac Configurator usually appears with he other official releases a few weeks later.

BenWThomas, 2021-05-24:
Unsupported iNav 3.0.0 F745nano worked very well on maiden flight of build of Explorer LR 4.  3" defaults need some PID tuning, FCU with NIN1404 w/4024 agile and very smooth.  iBus Fli14+ Rx, Holybro GPS/compass worked great. JDY-18 BlueTooth connects fine to SpeedyBee app.

1-handed movie of maiden hover here:  https://github.com/iNavFlight/inav/issues/6675

d4rken, 2021-05-24:
There is no support for the F745HEX right? For the hexplorer? None of the other F745 images seem to work.

harry1453, 2021-05-24:
@d4rken Flywoo have added support unofficially, and are working to get official support implemented. Please see above conversation. I'm pretty sure the F745HEX target is in their zip file, but please bear in mind that it's not an official build, nor is it even a release build, it's taken from some point in 3.0's development so won't include any bugfixes that were done between when it was created and when 3.0 was released.

gabrielgradinaru, 2021-05-29:
I has been merged boys: https://github.com/iNavFlight/inav/pull/6985
It is now available in 3.0.0-RC2: https://github.com/iNavFlight/inav/releases/tag/3.0.0-RC2

d4rken, 2021-06-03:
> Please see above conversation. I'm pretty sure the F745HEX target is in their zip file, but please bear in mind that it's not an official build, nor is it even a release build, it's taken from some [...]

The zip only contains a F4 HEX variant.

> It has been merged boys: #6985
> Now available in 3.0.0-RC2: https://github.com/iNavFlight/inav/releases/tag/3.0.0-RC2
> I suggest closing this issue now.

No F745HEX :frowning:

adragoset, 2021-06-07:
No F411 hex either. Looks like the soft serial port is not mapped properly on that board to pins PB0 and PB4. And since there is no resource remapping in inav i cant use one of those pads to control smart audio in my fixed wing plane with gps and compass.

gabrielgradinaru, 2021-06-07:
It looks like there is no special target for F745HEX, it is just the regular F745. More info here:
https://www.youtube.com/watch?v=6Zy7uyBVkHg&t=2680s

BenWThomas, 2021-06-09:
Yes, I checked just now, and the list of supported builds includes these from Flywoo:

inav_3.0.0_FLYWOOF411.hex
 <https://github.com/iNavFlight/inav/releases/download/3.0.0-RC2/inav_3.0.0_FLYWOOF411.hex>1.3 MB
inav_3.0.0_FLYWOOF411_V2.hex
 <https://github.com/iNavFlight/inav/releases/download/3.0.0-RC2/inav_3.0.0_FLYWOOF411_V2.hex>1.3 MB
inav_3.0.0_FLYWOOF745.hex
 <https://github.com/iNavFlight/inav/releases/download/3.0.0-RC2/inav_3.0.0_FLYWOOF745.hex>1.34 MB
inav_3.0.0_FLYWOOF745NANO.hex
 <https://github.com/iNavFlight/inav/releases/download/3.0.0-RC2/inav_3.0.0_FLYWOOF745NANO.hex>1.34 MB
inav_3.0.0_FLYWOOF7DUAL.hex
 <https://github.com/iNavFlight/inav/releases/download/3.0.0-RC2/inav_3.0.0_FLYWOOF7DUAL.hex>1.18 MB

Hooray!

d4rken, 2021-06-13:
> It looks like there is no special target for F745HEX, it is just the regular F745. More info here:
> https://www.youtube.com/watch?v=6Zy7uyBVkHg&t=2680s

Can't connect to the FC after flashing the regular F745 :frowning:

BenWThomas, 2021-06-17:
iNav 3.0.0 official release was posted a few days ago, and FlywooF745Nano is now supported.

    https://github.com/iNavFlight/inav-configurator/releases/tag/3.0.0

The two F745 Nano stacks I have reflashed are behaving very well with I2C compass, and UART-connected IRC, GPS, BT, and iBus Rx.  Very smooth.  2.6.1 backups made with CLI 'diff all' restored on fully-erased FCU behaved as expected in front-yard test flights.  Restore included accel and mag calibration.  (You will still have to choose a basic model-type and mixer settings before you restore.)

Hooray!

elbartnox, 2021-10-29:
> > Sembra che non ci sia un obiettivo speciale per F745HEX, è solo il normale F745. Maggiori informazioni qui: 
> > https://www.youtube.com/watch?v=6Zy7uyBVkHg&t=2680s
> 
> Impossibile connettersi all'FC dopo aver lampeggiato il normale F745 ?

Please reopen this thread.
As the title suggests, we should talk about the FC F745HEX, not Nano..Dual Etc ..
I confirm that for the FC F745HEX (hexacopter) the Firmware of the F745 (quad) does not work.
After the flash it is impossible to connect to the FC.
Tested Official Firmware and that of the Flywoo website.