iNavFlight/inav #6523: Auto repeat functionality for Waypoint missions.

jonasbouchraiet, 2021-01-22:
Want to automatically repeat waypoint missions for battery testing. For example: Create a circle out of 5 Waypoints and when the plane reaches Waypoint 5 it automatically goes to Waypoint 1 and repeats this until waypoint mode is disengaged.

I know theres a Jump function in EZ GUI but it doesn't do what I want.

stronnag, 2021-01-22:
That is easily doable with JUMP (inav 2.5 or later).

stronnag, 2021-01-22:
here's an example what does what you describe:
```
<?xml version="1.0" encoding="utf-8"?>
<mission>
  <!--mw planner 0.01-->
  <version value="2.3-pre8"></version>
  <mwp save-date="2021-01-22T07:08:21+0000" zoom="16" cx="-4.5257664" cy="54.3543680" generator="mwp (mwptools)"></mwp>
  <missionitem no="1" action="WAYPOINT" lat="54.3566492" lon="-4.5202802" alt="50" parameter1="0" parameter2="0" parameter3="0"></missionitem>
  <missionitem no="2" action="WAYPOINT" lat="54.3550945" lon="-4.5166085" alt="50" parameter1="0" parameter2="0" parameter3="0"></missionitem>
  <missionitem no="3" action="WAYPOINT" lat="54.3525792" lon="-4.5180111" alt="50" parameter1="0" parameter2="0" parameter3="0"></missionitem>
  <missionitem no="4" action="WAYPOINT" lat="54.3525792" lon="-4.5225492" alt="50" parameter1="0" parameter2="0" parameter3="0"></missionitem>
  <missionitem no="5" action="WAYPOINT" lat="54.3550945" lon="-4.5239518" alt="50" parameter1="0" parameter2="0" parameter3="0"></missionitem>
  <missionitem no="6" action="JUMP" lat="0.0000000" lon="0.0000000" alt="0" parameter1="1" parameter2="-1" parameter3="0"></missionitem>
</mission>
```

[image]

stronnag, 2021-01-23:
As there is no indication that JUMP is not worked as documented, I'm closing this issue.
Please reopen it if you have evidence to the contrary (for example. a misison files and blackbox log showing some unexpected JUMP behaviour).

closed 2021-01-23 by stronnag

jonasbouchraiet, 2021-01-24:
Is there a way to use the jump function in the INAV Configurator. Paste CLI?

jonasbouchraiet, 2021-01-24:
Or can you recommend a mission planner for mac where I can use jump?

stronnag, 2021-01-24:
## CLI

The CLI `wp` command, as described in the [project documentation](https://github.com/iNavFlight/inav/blob/master/docs/Navigation.md#cli-command-wp-to-manage-waypoints)

## MacOS

qgoundcontrol, apmplanner and [impload](https://github.com/stronnag/impload), or a virtual machine to run [mwp](https://github.com/stronnag/mwptools) or dronehelper.

jonasbouchraiet, 2021-01-24:
OK. Now I got it. Thank you so much for the very fast responses, it means a lot. And thank you for the recommendations.

jonasbouchraiet, 2021-01-26:
> here's an example what does what you describe:
> 
> ```
> <?xml version="1.0" encoding="utf-8"?>
> <mission>
>   
>   <version value="2.3-pre8"></version>
>   <mwp save-date="2021-01-22T07:08:21+0000" zoom="16" cx="-4.5257664" cy="54.3543680" generator="mwp (mwptools)"></mwp>
>   <missionitem no="1" action="WAYPOINT" lat="54.3566492" lon="-4.5202802" alt="50" parameter1="0" parameter2="0" parameter3="0"></missionitem>
>   <missionitem no="2" action="WAYPOINT" lat="54.3550945" lon="-4.5166085" alt="50" parameter1="0" parameter2="0" parameter3="0"></missionitem>
>   <missionitem no="3" action="WAYPOINT" lat="54.3525792" lon="-4.5180111" alt="50" parameter1="0" parameter2="0" parameter3="0"></missionitem>
>   <missionitem no="4" action="WAYPOINT" lat="54.3525792" lon="-4.5225492" alt="50" parameter1="0" parameter2="0" parameter3="0"></missionitem>
>   <missionitem no="5" action="WAYPOINT" lat="54.3550945" lon="-4.5239518" alt="50" parameter1="0" parameter2="0" parameter3="0"></missionitem>
>   <missionitem no="6" action="JUMP" lat="0.0000000" lon="0.0000000" alt="0" parameter1="1" parameter2="-1" parameter3="0"></missionitem>
> </mission>
> ```
> 
> [image]

Let me know if there are any mistakes in my Inav CLI; When reaching WP 14 it should jump to WP 1 again and repeat it for 100 times. I put the 165 end "flag" to the jump item, is that right or should i add it to the last Waypoint? And is there a value for p2 to iterate infinitely?

wp 0 1 473262367 84667233 7500 0 0 0 0
wp 1 1 473312941 84667334 7500 0 0 0 0
wp 2 1 473325224 84660616 7500 0 0 0 0
wp 3 1 473332957 84648351 7500 0 0 0 0
wp 4 1 473332874 84628639 7500 0 0 0 0
wp 5 1 473325464 84614358 7500 0 0 0 0
wp 6 1 473313502 84610561 7500 0 0 0 0
wp 7 1 473292055 84612766 7500 0 0 0 0
wp 8 1 473228600 84591677 7500 0 0 0 0
wp 9 1 473213640 84599699 7500 0 0 0 0
wp 10 1 473206193 84619673 7500 0 0 0 0
wp 11 1 473207501 84641396 7500 0 0 0 0
wp 12 1 473215799 84657139 7500 0 0 0 0
wp 13 1 473229355 84662138 7500 0 0 0 0
wp 14 6 0 0 0 0 100 0 165

stronnag, 2021-01-26:
> et me know if there are any mistakes in my Inav CLI; When reaching WP 14 it should jump to WP 1 again and repeat it for 100 times. I put the 165 end "flag" to the jump item, is that right or should i add it to the last Waypoint? And is there a value for p2 to iterate infinitely?

> wp 14 6 0 0 0 0 100 0 165

That's fine. The 165 should be the last item in the mission. You can use [-1 for infinite](https://github.com/iNavFlight/inav/wiki/MSP-Navigation-Messages#waypoint-and-action-attributes).

[image]

jonasbouchraiet, 2021-01-26:
Perfect. Will try today. Thanks for helping, thats aweome