iNavFlight/inav #6527: Support CX-OF optical flow module with yaw mode pin

Sebbb, 2021-01-23:
I bought the optical flow module https://www.aliexpress.com/item/33005508643.html?spm=a2g0s.9042311.0.0.27424c4dQOLL8p

This is compatible with the CX-OF protocol and works fine. Also, it has an additional pin for "yaw mode":

> Special note about Y pin:
> ①This pin is the yaw mode selection port. It is active low and defaults to high (normal mode) at startup.
> ②When the drone is in yaw motion, it is recommended to pull down the Y pin to make the module enter yaw mode.
> ③When the sensor is not in yaw motion, pull the Y pin high to put the module into normal mode.
> ④If the module does not turn on yaw mode during yaw motion, it may cause jitter and drift. It can be improved after turning on the yaw mode.
> ⑤If the module is in yaw mode when the drone is not in yaw motion, the linearity of the module data may be deteriorated. 

In yaw mode, the footer changes from 0xAA to 0xBB.

Example:

```
FE 04 00 00 00 00 00 1D AA    - normal mode
FE 04 01 00 00 00 01 1E AA    - normal mode
FE 04 00 00 00 00 00 1C BB    - yaw mode
FE 04 01 00 00 00 01 1B BB    - yaw mode

```

## Current Behavior
When in yaw mode, the drone starts beeping because it detects the optical flow module as down (displayed red in inav-configurator)

## Desired Behavior
Input from the module should not be considered, but the module should not be detected as down (no beeping)

## Suggested Solution
adjust main/io/opflow_cxof.c to not completely ignore packets with footer 0xBB. Probably the fix is really easy (set newPacket to true but don't adjust flowRateRaw?)

## Who does this impact? Who is this for?
People using this optical flow module

Sebbb, 2021-01-23:
This seems to work for me, it's probably not the best solution though - I'm not a coder..

```
diff --git a/src/main/io/opflow_cxof.c b/src/main/io/opflow_cxof.c
index 55e7a55d9..8adc26adf 100755
--- a/src/main/io/opflow_cxof.c
+++ b/src/main/io/opflow_cxof.c
@@ -116,6 +116,12 @@ static bool cxofOpflowUpdate(opflowData_t * data)
                     tmpData.flowRateRaw[2] = 0;
                     tmpData.quality = (constrain(pkt->squal, 64, 78) - 64) * 100 / 14;
                     previousTimeUs = currentTimeUs;
                     newPacket = true;
+                } else if (pkt->header == 0xFE && pkt->footer == 0xBB) {
+                    // Valid packet, yaw mode
+                    tmpData.deltaTime += (currentTimeUs - previousTimeUs);
+                    tmpData.quality = 0;
+                    previousTimeUs = currentTimeUs;
+                    newPacket = true;
                 }

```

int-smart, 2021-02-18:
Hi

How did you configure this sensor with INAV. I have a KAKUTEF7 with the optical sensor as above and a TFMini Plus. I am trying to configure this with INAV but am failing to do so. I have selected the correct optical flow hardware as well as the assigned it to the correct UART2 in my flight controller. I can recognize the TFMini Plus but the configurator fails to recognise the optical flow module.

Thanks for the help.

Sebbb, 2021-02-18:
No, it just worked out of the box. The only thing I'm struggling with is to see if it acutally helps positioning. I see debug data and it's aligned according to http://www.mateksys.com/downloads/setup_optic%20flow_in_INAV_rev1.pdf, but somehow my Quadcopter still moves several meters around.

Did you try if you see data output from your module by using serialpassthrough? I did that with hterm to check the output bytes.

int-smart, 2021-02-18:
The document above is for Mateksys module which works on MSP protocol. Did you go through the serial port configuring steps as described in the manual or did the steps I described above are fine. 
With regards to the debugging using serial data output, I will tear down my drone to get access to the Transmitter pin for the board.

Sebbb, 2021-02-18:
I know, but the steps for setting debug mode and aligning are be the same, regardless what module.

For the serial port, I set Sensors to "optical flow", baud rate 115200 (was already set) and chose CX-OF. That's it. Try the serialpassthrough also with 115200, that worked for me.

int-smart, 2021-02-18:
I suppose my sensor is bad as I dont see any readings. Thanks for the quick responses.

closed 2022-02-03 by DzikuVx

mxchinist, 2023-09-18:
> I suppose my sensor is bad as I dont see any readings. Thanks for the quick responses.

I've also had trouble making this exact sensor (Cheerson CXOF) work, making the "flow" icon in inav blue.
Check that your sensor is connected correctly to 3.3V, GND and RX(number) pad on your FC (my board is outputing 3.25V, which works as well. sensor is rated for 3.3-3.6V form what I've found)
Then in ports tab, select in "sensors" column "Optic flow" and change the number after it from 115200 to 19200. You have to do this in correct UART(number) row, where you've connected your sensor. All the other settings needs to be off (MSP, RX).
Last step is to choose CXOF in configurations tab in "sensors & busses" window.
I also pasted this command into CLI to see optical flow sensor inputs in sensors tab:
set debug_mode = FLOW_RAW
save
You should also calibrate the optical flow sensor before flying in calibration tab, it should be the same as in this video:
https://youtu.be/XUncioDjs1Q?t=460
Hope this helps !