iNavFlight/inav #6530: AKK ultra long range vtx - smartaudio - power level 5 not working

danielolsman, 2021-01-27:
## Current Behavior
When selecting the 5th power level via OSD or via the configurator it reverts back to the previous selected power output.

## Steps to Reproduce
Power level 1,2,3 and 4 are working fine when selected via OSD or configurator.
VTX reports back the current selected power level via smartaudio 
When selecting the next power level from  the 4th level via OSD a blank space is shown (see screenshot)

[image]

## Expected behavior
All power levels (1 to 5) should be available 

## Suggested solution(s)
-

## Additional context

https://www.akktek.com/ultra-long-range-vtx.html

https://pastebin.com/XGhTmWVw

---
1. # version
2.  
3. # INAV/MATEKF722WPX 2.6.0 Dec  4 2020 / 12:19:14 (cada68e4)
4.  
5. # GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] 

- FC Board name and vendor: Matek F722 WPX
- INAV version string: 2.6.0

Jetrell, 2021-01-29:
This VTX's highest power level will not work with iNav because the Smart Audio 2.1 addition to iNav was based on TBS hardware.  Which only have 4 power levels - 0 1 2 3 
 * 25 - 200 - 500 - 1000
 * 25 - 200 - 500 - 800
 * 25 - 100 - 200 - 400
 

I'm not sure where AKK's addition to the protocol comes from.  Tramp has 5 power levels.

There is a work around to use the highest power level of the Tramp VTX protocol, if it doesn't read correctly. `vtx_max_power_override`
But I'm unsure if it works with SA. And it does limit access to the lower power levels.

danielolsman, 2021-02-02:
vtx_max_power_override doesn't do anything unfortunately.
isn't there a way to manually override the max powerlevels?

expipiplus1, 2021-02-15:
@danielolsman I would suggest that you contact AKK directly to get some specifications on the protocol they're using.

danielolsman, 2021-02-16:
afaik there using smartaudio, for BF there are VTX tables for this VTX https://docs.google.com/document/d/1B97Ol2UQOWj8A2ZzzsRcIU3JlOwlX6hYAzeutjb6c1E/edit

it's only the 5th powerlevel i can't select

Jetrell, 2021-02-16:
> 
> 
> afaik there using smartaudio, for BF there are VTX tables for this VTX 
> 
> it's only the 5th powerlevel i can't select

As you said, BF uses VTX tables. They are user configurable. Adding a 5th power level for SA isn't a problem. While the iNav dev's chose not to go that way, due to increased complexity. And little need except in cases like this.

danielolsman, 2021-02-19:
I have got a response from AKK:

> Hello Daniël,
> 
> Just like Jetrell said, the iNav devs are unwilling to add a 5th power level option, there is nothing we can do, so I would suggest using SA.
> 
> Best regards
> Lisa
> Email: lisa@akktek.com

Jetrell, 2021-02-20:
@danielolsman Its not a case of iNav dev's, being unwilling to add support for an extra SA 2.1 power level. They were just not willing to add `VTX table` support.
Software can't be written without hardware to test it on. Here is the iNav hardware policy - https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

Did you try what Lisa suggested? Wiring up the VTX directly to a TBS CRSF receiver, and use the TBS Lua script in OpenTX to control its power levels.

closed 2022-02-03 by DzikuVx