iNavFlight/inav #6531: Do not immediately activate AIRMODE when arming with yaw stick arming safety bypass and mc_airmode_type = STICK_CENTER

avsaase, 2021-01-28:
## Current Behavior
When using `mc_airmode_type = STICK_CENTER` and arming using the yaw stick override as by `nav_extra_arming_safety = ALLOW_BYPASS`, AIRMODE tries to yaw the quad. This causes the motors to spin up a lot directly after arming and the quad to take off, which is a safety hazard.

## Desired Behavior
When using this combinations of settings, wait until the yaw stick is centered again before activating airmode.

breadoven, 2021-03-08:
I noticed in https://github.com/iNavFlight/inav/pull/6685 that you say you now Arm using the Arm switch flicking method. I had this Bypass Arming issue and changed airmode_type from STICK_CENTER to THROTTLE_THRESHOLD ... which stopped the motors spinning up when you Armed. Wondering why you don't use this which is recommended for smaller MR's. This is the first MR I've built so I'm new to this side of things, still trying to understand most of the MR specific settings.

avsaase, 2021-03-08:
I'm also new to the MR side of the hobby and I initially had problems getting airmode to activate using the `THROTTLE_THRESHOLD` option (#6532). Now I cannot reproduce the problem anymore so it is possible I didn't raise the throttle enough. Like you I have since switched to `THROTTLE_THRESHOLD` which indeed solves the problem. The default is still `STICK_CENTER`, though, so I think the behavior should be changed to only trigger airmode if the sticks were centered at least once after arming and then moved away from the center.

b14ckyy, 2024-03-20:
https://github.com/iNavFlight/inav/pull/7714 should also fix this issue here I think? @DzikuVx if not, feel free to reopen

closed 2024-03-20 by b14ckyy