iNavFlight/inav #6532: Airmode is deactivated when throttle is lowered below mc_airmode_threshold, instead of staying active until disarm

avsaase, 2021-01-28:
## Current Behavior
When using `mc_airmode_type = THROTTLE_THRESHOLD`, lowering the throttle below `mc_airmode_threshold` after raising it above this level disables airmode, instead of it staying active until disarm as documented. So you loose all stabilization when throttle is low, which defeats the purpose of airmode.

## Steps to Reproduce

1. Set settings as described above
2. Arm
3. Raise throttle above `mc_airmode_threshold`
4. Lower throttle below `mc_airmode_threshold` and notice the stabilization is turned off

## Expected behavior
Airmode should stay active until disarm.

## Additional context
I'm using the current master branch but as far as I know no changes have been made recently to the relevant code.

DzikuVx, 2021-01-29:
Are you testing on platform AIRPLANE or MULTIROTOR? Last time I checked on one of my quads, it was working just fine?

avsaase, 2021-01-29:
Sorry forgot to mention that. I'm using a multirotor.

DzikuVx, 2021-01-29:
I fly regularry with THROTTLE_THRESHOLD doing acro stuff and I do not notice any problems with airmode handling. The only way it can be deactivated is here:

```
if (!ARMING_FLAG(ARMED)) {
            /*
             * Disarm disables airmode immediately
             */
            DISABLE_STATE(AIRMODE_ACTIVE);
        } else if (
            !STATE(AIRMODE_ACTIVE) && 
            rcCommand[THROTTLE] > rcControlsConfig()->airmodeThrottleThreshold &&
            (feature(FEATURE_AIRMODE) || IS_RC_MODE_ACTIVE(BOXAIRMODE))
        ) {
            /*
             * Airmode is allowed to be active only after ARMED and then THROTTLE goes above
             * activation threshold
             */
            ENABLE_STATE(AIRMODE_ACTIVE);
        } else if (
            STATE(AIRMODE_ACTIVE) &&
            !feature(FEATURE_AIRMODE) &&
            !IS_RC_MODE_ACTIVE(BOXAIRMODE)
        ) {
            /*
             *  When user disables BOXAIRMODE, turn airmode off as well
             */
            DISABLE_STATE(AIRMODE_ACTIVE);
        }
 ```
 
 and it's either disarm or airmode disabling

avsaase, 2021-01-29:
Strange. I don't see any issue here. Perhaps I am misunderstanding what airmode is supposed to do. Here is a video: https://www.youtube.com/watch?v=LeLI5BI6M9M. I armed and moved the quad to show that the stabilization is not active. I then raised the throttle and the stabilization becomes active. But when I lower the throttle again the stabilization is disabled again. 

I should add that I'm am using a custom firmware but none of my changes related to throttle behavior (just my open PR's and a commit that makes the firmware pretend to be version 2.6 for compatibility with the SpeedyBee app).

avsaase, 2021-01-29:
Just tried it with the 2.6 release. Same problem, no stabilization and control when lowering the throttle below `mc_airmode_threshold `. Here is my diff: https://pastebin.com/raw/hWHxuCPd.

avsaase, 2021-02-23:
I must have not raised the throttle enough for airmode to trigger because I can't reproduce this problem anymore.

closed 2021-02-23 by avsaase