iNavFlight/inav #6535: New Target Request; MAMBAF405US_I2C

Animetronics, 2021-01-29:
## Current Behavior
Unknown; 
Possible I2C pin and timer assignment conflict with the new line if Mamba MK3 boards with barometers.:
[https://www.diatone.us/collections/new/products/mamba-stack-basic-f405-mk3-40a-6s-8bit](url)

## Desired Behavior
A new target , MAMBAF405US_I2C with the new resource assignments.

## Suggested Solution
Porting over these new assignments from 
[https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/DIAT-MAMBAF405US_I2C.config](url)

It appears that the pins for I2C are different from the existing MAMBAF405US target:
[https://github.com/iNavFlight/inav/tree/master/src/main/target/MAMBAF405US](url)

Namely:
#I2C
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09

#TIMERS
timer B09 AF3
#pin B09: TIM11 CH1 (AF3)
...
timer B08 AF2
#pin B08: TIM4 CH3 (AF2)

## Who does this impact? Who is this for?
New users of the upcoming MAMBAF405 MK3 stacks with integrated barometers.

## Additional context
The new board layout can be seen here:
[https://cdn.shopify.com/s/files/1/0027/2708/4144/files/6-MAMBA_Basic_F405_MK3.pdf?v=1610714257](url)

Animetronics, 2021-02-11:
The following modded hex file has potentially fixed the issue:
[TEST_MAMBAF405US_I2C.zip](https://github.com/iNavFlight/inav/files/5964693/TEST_MAMBAF405US_I2C.zip)

sq-meng, 2021-05-02:
I was not able to test the attached 2.7.0 build due to the lack of a suitable configurator. But...
Compiling from 2.6.1 source making the mentioned modifications, motor 4 won't spin if DSHOT protocol was selected. ONESHOT125 worked fine. I had additionally changed the line
#define USE_I2C_DEVICE_2 to 1, but that didn't make a difference.
The BF config file used a different DMA channel allocation scheme. I tried to change the target.c file and that seemed to work - all 4 motors spin under DSHOT300 now.
The change  to target.c was:
from
DEF_TIM(TIM1,   CH2,  PA9,   TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 1 ),     // S4_OUT – D(2, 2, 6)
to
DEF_TIM(TIM1,   CH2,  PA9,   TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0 ),     // S4_OUT – D(2, 2, 6)
Not sure if I traded one problem with another, but hope that helps.

closed 2021-05-08 by DzikuVx