iNavFlight/inav #6536: Target naming IFLIGHTF4_SUCCEXD incorrect / confusing

kreidler, 2021-01-29:
I opened this issue accidently in https://github.com/iNavFlight/inav-configurator/issues but closed it yet.

I assume that my request is similar to https://github.com/iNavFlight/inav-configurator/issues/1057. But meanwhile there is a Succex target (without TwinG) available.

On https://github.com/iNavFlight/inav/releases/tag/2.6.0 FC is named 'iFlight SucceX-E F4 Stack' under recommended hardware and the BG link points to the corresponding Succex-E FC.
As asset you can find only: https://github.com/iNavFlight/inav/releases/download/2.6.0/inav_2.6.0_IFLIGHTF4_SUCCEXD.hex
Unfortunately both namings are available FC from iflight: Succex-E and Succex-D

Or has BETAFLIGHTF4 still to be used for that FC?

I would like to buy the Succex-E but without a working target it will not make sense. Please clarify.

filipmaly, 2021-02-14:
Hi, I have same problem. I  wiped my iFlight SucceX-E F4 Stack new board and upload wrong version of iNav. What can I do now ? Do you have some one hex file for the board please ?

lavachemist, 2021-04-27:
@digitalentity did you know that the releases page for INAV configurator recommends an FC that has no INAV target?

[(https://github.com/iNavFlight/inav-configurator/releases/tag/2.6.1)]

lavachemist, 2021-04-27:
@kreidler Succex-D is a product line offered by Iflight. The target is not incorrectly named.

kreidler, 2021-05-02:
@lavachemist Please re-read my issue: I know that there are two different FC types from iflight but the link and naming on https://github.com/iNavFlight/inav/releases/tag/2.6.0 refers to a Iflight SucceX-E FC. That seems to be WRONG if the target is named SuccexD.

As @filipmaly realised the succexD target does not work on a succexE FC. Meanwhile I can find Succex-D F4 AIO whoop boards only which is in my opinon not really senseful for INAV but I do not know the origin intension.

closed 2022-02-03 by DzikuVx