iNavFlight/inav #6542: Home direction arrow drifts counterclockwise after takeoff

machupicchu27, 2021-01-31:
## Current Behavior
When I take off and fly away straight the RTH arrow rotates counterclockwise ~90 degrees. The direction is correct when I turn back towards home and the problem mostly goes away after flying around and rotating 360 degrees. I haven't had a problem with the actual RTH functionality. [This video](https://drive.google.com/file/d/1Snbbq99iZ8KVQDMtGYfZ8ulPWE22EMEe/view?usp=sharing) shows the issue (I don't fly perfectly straight away but the arrow is clearly not pointing towards home). It's 5 minutes long total in case you want to see how the problem improves but you can see the arrow drifting in the first 20 seconds of the video. 

## Steps to Reproduce
1. Arm the quad
2. Fly away from home

## Expected behavior
Arrow should point straight back towards home and rotate with the quad.

## Suggested solution(s)

## Additional context
I've calibrated the compass multiple times and it shows the correct heading on the bench. 
[CLI Dump](https://pastebin.com/hjQtbq9v)

- FC Board name and vendor: HOLYBRO KAKUTEF7
- INAV version string: INAV/KAKUTEF7 2.6.0 Dec  4 2020 / 11:38:46 (cada68e4) GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Jetrell, 2021-01-31:
Make sure your compass is away from any source of electromagnetic interference. High current wires etc.
Ensure `mag_declination` is set to your geographic location.
Looks to be compass interference.  When the quad is level the home arrow points correctly. When it tilts; the home arrow is incorrect.

machupicchu27, 2021-02-01:
> Make sure your compass is away from any source of electromagnetic interference. High current wires etc.
> Ensure `mag_declination` is set to your geographic location.
> Looks to be compass interference. When the quad is level the home arrow points correctly. When it tilts; the home arrow is incorrect.

It's on a mast away from everything. I never changed `inav_auto_mag_decl` so it should automatically calculate that. I did suspect tilting could be the issue as well but I don't think it is since the issue fixes itself after turning. Also, I reflashed INAV to my fc and I didn't have this problem anymore. Albeit, I flew in a new location and only one flight.

DzikuVx, 2021-06-10:
MAG heading drifting after take off suggest throttle corelation. Ergo, MAG is too close to motors, battery wires, etc.

closed 2021-06-10 by DzikuVx