iNavFlight/inav #6546: Quad Won't Arm - States there is no RC link but the receiver works

duncansshand, 2021-01-31:
## Current Behavior

Specs:
iNav 2.6
HGLRC F7222 
Spektrum 4650 SRLX2 Receiver
Spektrum DXe Transmitter

Everything is working perfectly on my quad, it is passing all pre-aming validation check but I cannot arm the quad. 

In the CLI it states it won't arm due to the RC link not being detected but the receiver tab shows inputs and movements from RC.  On the OSD it also states it is in '!FS!' and 'Unable to arm - No RC link' but the RSSI varies around 98 and the throttle is registered on stick movements. 

Due to the flight controller stating there is no RC link the quad is constantly in failsafe mode.

[image]

[image]

[image]

In the screenshot below it says in the first status that it is unable to arm due ARMSW RX CLI. After deactiving the armswitch, you can see in the second status that is no an arming disabled flag therefore the flight controller is registering the RC inputs.

[image]

I have tried 2 different UARTs and had no luck, both on TX pads as advised. 

## Suggested solution(s)

I thought that it could possibly be the receiver sending false signals for unused channels (e.g. channel 16) that are out of range and is triggering the failsafe. 

## Additional context

[image]

breadoven, 2021-01-31:
Channels 24 and 25 seem to be showing non zero values below the valid channel signal value (rx_min_usec = 750). Could be causing the problem.

duncansshand, 2021-01-31:
> Channels 24 and 25 seem to be showing non zero values below the valid channel signal value (rx_min_usec = 750). Could be causing the problem.

Yeah that was another thing I thought could be causing the problem. Sometimes the channels read ~65300 

I have no idea what to do here, is there anyway in the CLI to disable these unused channels?

duncansshand, 2021-02-01:
Solved the problem.

Downgraded the firmware on the receiver to a version released back in March 2019. 

One of the later firmware updates had a feature that sent the failsafe signal upon initiate which was locking iNav in failsafe mode.

The drone now arms and doesnt stay in failsafe.

Duncan

closed 2021-02-01 by duncansshand

kochge, 2024-08-20:
I've the same problem. SRLX2 works fine with DX8i transmitter, but it fails on DX6i. 
It seems that the SRLX2 Receiver reports invalid values on some channels when using the DX6i transmitter, but it works fine on DX8i. Is there any possibility to disable the failsafe check on these higher channels, as they are not required anyway?