iNavFlight/inav #6547: Allow to set "initialBaroAltitudeOffset"

poi2poi, 2021-02-01:
## Current Behavior

File: \inav-master\src\main\navigation\navigation_pos_estimator.c
Line: 306
static float initialBaroAltitudeOffset = 0.0f;

## Desired Behavior

Let user set different value for initialBaroAltitudeOffset because of varying atmospheric pressure

## Suggested Solution

## Who does this impact? Who is this for?

Important when setting close-to-ground waypoints i.e. flying over mountains or other obstacles

## Additional context

breadoven, 2021-02-01:
I assume this would be for long flights when atmospheric pressure might change enough to cause problems and you would have to set this offset by telemetry. Otherwise I don't see the need for an offset since atmospheric pressure is accounted for when you set the home altitude at arming. All other waypoint altitudes are relative to this home altitude.

poi2poi, 2021-02-01:
Hi Breadoven

The scenario you describe is one possible use-case, for me it's more important that I don't have to decide at home, when I'm planning the "mission", where I'm going to launch my wing. So I can set the waypoints with Altitude above sea level and then set the absolute altitude of my take off. Otherwise I would need to change the waypoints on take-off.

poi2poi, 2021-02-01:
well I figured out there is another variable 'baroGroundAltitude' in \inav-master\src\main\sensors\barometer.c that would be the better option to hook in

breadoven, 2021-02-01:
Yes I realised that is another way of doing it, relative to sea level. Makes it easier when you're ground skimming rather than having to work out height above ground relative to where you took off from.

poi2poi, 2021-02-01:
I think i will implement the option to set QNH (you can get it in any weather app or from your local airport) so setting the offset would be obsolete. I will try to make a push to github after testing it.

digitalentity, 2021-02-01:
A proper way of addressing your use-case would be to allow per-waypoint definition of altitude being relative to launch or absolute (above MSL). Feel free to open a feature-request for your use-case.

Internal sensor offset values are not intended to be user-configurable. Closing as wont-fix.

closed 2021-02-01 by digitalentity