iNavFlight/inav #6561: Separate flight pitch angle and board orientation with setting for pitch angle

fernandodlc, 2021-02-02:
Many users have trouble getting the pitch angle right for level flight in angle mode.

The current method involves changing the board orientation to suit the desired angle. Whilst this is problem free once done it is far from the obvious solution, involves guesswork and can only be changed hooked up to a configurator.

I propose separating the "flight pitch angle" from the angle of the board in the plane, ideally with a configurator and OSD setting. This would be a logical separation to make. The angle the board is level in the plane and the angle you desire to fly at are not logically the same thing. This means it could be included in things like presets to make angle mode work most of the time independent of the flight controller mounting.

fernandodlc, 2021-02-02:
Additional reasoning: The angles you use in the advanced tuning tab and auto-launch do not represent true values if the board orientation has been changed to account for level flight.

avsaase, 2021-02-03:
@DzikuVx, is there a PR for this already? I'd like to follow along.

DzikuVx, 2021-02-03:
@avsaase not really. Very early code resides only on my local drive ATM

digitalentity, 2021-02-03:
By design "bord is level" equals "flying level". It has nothing to do with "aircraft laying flat on the floor", in the end the aircraft is intended to fly, right. Pitch, roll, launch langes are relative to **level flight**, not "level on the ground". Ultimately the two should be identical, but that's a mechanical problem of deciding wing angle of attack, not the software problem IMO.

digitalentity, 2021-02-03:
Decoupling trimming from board alignment might make sense though.

fernandodlc, 2021-02-03:
Thanks @DzikuVx and @digitalentity!

DzikuVx, 2021-02-03:
@avsaase see #6564 but this is not operational yet

fernandodlc, 2021-02-03:
@digitalentity I'm not sure I agree with level flight being the reasonable starting point for config values. I will happily work with whatever you guys decide but in the spirit of constructive discussion:

- Pitch in level flight is dependent on airspeed. Airspeed is dependent on a number of things. Level on the ground is always consistent.
- In manned aviation the longitudinal axis of the plane is commonly used when talking about AoA. From the quote below it seems there's no real issue as long as it's consistent but "plane flat on the ground" is what rc manufacturers tend to build shelves for so it's probably the easiest thing for us to be consistent with.

"Angle of attack (AOA) is the angle between the oncoming air or relative wind and a reference line on the airplane or wing. Sometimes, the reference line is a line connecting the leading edge and trailing edge at some average point on the wing. Most commercial jet airplanes use the fuselage centerline or longitudinal axis as the reference line. It makes no difference what  the reference line is, as long as it is used consistently."
https://www.boeing.com/commercial/aeromagazine/aero_12/whatisaoa.pdf

PR #6568 (2021-02-05): Ability to trim pitch angle for level flight
Closes #6561

The `fw_level_pitch_trim` can be used to trim pitch angle and is separated from the board orintation option.
To make it more user friendly, **positive** values means that nose should be raised **UP** for level flight.  

The next step would be to add it to the Adjustments section and remove ROLL and PITCH board adjustments from Configurator GUI and Adjustments
closed 2021-02-05 by DzikuVx