iNavFlight/inav #6571: RC via Mavlink

harry1453, 2021-02-04:
## Current Behavior

Separate Mavlink and RC UART connections are required when using Crossfire. Or any control link, for that matter.

## Desired Behavior

Be able to control an aircraft using a single Mavlink UART via Crossfire. Also, control via joystick on GCS etc.

## Suggested Solution

Add Mavlink as a serial RX provider. Once selected, RC commands are taken from Mavlink.
**I plan to implement and PR this myself**.

## Who does this impact? Who is this for?

Anyone using Crossfire who wants Mavlink mission upload / telemetry and RC on the same UART (especially important on F411 chips!)

## Additional context

This is just a tracking issue to gauge interest (as I've seen this asked for before) and for any relevant discussion, **I plan to implement and PR this myself as long as maintainers think this would be a useful contribution**.

cgrrty, 2021-02-13:
Hi
I want RC via Mavlink with openhd
Thanks

cgrrty, 2021-03-20:
How's it going? when will it be released?

wx4cb, 2021-03-22:
would be nice, but i think they need to fix the mavlink implementation itself first :) that said, a full mavlink v2 implementation would be cool

cgrrty, 2021-03-23:
PX4 and ardupilot support RC with mavlink now

harry1453, 2021-03-24:
@cgrrty as I said in telegram I've been working on it, I have an implementation but have not had the time to try it out beyond compiling it, so it's not ready yet at all. unfortunately other things are taking up my time at the moment. I promise I'll get back on this soon

@wx4cb I've implemented MAVLink V2 as part of my changes.

wx4cb, 2021-03-24:
@harry1453 cant wait

PR #6738 (2021-04-10): RC via MAVLink & MAVLink V2 Support
Closes #6571 

This PR

- Updates the MAVLink library to V2 (**this is the reason this PR is so huge, it's mostly auto generated, sorry 😢**)
- Implements some of the new V2 fields in the messages already in use
- Adds scripts for both Windows and Mac/Linux to easily automatically generate the MAVLink library should any updates be needed in the future
- Adds command line option `mavlink_version` to switch back to MAVLink V1 for backwards compatibility (defaults to V2)
- Adds serial receiver option "MAVLink". With this receiver option, MAVLink [RC Channels Ove [... cut]
closed 2021-04-10 by DzikuVx