iNavFlight/inav #6573: Mavlink default implementation compatibility

padcom, 2021-02-05:
## Current Behavior
Connecting to QGroundControl over serial or UDP (with Crossfire) results in the following logs in QGroundControl:

```
Switching outbound to mavlink 2.0 due to incoming mavlink 2.0 packet: 0x562fb4826e18 3 2
```

## Steps to Reproduce

1. Define a serial port as MAVLink telemetry
2. Connect QGroundControl either via crossfire UDP proxy or directly via serial port (checked with both)

## Expected behavior
QGroundControl connects to iNAV

## Additional context

# INAV/MATEKF722SE 2.6.0 Dec  4 2020 / 12:21:50 (cada68e4)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Please note that QGroundControl is the canonical implementation of MAVLink. Everything else is basically an offspring of that GCS. Therefore, naturally, it should work out of the box. Maybe it is some checksum being sent initially that lets the QGroundControl know it tries to use MAVLink 2.0 and it is not really a 2.0 client...

digitalentity, 2021-02-05:
Does QGC work if you connect it directly to the serial port of the FC? If so - it's probably CrossFire messing up the MAVLink stream.

padcom, 2021-02-07:
As I have already stated in the issue it doesn't work with either MAVLink proxy via CRSF nor with direct serial connection. At least I have not been able to make it work with any of those.

digitalentity, 2021-02-08:
Noted. We likely need to upgrade to a fresh MAVLink library.

wx4cb, 2021-03-22:
mavlink into mission planner works fine via UDP on crossfire - that's how i use it and also via bluetooth to my tracker from crossfire.

that said, i've never gotten QGC to work properly

personally i use mwptools on linux or MP on windows

wx4cb, 2021-03-23:
@padcom all i can say is that mp and mwp dont have any noticeable issues with tracking the plane 2+ miles out and neither does my antenna tracker that is on the crsf bluetooth

DzikuVx, 2021-04-10:
@padcom MAVLink was updated in INAV via #6738 
Please check with INAV 3.0 if problem persists

closed 2021-04-10 by DzikuVx